emsesp/EMS-ESP32 #111: Errors reading from Solar Module

proddy, 2021-09-16:
hi, im not sure that im writing in right place, 
i get
2021-09-16 15:46:26.392   ERROR 671:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 62
2021-09-16 15:55:18.997   ERROR 672:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 58
2021-09-16 15:58:03.506   ERROR 673:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 19 03 20
2021-09-16 15:58:40.901   ERROR 674:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 63
2021-09-16 16:10:30.787   ERROR 675:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 62

i have replaced esp8266 to esp32, and added a sm200 module,  im not sure what device generate this frames,  could be sth wrong with solar module ?

_Originally posted by @Bartekn86 in https://github.com/emsesp/EMS-ESP32/issues/107#issuecomment-920940048_

proddy, 2021-09-16:
@Bartekn86  did it work previously in V2 (the ESP8266)?

Bartekn86, 2021-09-16:
I'm not sure, I wouldn't want to mislead you

proddy, 2021-09-16:
can you post your system info (from web console, go to help)

Bartekn86, 2021-09-16:
sure
[emsesp_system_info-1.txt](https://github.com/emsesp/EMS-ESP32/files/7179078/emsesp_system_info-1.txt)

MichaelDvP, 2021-09-16:
I was a bit confused by the offset 03, but this was only a bug in reporting (fixed now). There is also a read fail on boiler telegram 19, it's not only solar. Tx-errors are often power issues. How is the emsesp powered? If 8266 was bus-powered, and esp32 is usb-powered don't forget to remove the jumper/powermodule.

BTW: A few days ago i've tested MH-ET32 buspowered with low clock, low bandwidth and tx-power 8 dBm, working with ~90% tx and 90% rx quality, that's mainly the same i get with 8266. (with servicejack power it's always 100% quality).

Bartekn86, 2021-09-16:
i have esp32 with usb-power, jumper removed firstly :-)

MichaelDvP, 2021-09-16:
Your system info looks ok, there is only one tx-error in 55 sends and 99% quality. Also the first post shows for solar telegram 02 62 the errors 671 and 675 with time difference of 24 minutes. The telegram is polled every minute, so we have 23 successfull polls between these errors. It's not perfect but nothing to worry about and the solar telegrams are working.
You can try to if it gets better with different bus connections, shorter/longer/thicker cables, etc.

Bartekn86, 2021-09-16:
thank you, i will try with a bit longer cables between esp and bc25. theses  are shorter than 100mm, meyby this is a problem.. hm...

Bartekn86, 2021-09-18:
nothing change after replace wire, prolong wires, itp, itd.  I have 100 % rx and tx quality, and still 2021-09-18 22:28:05.134   ERROR 93784:  [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 19 20 01 A5

proddy, 2022-01-26:
@Bartekn86 did you try with v3.4 and lowering the WiFi transmit power?

closed 2022-10-31 by proddy