emsesp/EMS-ESP32 #112: wwonetimecharging not working

Ivannefit, 2021-09-18:
Hello,

The ww one time charging does not work on my nefit enviline tower 2 (monoblock)

The option to adjust is not available EMS-ESP web interface.

In Domoticz is no hardware available for wwonetimecharging.

[image]

When i use the Nefit phone app the ww temperature is default on 70 degrees, The only thing i need to set is the time that extra ww wil be active .( varies from 15 min to 48 hours in steps of 15 min)

[image]

Device information
[image]

proddy, 2021-09-18:
can you post your system info (from the help tab) so we can see the devices

Ivannefit, 2021-09-18:
Here is the info.

{
  "System": {
    "version": "3.2.1",
    "uptime": "003+01:45:06.741",
    "uptime_sec": 265506,
    "freemem": 174
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 808472,
    "read requests sent": 57535,
    "write requests sent": 0,
    "incomplete telegrams": 21,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT publishes": 45073,
    "MQTT publish fails": 0,
    "dallas sensors": 0,
    "dallas reads": 0,
    "dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco (DeviceID:0x08, ProductID:172, Version:01.15)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:13.07)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Heatpump",
      "name": "HP Module (DeviceID:0x38, ProductID:200, Version:40.04)",
      "handlers": "0x42B 0x47B"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)",
      "handlers": ""
    }
  ]
}

proddy, 2021-09-18:
thanks. Could you also send the output of http://ems-esp/api/boiler/wwonetime

Ivannefit, 2021-09-18:
You mean with MQTT exploder?

proddy, 2021-09-18:
no, EMS-ESP has an REST API. The URL will query EMS-ESP and bring back the ww onetime details. You may need to replace ems-esp with the IP address

Ivannefit, 2021-09-18:
Ah ok.

  "name": "wwonetime",
  "fullname": "ww one time charging",
  "circuit": "ww",
  "value": "off",
  "type": "boolean",
  "writeable": true
}

proddy, 2021-09-18:
So we need to figure out why you can't change it from the Web UI. Just to be safe could you restart EMS-ESP and also Shift-Refresh your browser to make sure nothing is cached.

Also you can try setting the value directly from the Console. Telnet into EMS-ESP, do a `su` with the default system password and type `call boiler wwonetime` to see the settings and then try `call boiler wwonetime on` to change a value.

The feature from the Nefit app is interesting. We could consider making a new feature that does the same if you want - but then please create a new GitHub enhancement request with some specifications on how it works.

MichaelDvP, 2021-09-18:
@Ivannefit Please use the [latest dev-build](https://github.com/emsesp/EMS-ESP32/releases), sunbuzz has [recently fixed](https://github.com/emsesp/EMS-ESP32/commit/1b9a2f21d2778215c707a5a63435a30b1aaa10a0) the (for some heatpumps) too small json buffer, cutting the pencil and some other paramters from webpage. 

On some systems ww is only managed by thermostat and wwonetime-command on boiler does not work, than try thermostat wwcharge command.

Ivannefit, 2021-09-18:
Ok i have done it with versie 3.2.1 and 3.2.2b5
[image]

[image]

MichaelDvP, 2021-09-18:
> Both not working

As mentioned: Have you tried `call thermostat wwcharge on`. The command from terminal is send to the boiler, but the boiler ignores it. The dev fixes only the missing values on web-page.

Ivannefit, 2021-09-18:
now i have tried call thermostat wwcharge on.

[image]

Still nothing happens

MichaelDvP, 2021-09-18:
From [here](https://github.com/emsesp/EMS-ESP/issues/589#issuecomment-726140292): Try with `call system send "0B 10 FF 0B 01 F5 FF"`

I think the thermostat command sends 1 instead of 0xFF.

Ivannefit, 2021-09-18:
This one works, but default time is now 45 min. it would be nice to set the timer and be able to use it in domoticz

[image]

MichaelDvP, 2021-09-18:
Ok, i'll change the wwcharge-command to make it work. I think `wwchargeduration` can also be set in steps of 15min, doesn't it work?
With domoticz i can't help, i've never tried it.

Ivannefit, 2021-09-18:
wwchargeduration is not in my web interface, and with telnet i get error ( command not found)

MichaelDvP, 2021-09-18:
Should be right before wwcharge in thermostat values. Please check `call thermostat info` and `read 10 2F5`

Ivannefit, 2021-09-18:
Ya sorry was looking in boiler, wwchargeduration works properly

closed 2021-09-26 by proddy

Ivannefit, 2021-09-27:
i'm using 3.2.2.b8 now and all value that are writable in the web intrerface dont have a pull down menu.

[image]

MichaelDvP, 2021-09-27:
I think you're using a cached webpage, please reload page in browser.

Ivannefit, 2021-09-27:
I have reload the de page and pull down menu's are back, Only wwonetimecharging isn't working

MichaelDvP, 2021-09-27:
> Only wwonetimecharging isn't working

As mentioned: if ww is managed by thermostat the `boiler wwontime` is not working, you have to use `thermostat wwcharge`, this should work now.
For systems with working boiler command it is only starting if the current temp is more than 5 degrees under setpoint, otherwise the boiler ignores the command.

Ivannefit, 2021-09-27:
Ah ok ,sorry forgot about that. wwcharge is working!!!