emsesp/EMS-ESP32 #113: Timer ww one time charging

Ivannefit, 2021-09-18:
The Nefit app has a option to set temporary ww. It would be nice to be able to use this option in domoticz.

On de Nefit app you can set the time from 15 min to 48 hours in steps of 15 min.

The default temp of extra ww is set to 70 degrees  and the only thing you neet to set is the time you want extra ww.
[image]

This option can be handy example:if you take a bath set ww to 1 hour or when you take a shower 15 min,

proddy, 2021-09-18:
Thanks for adding. @MichaelDvP is this something you were planning to take a look at?

MichaelDvP, 2021-09-18:
@proddy I think we have wwchargeduration already. But new to me is that it can be set up to 48h, now it is limited to 120min by enum_wwChargeDuration. I think we should change from enum to uint (minutes), but need a multipier. Now we have only dividers (div2), (div10) as uint8_t parsed to render and round functions.

Change this to int8_t and use negative values for multipliers?
i.g.
``` 
MAKE_PSTR_LIST(mul15, F("-15"))
...
double Helpers::round2(double value, const int8_t divider) {
    double  val = (value * 100 + 0.5);
    if (divider > 0) {
        val = ((value / divider) * 100 + 0.5);
    } else if (divider < 0) {
        val = value * -100 * divider;
    }
```

proddy, 2021-09-19:
Yes, I can't think of a better way. But to keep the code clean and readable I would suggest some more changes like
- rename the variable `divider` into something else since it could be also a multiplier
- add an additional `bool` parameter to round2() which signifies if its a multiplier or divider

MichaelDvP, 2021-09-19:
> But to keep the code clean and readable

That's the problem, a negative divider for multiplier is a hack and mathematical incorrect. But making the divider float will give rounding errors for factor 15, which is divider 0.066666. Also round2 is only usefull for division, a multiplication can be handled with int. 
Better way: 
```
MAKE_PSTR_LIST(mul15, F("*15"))
...
            uint8_t divider = 0;
            uint8_t factor  = 1;
            if (dv.options_size == 1) {
                const char * s =  uuid::read_flash_string(dv.options[0]).c_str();
                if (s[0] == '*') {
                    factor = Helpers::atoint(&s[1]);
                } else {
                    divider = Helpers::atoint(s);
                }
            }
...
           json[value] = divider ? Helpers::round2(*(uint8_t *)(dv.value_p), divider) : *(uint8_t *)(dv.value_p) * factor;
```
and we need to check in `atoint()` for only valid numbers `while (*value >= '0' && *value <= '9') {`

Do you agree?

proddy, 2021-09-19:
yes agree, that's much tidier. I wrote a small 'entities' command so maybe wait until I've checked that in first

proddy, 2021-11-10:
already fixed.

closed 2021-11-10 by proddy