emsesp/EMS-ESP32 #117: Question: Command for wwdisinfecting on/off or onetime?

giovanne123, 2021-09-19:
I'm currently searching if it is posible to send a command for setting the "wwdisinfecting" on/off or something like wwdisinfectingonetime? (Like for the wwonetime?)

curl http://192.168.0.211/api/boiler/commands doesn't list a command for that so I though that it isn't supported, but wanted to ask if I'm right?

EMS-ESP32 v3.1.1
Buderus GB132

MichaelDvP, 2021-09-19:
We have the schedule of disifecting for RC35 thermostat, but no commands for other thermostats and for the boiler. 
You can try to fetch the right telegram when starting manual while logging. 
I guess it could be something with bit 2/6 of telegram 0x35, NorbertS list a flag in this position. Try `call system send "0b 08 35 00 44"`

MichaelDvP, 2021-09-19:
I've tested on my GB125 and this command works. Disinfection is started. Like onetime-command  it's only triggering and stopping `call system send "0b 08 35 00 04"` does not work for my boiler, i have to wait until the disinfection is done. Changing disinfectiontemp while running is possible.
I'll add the command to wwdisinfecting. @proddy should we rename to `wwdisinfect`?

proddy, 2021-09-19:
> I'll add the command to wwdisinfecting. @proddy should we rename to wwdisinfect?

yes, lets do that

giovanne123, 2021-09-20:
> We have the schedule of disifecting for RC35 thermostat, but no commands for other thermostats and for the boiler.
> You can try to fetch the right telegram when starting manual while logging.
> I guess [...]

Sounds great, I will test the command for my GB132 when I need the next disinfect cycle...

giovanne123, 2021-09-30:
Sorry for late reply.
I have tested the call for starting the disinfect and it is working (call system send "0b 08 35 00 44")
For my boiler stopping does also not work with (call system send "0b 08 35 00 04"). I have to wait too.

(I have two entities, maybe old artifacts?)
[image]

after a while changed back to "off"
[image]

**Additional question for me is, what setting does the call change?**

In the Buderus software there is a flag and configs for activating and defining the weekday and time:
Will this be additional information and currently not supported in EMS-ESP32?
[image]
I checked the disinfect (Thermische Desinfektion) in EMS RC30 Controller and the setting there was all the time "off" when the disinfect was running after the call. I think it will be in the Buderus software in the screenshot above also all the time "off" but I can't check - would need to reactivate the software/solution for that to check.

But I think would also be interesting to know im EMS-ESP32/HA is the weekly disinfect is activated or not, or?

MichaelDvP, 2021-10-01:
> (I have two entities, maybe old artifacts?)

Yes, both ;-) Double `ww` and missing `ww` are wrong, actual iteration is one `ww`.

The disinfection was added with some other parameters to RC35 short time ago, i think these values are also in RC30:
wwProgramm, wwCirculationProgram, wwDisinfect, wwDisinfectDay, wDisinfectHour, wwMaxTemp and wwOneTmeKey. 
I'll add it to dev, please check.

What buderus software/solution is this? This could be helpfull to figure aut some other settings.

giovanne123, 2021-10-01:
> > (I have two entities, maybe old artifacts?)
> 
> Yes, both ;-) Double `ww` and missing `ww` are wrong, actual iteration is one `ww`.
> 
Hm, I will check my entities...

> The disinfection was added with some other parameters to RC35 short time ago, i think these values are also in RC30: wwProgramm, wwCirculationProgram, wwDisinfect, wwDisinfectDay, wDisinfectHour, wwMaxTemp and wwOneTmeKey. I'll add it to dev, please check.
> 
Ok, I will have a look to the dev...

> What buderus software/solution is this? This could be helpfull to figure aut some other settings.
>
I was using the Buderus Eco-Soft Software (Logamatic_ECO-SOFT_4000_EMS) with Buderus Easycom before...

giovanne123, 2021-10-02:
Updated to latest DEV - v3.2.2b10
In console I think this will be the command for one time disinfecting to start, or? (haven't tried because no need for disinfect at the moment ;-)
(I will add/test it to my HA for a script call using mqtt with **topic: ems-esp/boiler** **Payload: {"cmd":"wwdisinfect","data":1}**)
```
ems-esp32:# show commands
...
 boiler:
   wwdisinfect           ww disinfection *
...
```

My entyties are fine now after recreating them in HA.

Also the Boiler/Thermostat informations for disinfect are now available:
[image]

[image]

May I allowed to ask one question not directly related here, but "wwOneTmeKey" you mentioned I can't see it. What does this stand for? Maybe not related to me but I'm interested ;-)

MichaelDvP, 2021-10-03:
>  think this will be the command for one time disinfecting to start, or?

Yes, the boiler command starts a single disinfection, the setting in thermostat is for the scheduled disinfection, maybe we should change the full name to `thermostat ww disinfection schedule`?
> but "wwOneTmeKey" you mentioned I can't see it.  

Maybe this is only in RC35, check `read 10 37` the byte in offset 9. It's only the function of the water-key on the RC35, if pressing starts a oneTimeHeating or only circulation.

proddy, 2021-11-10:
@giovanne123 do you still want us to pursue this issue?

giovanne123, 2021-11-10:
@prddy, yes would be great if it get's into stable version.
I have tested in Version v3.2.2b10
- a) command for one time disinfect: call system send "0b 08 35 00 44"
- b) support desinfect infos for RC30 was added Form  MichaelDvP  to dev: https://github.com/emsesp/EMS-ESP32/issues/117#issuecomment-931985813

proddy, 2021-11-10:
> a) command for one time disinfect: call system send "0b 08 35 00 44"

@MichaelDvP already added that with the command is `wwdisinfect` and takes a boolean, but it doesn't work on your boiler which uses a different command (0x35 UBAFlags instead of 0x33 UBAParameterWW) correct?

closed 2021-12-29 by proddy