emsesp/EMS-ESP32 #120: EMS ESP Console problems

kempa007, 2021-09-20:
Hi,

I have just purchased  EMS E32 gateway. It works, but when I log in to the console I cant enter boiler, thermostat submenu. Do I need to turn it on somewhere?

kempa007, 2021-09-20:
Thanks for making it clear. I know it is not openhab forum, but I want to send command to change thermostat temperature. I'm using mosquitto MQTT broker. I tried to run this:

Type number : thermostat_cmd_temp "Thermostat setpoint temp" [commandTopic="ems-esp/thermostat_cmd_temp" ]

But it doesn't work. Do you know how to send command by Openhab MQTT binding? Reading values from EMS bus works like a charm.

MichaelDvP, 2021-09-20:
See the doc [here](https://emsesp.github.io/docs/#/MQTT?id=sending-commands):
You have to send to topic *"thermostat"* `{"cmd":"seltemp", "data":21,"hc":1}`
or activate the single subscriptions in settings and you can write the temperature as number direct to topic *"thermostat/hc1/seltemp"* or *"thermostat/seltemp"* if you have choosen main_heatingcircuit.

kempa007, 2021-09-20:
Great hint :) I'm a bit closer to solve my problem. But after thing:

[commandTopic="ems-esp/thermostat", formatBeforePublish="{\"cmd\":\"seltemp\", \"data\": %s, \"hc\":\"1\"}

I am getting an error:
000+21:17:50.903 W 0: [command] Command seltemp on thermostat not found
000+21:17:50.903 E 1: [mqtt] No matching cmd (seltemp)

proddy, 2021-09-20:
Some more information about your system will help. Like what thermostat do you have (do a http://ems-esp/api/system)?

If your thermostat does not support Write commands then the command will not work. You can also check by going to the console and typing `# show commands`

kempa007, 2021-09-20:
Yes, forgive me. I think it should, cause I can change termostat temperature from EMS-ESP web panel. This is an output form show commands:

` thermostat:
  building              building *
  [hc] comforttemp      comfort temperature *
  commands              list all commands
  [hc] controlmode      control mode *
  datetime              date/time *
  [hc] designtemp       design temperature *
  [hc] ecotemp          eco temperature *
  info                  list all values
  [hc] manualtemp       manual temperature *
  [hc] maxflowtemp      max flow temperature *
  minexttemp            minimal external temperature *
  [hc] minflowtemp      min flow temperature *
  [hc] mode             mode *
  [hc] nofrosttemp      nofrost temperature *
  [hc] offsettemp       offset temperature *
  [hc] program          program *
  [hc] roominfluence    room influence *
  [hc] seltemp          selected room temperature *
  [hc] summersetmode    set summer mode *
  [hc] summertemp       summer temperature *
  [hc] tempautotemp     temporary room temperature automode *
  wwcircmode            circulation pump frequency *
  wwmode                mode *
  wwsettemp             set temperature *
  wwsettemplow          set low temperature *
`

proddy, 2021-09-20:
could you still do a http://ems-esp/api/system ?

kempa007, 2021-09-20:
Yes of course. Thanks for your time: 

<html><body>
<li class="tabs-menu-item rawdata " role="presentation"><span class="devtools-tab-line"></span></li><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div class="toolbar"><div class="devtools-separator"></div><input class="searchBox devtools-filterinput" placeholder="Filtruj JSON"></div><div class="panelContent">

  |  
-- | --
System |  
version | "3.2.1"
uptime | "000+22:31:01.667"
uptime_sec | 81061
freemem | 132
Status |  
bus | "connected"
bus protocol | "Buderus"
telegrams received | 134314
read requests sent | 31095
write requests sent | 2
incomplete telegrams | 2
tx fails | 1
rx line quality | 100
tx line quality | 100
MQTT publishes | 33827
MQTT publish fails | 0
dallas sensors | 0
dallas reads | 0
dallas fails | 0
Devices |  
0 |  
type | "Boiler"
name | "Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.08)"
handlers | "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
1 |  
type | "Thermostat"
name | "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)"
handlers | "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
2 |  
type | "Thermostat"
name | "RC100/Moduline 1000/1010 (DeviceID:0x39, ProductID:165, Version:40.04)"
handlers | "0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
3 |  
type | "Mixer"
name | "MM100 (DeviceID:0x21, ProductID:160, Version:24.05)"
handlers | "0x2D8"
4 |  
type | "Mixer"
name | "MM100 (DeviceID:0x20, ProductID:160, Version:24.05)"
handlers | "0x2D7"
5 |  
type | "Controller"
name | "BC25 (DeviceID:0x09, ProductID:125, Version:03.03)"
handlers | ""

</div>
</body>
</html>

proddy, 2021-09-20:
thanks! I see the problem. You have multiple Thermostats and the command is going to the wrong one. This was fixed in the dev release but has yet to make it to the stable. You can get the binary from https://github.com/emsesp/EMS-ESP32/releases/tag/latest 

and then re-try

sorry and thanks for your patience!

kempa007, 2021-09-20:
Okay, I will try, but have you got some tutorial how to upgrade my hardware? I am using E32 gateway form https://bbqkees-electronics.nl/.

kempa007, 2021-09-21:
I just upgraded firmware, but the error is still the same: 

000+00:04:05.194 W 0: [command] Command seltemp on thermostat not found
000+00:04:05.194 E 1: [mqtt] No matching cmd (seltemp)

kempa007, 2021-09-21:
I figured it out. It was a problem with formating the command:
this one didnt work:

[commandTopic="ems-esp/thermostat", formatBeforePublish="{"cmd":"seltemp", "data": %s, "hc":"1"}

but this one runs well:

[commandTopic="ems-esp/thermostat", formatBeforePublish="{cmd:\"seltemp\", data: %s, hc:\"1\"}

proddy, 2021-09-21:
good!