emsesp/EMS-ESP32 #121: Bosch Condens GC2300W/Logamax Plus GB122 - missing error message

partikus, 2021-09-20:
**Bug description**

Once the boiler stopped working due to an issue, the error sent is not being handled. 

```
?(0xBF), data: 08 EA 00 06 11 20 2D 2D 0B 87 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Based on [the error message type](https://emsesp.github.io/docs/#/MC110-controller?id=type-0xbf-deviceserrormessage) I've decoded the following values:

```
# pattern
[byte:{number}] | [value] - [notes]

[b:0] | 08 - [8] device ID
[b:1] | EA - [234] - model ID
[b:2] | 00 - [0] - blocking error: always 0x0A (fault class, see service manual) 
[b:3] | 06 - [6] blocking error: 0x00 or 0x10 = interlocking error, 0x11 = blocking error, 0x12 = system error, 0x13 = reset system error
[b:4] | 11 - [DC1] - device control ASCII | Fault code 0. character (possibly 0x0a = space)
[b:5] | 20 - [ ] ASCII | Fault code (operation code) 1st character
[b:6] | 2D - [E] Fault code (operation code) 2nd character
[b:7] | 2D - [E] additional code (status code)
[b:8] | 0B - [...
[b:9] | 87 - ...] - 2951 - data 
```

```
Boiler: Bosch Logamax Plus GB122 (DeviceID:0x08 ProductID:234, Version:05.03)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA
 ```

**Device information**
```json
{
  "WIFI": {
    "ssid": "XXXX",
    "hostname": "ems-esp",
    "static_ip_config": "off"
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": "on",
    "host": "192.168.98.134",
    "port": 1883,
    "username": "mqtt",
    "client_id": "ems-esp",
    "keep_alive": 120,
    "clean_session": "on",
    "max_topic_length": 128,
    "publish_time_boiler": 0,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 5,
    "publish_time_sensor": 0,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "on"
  },
  "NTP": {
    "enabled": "on",
    "server": "0.pl.pool.ntp.org",
    "tz_label": "Europe/Warsaw",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": "on",
    "port": 8266
  },
  "Settings": {
    "tx_mode": 2,
    "ems_bus_id": 11,
    "syslog_level": -1,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "master_thermostat": 0,
    "shower_timer": "off",
    "shower_alert": "off",
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "on",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  }
}
```

MichaelDvP, 2021-09-20:
I can not see the error structure in this c2 telegram, please @partikus explain the structure and what errors are stored there.
My boiler does not have the c2 telegram, i can not test.

As mentioned before the error log is `read 8 10` for blocking errors, `read 8 11` for nonblocking errors, `read 10 12` and `read 10 13`  for device errors (not sure what is the difference of 12/13). These telegrams have some blocks of 12 bytes containing display code (2), errornumber (2), year, month, hour, day, minute, duration, source. As `last-error` we read the first (newest) dataset of these telegrams, if logged with the homeautomation-server you can also store all errors. 

The BE/BF telegrams are for actual reporting an error from any device and the telgrams are empty if there is no error.

partikus, 2021-09-20:
I'm not sure... my suggestion can be just overinterpretation of the response but for sure it's possible to read more than last 2 errors.

`0B 84` which is `2948`
`0B 6F` which is `2927`

e.g. I did not find EN version so I believe NL should be better than PL.
[image]

I do not get fully why when I call `read 08 C2` I'm getting limited response but when I set offset bigger then 0 like `02` or so I'm getting much longer response which I've shown above.

MichaelDvP, 2021-09-21:
Indeed, i don't have such high errorcodes in my ems1.0 system, but the three bytes before the code looks like ascii-display-code:
`20 2D 2D 0B 84` => Error: `" --" (2948)` 
But i have no idea what the other bytes are, i can't see a date information, also as unix-timestamp i don't see it.
What does `read 8 10` etc. shows on your system?

partikus, 2021-09-21:
```
ems-esp:/# read 08 10
005+07:44:38.246 N 20: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage1(0x10), data: <empty>
ems-esp:/# read 08 11
005+07:44:39.728 N 21: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage2(0x11), data: <empty>
```

partikus, 2021-09-21:
I was looking for a pattern for data offset for the error messages. Looks like the 12 bytes length (0C) offset +20 bytes.

```
send telegram "0b 88 C2 08 0C"
005+09:15:36.427 N 830: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0B 84 00 0F 3D 6A 00 00 0F 3D 6B 00 (offset 8)
send telegram "0b 88 C2 1C 0C"
005+09:20:37.670 N 838: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0B 84 00 0F 2F 87 00 00 0F 2F 88 00 (offset 28)
ems-esp:/# send telegram "0b 88 C2 30 0C
005+09:23:17.250 N 841: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0B 6F 00 0F 2F 07 00 00 0F 2F 08 00 (offset 48)
ems-esp:/# send telegram "0b 88 C2 58 0C"
005+09:23:37.684 N 842: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0B 86 00 0F 1A F2 00 00 0F 1A F3 00 (offset 88)
ems-esp:/# send telegram "0b 88 C2 6C 0C"
005+09:24:33.377 N 843: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0B 84 00 0F 1A EA 00 00 0F 1A EA 00 (offset 108)

```

partikus, 2021-09-21:
I'm trying to debug the data using the following pages:

https://r12a.github.io/uniview/
https://r12a.github.io/app-conversion/

[image]

`0B 86 00 0F 1A F2 00 00 0F 1A F3 00` gives:

* `0B 86` - `2950`
* `00` - ` ` (space)
* `0F` - `15` 
* ...

MichaelDvP, 2021-09-22:
If you sort the telegram to 10 blocks of 20 bytes you get a set of messages:
```
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 3D 6A 00 00 0F 3D 6B 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 2F 87 00 00 0F 2F 88 00 
08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 2F 07 00 00 0F 2F 08 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 2F 06 00 00 0F 2F 07 00 
08 EA 00 0A 09 20 2D 2D 0B 86 00 0F 1A F2 00 00 0F 1A F3 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 1A EA 00 00 0F 1A EA 00 
08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 18 CB 00 00 0F 18 CC 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 18 CB 00 00 0F 18 CB 00 
08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 13 06 00 00 0F 13 06 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 13 05 00 00 0F 13 06 00
```
Offset 5,6,7 are the ascii error message, 8,9 the error-number. 
The first bytes `08 EA 00 0A 09` are all identical, 08 can be the device-source of the error. 
Offset 10-14 and 15-19 looks like a time code for start and end of error. But i can not decode it.
Maybe four bytes 10-13 as minutes of boiler uptime? Can you check against `ubauptime`
 
What information to the errors is shown on the thermostat service menu?

partikus, 2021-09-22:
**call boiler info**

```
{
  "heatingActive": "on",
  "tapwaterActive": "on",
  "selFlowTemp": 0,
  "selBurnPow": 0,
  "curBurnPow": 0,
  "pumpMod": 0,
  "pumpMod2": 0,
  "curFlowTemp": 46.2,
  "burnGas": "off",
  "flameCurr": 0,
  "heatPump": "off",
  "fanWork": "off",
  "ignWork": "off",
  "heatingActivated": "on",
  "heatingTemp": 35,
  "burnMinPeriod": 3,
  "burnMinPower": 1,
  "burnMaxPower": 0,
  "boilHystOn": -10,
  "boilHystOff": 6,
  "UBAuptime": 1006828,
  "burnStarts": 26700,
  "burnWorkMin": 139792,
  "heatWorkMin": 113189,
  "wWSelTemp": 0,
  "wWSetTemp": 40,
  "wWDisinfectionTemp": 70,
  "wWCircPump": "off",
  "wWCircPumpMode": "0x3min",
  "wWCirc": "off",
  "wWCurTemp": 38.5,
  "wWCurTemp2": 0,
  "wWActivated": "on",
  "wWOneTime": "off",
  "wWDisinfecting": "off",
  "wWCharging": "off",
  "wWRecharging": "off",
  "wWTempOK": "off",
  "wWHeat": "off",
  "wWStarts": 26603,
  "wWWorkM": 17348
}
```

```
ems-esp:/# read 88 14
006+02:19:43.905 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 0F 5C ED
```

Currently I have no any thermostat at all. [Tech Controllers](http://tech-controllers.com/p/eu-l-8e) manages the automation. Unfortunately, they do not support EMS to control the boiler. 

<details>
<pre>
006+02:22:42.237 N 11: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 C9 00 64 23 01 00 00
006+02:22:42.446 N 12: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 C9 00 00 00 02 01 DC 00 00 00 00 00 00 14 00 01 C9 00 00
006+02:22:42.634 N 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
006+02:22:42.924 N 14: [emsesp] Boiler(0x08) -> All(0x00), UBADHWStatus(0xE9), data: 28 01 75 00 00 00 00 01 75 46 00 00 00 00 00 67 EB 00 43 C4 00 00 00 00 01
006+02:22:45.523 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 C7 (offset 23)
006+02:22:52.277 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 C9 00 64 23 01 00 00
006+02:22:52.486 N 17: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 C9 00 00 00 02 01 DC 00 00 00 00 00 00 14 00 01 C9 00 00
006+02:22:52.675 N 18: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
006+02:22:52.919 N 19: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 CA (offset 23)
006+02:22:53.124 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBADHWStatus(0xE9), data: 28 01 76 00 00 00 00 01 76 46 00 00 00 00 00 67 EB 00 43 C4 00 00 00 00 01
006+02:22:59.168 N 21: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xBF), data: 00 00 (offset 8)
006+02:23:02.292 N 22: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00
006+02:23:02.478 N 23: [emsesp] Boiler(0x08) -> All(0x00), UBAOutdoorTemp(0xD1), data: 80 00
006+02:23:02.677 N 24: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 C9 00 64 23 01 00 00
006+02:23:02.906 N 25: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 C9 00 00 00 02 01 DD 00 00 00 00 00 00 14 00 01 C9 00 00
006+02:23:03.095 N 26: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
006+02:23:03.306 N 27: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 00 40 00 00 78 00 00 00 00 00 68 4C 02 22 10 00 00 00 01 BA 25 00 24 88 00 64
006+02:23:03.571 N 28: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 DE 30 (offset 29)
006+02:23:03.784 N 29: [emsesp] Boiler(0x08) -> All(0x00), UBADHWStatus(0xE9), data: 28 01 77 00 00 00 00 01 77 46 00 00 00 00 00 67 EB 00 43 C4 00 00 00 00 01
006+02:23:09.693 N 30: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 0F 5C F1
006+02:23:09.888 N 31: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParametersPlus(0xE6), data: 01 23 00 00 32 12 00 01 06 F6 03 01 05 64 01 00 00 1E 00 2D 01 01 01 00 01 00
006+02:23:10.139 N 32: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWWPlus(0xEA), data: 00 00 3C 08 00 01 28 00 00 00 00 00 46 00 00 01 00 00 00 00 00 5F 00 00 00 00 01
006+02:23:10.261 N 33: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWWPlus(0xEA), data: 01 (offset 26)
006+02:23:12.270 N 34: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 C9 00 64 23 01 00 00
006+02:23:12.478 N 35: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 C8 00 00 00 02 01 DB 00 00 00 00 00 00 14 00 01 C8 00 00
006+02:23:12.667 N 36: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
006+02:23:12.872 N 37: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 C9 (offset 23)
006+02:23:13.076 N 38: [emsesp] Boiler(0x08) -> All(0x00), UBADHWStatus(0xE9), data: 28 01 78 00 00 00 00 01 78 46 00 00 00 00 00 67 EB 00 43 C4 00 00 00 00 01
006+02:23:17.799 N 39: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 CA (offset 23)
006+02:23:22.274 N 40: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 C8 00 64 23 01 00 00
006+02:23:22.482 N 41: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 C8 00 00 00 02 01 DB 00 00 00 00 00 00 14 00 01 C8 00 00
006+02:23:22.670 N 42: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
006+02:23:22.900 N 43: [emsesp] Boiler(0x08) -> All(0x00), UBADHWStatus(0xE9), data: 28 01 79 00 00 00 00 01 79 46 00 00 00 00 00 67 EB 00 43 C4 00 00 00 00 01
006+02:23:23.317 N 44: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 C7 (offset 23)
</pre>
</details>

partikus, 2021-09-22:
I've wrote simple JS code for parsing it. It works like you expected:

```javascript
function toNumber(bytes, from, length) {
	// console.log('selected bytes', bytes.slice(from, from + length), 'from', bytes);
	return parseInt(`${bytes.slice(from, from + length).join('')}`, 16)
}
function parseError(totalUptime, message) {
	const bytes = message.split(' ');
 	const errorNo = toNumber(bytes, 8, 2);
  	const startedAtOffset = toNumber(bytes, 10, 4);
	const stoppedAtOffset = toNumber(bytes, 15, 4);
	const from = startedAtOffset * 60 * 1000;
	const to = stoppedAtOffset * 60 * 1000;
	const uptimeMinutes = totalUptime * 60 * 1000;
	const launched = new Date(Date.now() - uptimeMinutes);
	const startedAt = new Date(launched.getTime() + from);
	const stoppedAt = new Date(launched.getTime() + to);

	return { errorNo, startedAt, stoppedAt };
}

const errors = `08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 3D 6A 00 00 0F 3D 6B 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 2F 87 00 00 0F 2F 88 00 
08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 2F 07 00 00 0F 2F 08 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 2F 06 00 00 0F 2F 07 00 
08 EA 00 0A 09 20 2D 2D 0B 86 00 0F 1A F2 00 00 0F 1A F3 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 1A EA 00 00 0F 1A EA 00 
08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 18 CB 00 00 0F 18 CC 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 18 CB 00 00 0F 18 CB 00 
08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 13 06 00 00 0F 13 06 00 
08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 13 05 00 00 0F 13 06 00`.split("\n");

errors.map(msg => parseError(1006828, msg));
```

which gives:

```
"[
    {
        \"errorNo\": 2948,
        \"startedAt\": \"2021-09-16T22:26:15.793Z\",
        \"stoppedAt\": \"2021-09-16T22:27:15.793Z\"
    },
    {
        \"errorNo\": 2948,
        \"startedAt\": \"2021-09-14T11:11:15.794Z\",
        \"stoppedAt\": \"2021-09-14T11:12:15.794Z\"
    },
    {
        \"errorNo\": 2927,
        \"startedAt\": \"2021-09-14T09:03:15.794Z\",
        \"stoppedAt\": \"2021-09-14T09:04:15.794Z\"
    },
    {
        \"errorNo\": 2948,
        \"startedAt\": \"2021-09-14T09:02:15.794Z\",
        \"stoppedAt\": \"2021-09-14T09:03:15.794Z\"
    },
    {
        \"errorNo\": 2950,
        \"startedAt\": \"2021-09-10T19:22:15.795Z\",
        \"stoppedAt\": \"2021-09-10T19:23:15.795Z\"
    },
    {
        \"errorNo\": 2948,
        \"startedAt\": \"2021-09-10T19:14:15.795Z\",
        \"stoppedAt\": \"2021-09-10T19:14:15.795Z\"
    },
    {
        \"errorNo\": 2927,
        \"startedAt\": \"2021-09-10T10:11:15.796Z\",
        \"stoppedAt\": \"2021-09-10T10:12:15.796Z\"
    },
    {
        \"errorNo\": 2948,
        \"startedAt\": \"2021-09-10T10:11:15.796Z\",
        \"stoppedAt\": \"2021-09-10T10:11:15.796Z\"
    },
    {
        \"errorNo\": 2927,
        \"startedAt\": \"2021-09-09T09:34:15.796Z\",
        \"stoppedAt\": \"2021-09-09T09:34:15.796Z\"
    },
    {
        \"errorNo\": 2948,
        \"startedAt\": \"2021-09-09T09:33:15.796Z\",
        \"stoppedAt\": \"2021-09-09T09:34:15.796Z\"
    }
]"
```

MichaelDvP, 2021-09-22:
I think `const launched = new Date(Date.now() - uptimeMinutes);` gives wrong results. The boiler can be switched off in summer, not counting the uptime. If there is a thermostat with RTC, i think the telegram 10/11 is used and C2 only for systems without clock. 
I suggest to publish only the last error as ` --(2948) 8067 minutes ago`  or  ` --(2948) 5d14h27m ago`

partikus, 2021-09-22:
Oh, I forgot to mention the boiler is working a full year so the `launchedAt` is a valid value.

partikus, 2021-11-27:
@proddy & @MichaelDvP thank you for your time and help :+1: