emsesp/EMS-ESP32 #122: fastheatupfactor - detailled info missing in api V3.

tp1de, 2021-09-20:
ems-esp/api/thermostat/hc1/fastheatupfactor returns 
```
{
"message": "Problems parsing elements"
}
```

uom should be %. min = 0 / max = 100.

BTW: API V3 changed from unit into uom ..... I need to adjust my ioBroker adapter. Please do not change without need.

proddy, 2021-09-20:
unit is the wrong English word. And the `dev` branch is for changes. Any breaking changes will be documented between the official versions. If it bothers you stick to the main branch

proddy, 2021-09-20:
what does http://ems-esp/api/thermostat/hc1 come back with?

tp1de, 2021-09-20:
> what does http://ems-esp/api/thermostat/hc1 come back with?

```
{
  "seltemp": 22,
  "currtemp": 22.9,
  "mode": 1,
  "modetype": 1,
  "ecotemp": 19,
  "manualtemp": 16,
  "comforttemp": 22,
  "summertemp": 16,
  "designtemp": 42,
  "offsettemp": 0,
  "minflowtemp": 35,
  "maxflowtemp": 55,
  "roominfluence": 0,
  "nofrosttemp": -5,
  "targetflowtemp": 0,
  "heatingtype": 1,
  "summersetmode": 0,
  "summermode": 1,
  "controlmode": 1,
  "program": 1,
  "fastheatupfactor": 20
}
```

tp1de, 2021-09-20:
> unit is the wrong English word.

maybe ... but in ioBroker it is since years `unit` as well.

> And the `dev` branch is for changes. Any breaking changes will be documented between the official versions.

Just tell me how to track changes within API V3 so that I can adjust accordingly.
Can't you inform ahead within discord channel DEVELOPMENT/api?

proddy, 2021-09-20:
> Can't you inform ahead within discord channel DEVELOPMENT/api?

that's a good idea. I do agree it needs to be listed somewhere. It should be highlighted in the changelog but often we (I) forget

proddy, 2021-09-20:
Ok I found a number of errors in the API which I fixed in another branch, so will need to merge later. For example http://ems-esp/api/boiler/id would cause EMS-ESP to crash and restart. not good!

MichaelDvP, 2021-09-21:
If *"fastheatupfactor"* is a percentage value, it is not a factor. Factors are scalar values, percentage is a relation to a reference scaled to 100. I've renamed to `fastheatup` and added the percent.

Parsing does not work because `hc1/heatupfactor` was one byte to long for the parsing-buffer.

PS: @proddy Other value-names like `cylinderpumpmodulation`are also to long, i've enlarged the buffer.  But long names also blowing up mqtt and api response, especially if in a heatingcircuit.

MichaelDvP, 2021-09-21:
> For example http://ems-esp/api/boiler/id would cause EMS-ESP to crash and restart. not good!

Working for me, i think it crashes in the new `EMSdevice::device_entity_ha(`, which does not check for fullname == nullptr.

proddy, 2021-09-21:
yes it only crashes when HA is enabled. I've made the fixes, just in another branch so will merge later along with some React Hooks modifications

MichaelDvP, 2021-09-21:
> Nevertheless the km200 field-name is even with percentage fastHeatupFactor

I know, the name was taken from [here](https://github.com/emsesp/EMS-ESP32/issues/97#issuecomment-907656969), but it was not mentioned that it is a percent value. If Bosch names are wrong we don't need to copy them.

proddy, 2021-09-22:
fixed in latest dev. re-open if needed

closed 2021-09-22 by proddy