emsesp/EMS-ESP32 #123: Received message on illegal discovery topic

swa72, 2021-09-21:
**Question**
Recent HA log reports
```
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:101
Integration: MQTT (documentation, issues)
First occurred: 20:25:19 (5 occurrences)
Last logged: 20:25:19

Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwsettemp/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwmode/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwsettemplow/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwcircmode/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwextra1/config'
```

**Device information**
```
{
  "System": {
    "version": "3.1.0",
    "uptime": "002+04:22:15.388",
    "freemem": 175088
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 358860,
    "#read requests sent": 116325,
    "#write requests sent": 33,
    "#incomplete telegrams": 0,
    "#tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.04)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:11.09)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x21, ProductID:160, Version:24.03)",
      "handlers": "0x2D8"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x22, ProductID:160, Version:24.03)",
      "handlers": "0x2D9"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x28, ProductID:160, Version:24.03)",
      "handlers": "0x331"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x20, ProductID:160, Version:24.03)",
      "handlers": "0x2D7"
    },
    {
      "type": "Solar",
      "name": "SM200/MS200 (DeviceID:0x30, ProductID:164, Version:25.04)",
      "handlers": "0xF9 0x358 0x35A 0x362 0x363 0x366 0x364 0x36A 0x380 0x38E 0x391"
    },
    {
      "type": "Controller",
      "name": "9000i (DeviceID:0x09, ProductID:224, Version:35.03)",
      "handlers": ""
    }
  ]
}
```

proddy, 2021-09-21:
I haven't seen this, but will keep my eyes peeled.

MichaelDvP, 2021-09-22:
@swa72 v3.1.0 is very old, there are a lot of fixes since that, also the wrong `warm water` prefix in ha has changed to `ww`. Please update.

swa72, 2021-09-22:
Hi, my bad. Will do. Hopefully, this does not end in a big rename orgy of
entities in HA ;-)

swa72, 2021-09-23:
Well, updated to 3.2.1. Same warnings.
```
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:101
Integration: MQTT (documentation, issues)
First occurred: 5:09:41 (5 occurrences)
Last logged: 5:09:41

Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwsettemp/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwmode/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwsettemplow/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwcircmode/config'
Received message on illegal discovery topic 'homeassistant/sensor/ems-esp/thermostat_warm water_wwextra1/config'
```

MichaelDvP, 2021-09-23:
These are old mqtt entries with retain-flag. You have to remove them from mqtt-broker. 
Best way:
1 power off emsesp
2 remove all emsesp entries from mqtt-broker
3 power up emsesp again

proddy, 2021-09-23:
Michael is correct, these are the old names and need to be removed manually. I've been exploring how to get EMS-ESP to clean up old MQTT topics (I'll create a new issue for this)

swa72, 2021-09-23:
Ok, removed the messages using MQTT Explorer (HA and EMS-ESP were still running). Rebooted HA. All clear :-)