emsesp/EMS-ESP32 #125: replace DeviceValueUOM::LIST with a valid UOM

proddy, 2021-09-22:
'LIST' is not a unit of measure (like minutes, kW, % etc) so should be removed and replaced with the `DeviceValueUOM::NONE` as typically lists hold strings.

Then in the code, remove the checks and replace by checking the dv.type for `DeviceValueType::ENUM`

@MichaelDvP shall I do this or you since this was your addition I believe

MichaelDvP, 2021-09-22:
NUM, BOOLEAN, and LIST are no uoms. But as mentioned [here](https://github.com/emsesp/EMS-ESP32/blob/80fedf3fa3c95e4cfc002868a2cea26514486405/src/locale_EN.h#L186) this was your hack for HA. But now it is also used for [web-commands](https://github.com/emsesp/EMS-ESP32/blob/80fedf3fa3c95e4cfc002868a2cea26514486405/interface/src/project/ValueForm.tsx#L69) to render the dialog for bools and enums. Not easy to remove.

proddy, 2021-09-22:
Right, I'll take a stab at fixing it. Should be able to achieve the same using `dv.type`

MichaelDvP, 2021-09-22:
The LIST can easily removed, in emsdevice we can use `DeviceValueType::ENUM` and in ValueForm we can check for `devicevalue.l`.
BOOLEAN and NUM is additional  used for formatting the web-values, there we need some more checks.

proddy, 2021-09-22:
ok thanks for checking. I think its worth it. You know me, I'm a sucker for clean and readable code :-)

MichaelDvP, 2021-09-22:
OK, i'll remove UOMs: LIST, BOOLEAN and NUM. It's working in my test now.

proddy, 2021-09-22:
great! that was quick. I'm still trying to figure out React Hooks to get the i18n localization working

MichaelDvP, 2021-09-23:
Done. Please check the HA-icon for numbers working proper.
Booleans are handled now as on/off. The command input can not handle bool native and gives "true" as string back. My first attempt with `if (typeof devicevalue.v == "boolean")` does not work after changing the value. Also the new values is send to emsep as string. Now enum and bool commands handled the same way.

proddy, 2021-09-23:
@MichaelDvP it looks great, I couldn't find any issues with changing booleans or enums from the WebUI. Are there still issues you need help with?

MichaelDvP, 2021-09-23:
>  I couldn't find any issues

Good, then i close this.

closed 2021-09-23 by MichaelDvP