emsesp/EMS-ESP32 #129: Home Assistant - State class support

KNXBroker, 2021-09-24:
**Is your feature request related to a problem? Please describe.**
Since some home assistant releases home assistant supports state_class, which is not support (yet) by EMS-ESP.

**Describe the solution you'd like**
It would be great if automatic entity creation would support state_class definition. E.g. for temperatures `state_class: measurement`

**Describe alternatives you've considered**
Manual definition / creation of home assistant entities.

**Additional context**
I know that ESP8266 is no longer supported, but I am still using it... In general the feature would be great for both V2 and V3. Is the code definition part of both V2 and V3 the same...? I could support if someone shows me the respective lines. I assume it's only an additional parameter which has to be added to the respective sensors.

proddy, 2021-09-26:
@swa72 mentioned in #128 

> The stock configuration does not list sensor.solar_energy_today in HA's new energy dashboard. Adding
>
> ```
> sensor.solar_energy_today:
>   device_class: energy
>   state_class: total_increasing
> ```
>
> to customize.yaml did the trick for me.

proddy, 2021-09-28:
Added to 3.2.2b10

* WH uses `measurement` (E.g. solar/energyLastHour, solar/energyTotal)
* KWH uses `total_increasing` (E.g. solar/energyTotal, boiler/*nrg*)
* KW uses `measurement` (E.g. boiler/hpPower)
* W uses `measurement` (E.g. solar/solarPower)

everything else will not add the `state_class` attribute in the Home Assistant entity

closed 2021-09-28 by proddy