emsesp/EMS-ESP32 #136: Sense II / CW400 - Grab "Flow Setpoint Modifier" from thermostat and publish

glitter-ball, 2021-10-02:
My Sense II (identical to the CW400) maintains a value in the Diagnostics menu [unhelpfully] called "Room Influence". This shows how the realtime calculated flow setpoint is being altered by the error between current and set room temperatures, in weather comp systems with room influence. It's useful because it can help tweak the weather comp settings.

NB This is distinct from the "Room Influence" factor which is set in the weather comp heating curve menu for each heating circuit and remains static.

The thermostat alters the setpoint flow temperature up or down based on the error and the static "Room Influence" value.

My system has a static Room Influence factor of 9K, current setpoint is 18.5ºC and current room temp is 18.3ºC. This produces a dynamic "Room Influence" of 1.4K, currently.

Having both called "Room Influence", as they are in my Sense II is silly and very confusing. Perhaps this new value should be called "Flow Setpoint Modifier" or similar?

Very happy to do some snooping to find out which telegram this is in. Can anyone point me in the rough direction to start with please? It would be great to get it added. Thanks.

MichaelDvP, 2021-10-03:
> My system has a static Room Influence factor of 9K

It's not a factor, afaik it's a limiter, the roomtemperature difference will affect the flowtemp up to 9 K.
> Perhaps this new value should be called "Flow Setpoint Modifier" or similar?

What do you think about: rename the setting to `max. room influence` and the new measurement to `curr. room influence`? Then it's clear that these values are connected.
 > Can anyone point me in the rough direction to start with

`Log all` or use `watch 10` t o see the telegrams from thermostat, go to diagnostic menu and watch the room-influence value, note time of the change and the new value. Best to note 2 or more changes to be sure. If changes goes to slow, ask someone to manipulate the room thermostat like open the window or put a light bulb/candle near the thermostat.
With time of change, the new value and the log we can indentify the telegram and offset.

glitter-ball, 2021-10-03:
I think it's probably a multiplier, rather than a limit, otherwise it wouldn't do much at low errors. The manual says, "The difference beween set room temperature and measured room temperature leads to a parallel shift of the heating curve. The higher the set value for room influence, the bigger is the shift of the heating curve for a given temperature difference"

Anyway... I'm quite happy to try and find the data in a telegram. Can someone remind me how signed floating numbers are stored? I know some were two hex bytes divided by 10, but I can't remember how signed numbers were handled. Thanks.

MichaelDvP, 2021-10-03:
> I think it's probably a multiplier

A multiplier does not have a unit, with `Kelvin` it should be a absolute value. My RC35 manual calls the setting "maximaler Raumeinfluss (0K - 10K)" == maximum roominfluence.
> but I can't remember how signed numbers were handled

As 2'th complement, e.g. 0xFFFF = -1 (-0.1K), 0xFFFE = -2(-0.2K), You can post log and readings here, more eyes see more.

glitter-ball, 2021-10-03:
OK. I think I've found it. In telegrams starting "RC300Monitor(0x2A5), data: 06 07 ..." the next two bytes are the calculated influence in 2s complement notification.

With -4.5K showing, I see...
`000+09:08:03.286 N 3: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 07 FF D3 00 00 FF 00 52 00 B4 02 FF 01 02 (offset 25`) - FFD3 is -45 in 2s complement

`000+09:12:03.317 N 44: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 07 FF A6 00 00 FF 00 52 00 B4 02 FF 01 02 (offset 25`) - FFA6 is -90 in 2s complement and display eventually showed -9.0K

(Full dump below - FFAF is -81 and ties in with -8.1K on display...)

Yes, it does seem as though the value sets a limit. But it also affects the value calculated by a multiplier. For example, with room temp 0.5ºC above setpoint, the calculated value changes as follows...

Room influence setting, Calculated value

6K, -1.2K
7K, -1.4K
8K, -1.6K
9K, -1.8K

So the setting is acting both as a multiplier and a limit (with it set at 9K, I couldn't get the influence value to go above 9K).

```
ems-esp:$ watch on 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x10
000+00:00:41.130 E 0: [telegram] Rx: EF 4F 10 09 (CRC 09 != 19)
000+00:00:43.038 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+00:00:43.088 N 2: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
000+00:00:44.006 N 3: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 06 07 FF A6 00 00 FF 00 52 00 B4 02 FF 01 02 (offset 25)
000+00:00:44.204 N 4: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 0A 07
000+00:01:01.216 N 7: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 C2
000+00:01:02.498 N 8: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Monitor(0x2A5), data: 20
000+00:01:02.573 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 C2 21 25 00 00 25 1E 00 CA 03 03 01 00 CA 01 7F 00 00 11 01 03 07 98 00
000+00:01:02.824 N 10: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 25)
000+00:01:02.879 E 11: [telegram] Rx: 90 0B FF 19 17 A5 06 07 FF A6 00 00 FF 00 52 00 B4 02 FF 01 02 5A (CRC 5A != 6E)
000+00:01:03.223 N 12: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Set(0x2B9), data: 20
000+00:01:04.974 N 13: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Set(0x2B9), data: 20
000+00:01:05.774 N 15: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Summer(0x2AF), data: 20
000+00:01:05.815 N 16: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2AF), data: 09 28 00 00 41 2D 11 01 19 1E 00
000+00:01:06.123 N 17: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Curves(0x29B), data: 20
000+00:01:06.171 N 18: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Curves(0x29B), data: 01 01 00 FF FF 01 05 30 52
000+00:01:06.474 N 19: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300WWmode(0x2F5), data: 20
000+00:01:06.541 N 20: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
000+00:01:06.799 N 21: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300WWtemp(0x31B), data: 20
000+00:01:06.825 N 22: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 2D
000+00:01:07.173 N 23: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300OutdoorTemp(0x23A), data: 20
000+00:01:07.205 N 24: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300OutdoorTemp(0x23A), data: 00 8D
000+00:01:07.498 N 25: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Settings(0x240), data: 20
000+00:01:07.561 N 26: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Settings(0x240), data: 32 00 01 00 00 00 00 00 FF 02 FC 06 FF 00 00
000+00:01:11.375 N 27: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBADevices(0x07), data: 0F
000+00:01:11.399 N 28: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBADevices(0x07), data: 1B 01 00 00 00 00 00 00 00 00 00 00 00
000+00:01:11.764 N 29: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: FF AF (offset 27)
000+00:01:31.130 N 34: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 15 0A 13 03 08 39 06 01 10 FF 02)
000+00:01:36.171 N 38: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 C9 01 80 (offset 13)
000+00:01:42.357 N 41: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
000+00:01:42.411 N 43: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 3C (offset 3)
000+00:01:43.015 N 44: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+00:01:43.261 N 46: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 0A 07
000+00:02:02.401 N 50: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Monitor(0x2A5), data: 20
000+00:02:02.479 N 51: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 C2 21 25 00 00 25 1E 00 C9 03 03 01 00 C9 01 80 00 00 11 01 03 07 97 00
000+00:02:02.801 N 52: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Monitor(0x2A5), data: 20 (offset 25)
000+00:02:02.857 N 53: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 07 FF AF 00 00 FF 00 52 00 B4 02 FF 01 02 (offset 25)
000+00:02:03.176 N 54: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Set(0x2B9), data: 20
000+00:02:03.237 N 55: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 4E FF FF 00 1C 01 E1 20 01 0F 05 00 00 02 1F
[... cut ...]
000+00:03:06.229 N 102: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300Settings(0x240), data: 20
000+00:03:06.293 N 103: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Settings(0x240), data: 32 00 01 00 00 00 00 00 FF 02 FC 06 FF 00 00
000+00:03:32.160 N 104: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 15 0A 13 03 0A 3A 06 01 10 FF 02
000+00:03:35.471 N 106: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 C7 01 82 (offset 13)
000+00:03:39.019 N 112: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 C1
```

MichaelDvP, 2021-10-04:
Also found it in [Norberts list, page 32](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf), there it is called `Offset on room temperatur setpoint (room influence)`.
I'm not sure what the value really is. Is it added to flowtemp or is the flowtemp calculated from heatingcurve at modified roomtemp (measured + this offset). Norberts name points to the second.
How should we name this value? Shortname and fullname? Any suggestion? @proddy , @glitter-ball ?
And should we rename the roominfluence? That's a breaking change for HA-users.

proddy, 2021-10-04:
I don't think offset or influence are the correct words. Perhaps 'adjustment' or 'calibration' ?

glitter-ball, 2021-10-04:
This feature only applies to setups where the thermostat is set to do weather compensation with room influence.

The weather comp routine calculates a flow setpoint, based on the heating curve and outside temperature. The room influence then tweaks the flow setpoint - it's often described as a parallel shift in the heating curve.  If the room is too cold, the flow setpoint is increased and if it's too hot, the flow setpoint is decreased. Without this feature, it could take a long time for the room to reach the setpoint temperature because the thermostat will be trying to match heat loss from a space that is already too hot or too cold. With it, the set room temperature is reached more quickly and it also mitigates errors in the heating curve, to some extent. If the curve isn't quite right and, for example, the flow temp is too high to balance heat loss, the room temp will rise and room influence will trim the flow setpoint downwards, counteracting the heating curve error.

The value we're looking at is added to the weather comp's calculated flow temp, I think. It's not 100% straightforward. The room influence value does seem to set an upper limit as @MichaelDvP suggests from the RC35 manual. But, it also acts as a multiplier on the temperature error, as I showed above, when I changed the influence setting and the flow temp correction changed in realtime too.

'Calibration' probably isn't the right word - this correction is being done in realtime and is part of the normal operation. In fact, 'correction' is probably a good name. Room temp flow setpoint correction / adjustment?

I'd be tempted to leave the existing room influence point well alone - that's not a bad description and it avoids upsetting all the HA setups (mine included...) out there. Maybe we just add a new point?

MichaelDvP, 2021-10-04:
If you are sure that this value is direct affecting the flow temperature (after calculation from heatingcurve) and not as norbert describes the the room-setpoint before heatcurve calculation. 
Then use these (long) names: mqtt: `flowtempcorrection`, full name: `room temperature flow temperature correction`, or `room influenced flow temperature correction` to avoid double temp?

glitter-ball, 2021-10-09:
OK - I've done some measurements. Room Temp was 18.2ºC and I dialled room setpoint up to 20ºC. With my hands on the thermostat and the heating on, inevitably the room temp changed during the test, which made it a bit tricky. But interesting results...

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/user/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/user/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

A | B | C | D | E
-- | -- | -- | -- | --
RI setting (K) | Calculated RI (K) | Calculated Flow (ºC) | Flow Boost (ºC) : C-41 | Boost / Calc RI : D/B
0 | 0.0 | 41 | 0 |  
1 | 1.5 | 46 | 5 | 3.3
2 | 3.0 | 49 | 8 | 2.7
3 | 4.2 | 52 | 11 | 2.6
4 | 5.6 | 56 | 15 | 2.7
5 | 7.0 | 60 | 19 | 2.7
6 | 6.5 | 58 | 17 | 2.6
7 | 7.8 | 63 | 22 | 2.8
8 | 8.8 | 64 | 23 | 2.6
9 | 9.9 | 67 | 26 | 2.6
10 | 10.0 | 67 | 26 | 2.6

</body>

</html>

So, what do we learn from this?

- The boost in flow temperature appears to be directly related to the room influence setting. Here, most of the factors are around 2.6. I'm guessing this factor is governed by the room temperature error. The bigger the error, the more the flow is boosted. So, the flow temp is calculated by a formula like : Tcalc = Tbase + (n * RI setting * (setpoint - actual)) where Tbase is the basic temperature calculated by the weather compensation from the heating curve and n is a constant multiplier)
- The flow temp boost (in ºC) can exceed the setting (I'm seeing 26ºC boost here for 10K setting and 9.9ºC for 9K setting)
- The boost does tail off at higher values - calc flow temp doesn't change between RI setting of 9K and 10K

So hopefully answers @MichaelDvP's question above - the factor seems to be affecting the flow setpoint on top of the basic heating curve, which is implied by the manual saying that it introduces a parallel shift in the heating curve.

MichaelDvP, 2021-10-09:
If i get it right, A is the `roomInfluence` setting, B the new value, C is `targetflowtemp` from thermostat, D and E calculated from you.
It's interesting that the RI-value can be higher than setting, but it's not direct flow correction. 

What was the damped outside temperature? And the heating-curve parameters. 
I'm not sure what the value is, It can also be addition to room-setpoint:  
`flowtemp = heatingcurve(dampedoutsidetemp, roomsetpoint+B)`
This can be checked if you can check the flowtemp for roomsetpoint 20°C with A set, note the B-value and compare to flowtemp with A zero and roomsetpoint = 20 + noted-B. 

IMHO the roominfluence and the new value are associated and that should be expressed in the names. Also the mqtt-name should be short. I'm not happy with the suggestions now.

glitter-ball, 2021-10-09:
FAQ for Buderus Logamatic TC100 uses room influence like this...

> Flow temperature correction = (Setpoint temp - actual temp) * room influence factor

In the TC100, the RI factor can be set to 0, 1, 2 or 3.

No evidence that CW400 / Sense II works exactly like this but principle looks similar.

MichaelDvP, 2021-10-09:
Your heatingcurve gives a slope of 2.7, which is nearly the same as you calculate in col. E and fits to my formular.
But 9.5°C outdoor and 20°C setpoint should give 48°C flow without roominfluence.

I thought the TC100 is a single hc room controller? Does it also work outdoor-controlled? 
And in your data the setting A is not a factor.

glitter-ball, 2021-10-09:
Great. Happy to go with any suitable names and to get this working.

glitter-ball, 2021-10-10:
Thanks. Just loaded the dev binary. Current Room Influence is showing 9K in EMS-ESP & HA but 0.9K on the thermostat. Looks like we might be missing division by 10...

glitter-ball, 2021-10-10:
Thanks. Loaded the latest build and looks good. *8)

closed 2021-10-15 by glitter-ball