emsesp/EMS-ESP32 #138: Worcestor Comfort II RF Support

stewiem2000, 2021-10-04:
Hi,
I have connected my EMS-ESP (an awesome EMS Bus Gateway Premium II (P2) from @bbqkees!) to my Worcester Bosch Greenstar 36CDi Compact Erp combi boiler with Comfort II RF wireless controller and "CRF200S" remote.  I'm running the latest dev firmware, "2.2.2b0" via telnet, "2.2.2-dev(38a443e)" from the web-interface.

The boiler seems to be detected OK (I think, certainly getting sensible values out of it), but the thermostat isn't recognised and I'm not entirely sure about the controller, _show devices_ yields
```
(1) Boiler: Worcester GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.02)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

(2) Controller: Controller (DeviceID:0x09, ProductID:152, Version:03.01)

(2) Unknown: Worcester unknown (DeviceID:0x18 ProductID:216, Version:01.02)
```

_show ems_ bus information:
```
EMS Bus info:
  Tx mode: 1
  Bus protocol: HT3
  #telegrams received: 583
  #read requests sent: 132
  #write requests sent: 0
  #incomplete telegrams: 19
  #tx fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 100%
```

From the following log, I'm slightly concerned by the CRC-mismatches, but I can see that dev:0x18 is the thermostat because adjusting its target temperature emits 0x2AF IDs where the third octet (value 0x20) seems to be the call for heat (value 0x21 when wanted, 0x20 when not) and the the last octet (value 0x25) seems to be twice the selected temperature, ie. 0x25 = 37 = 18.0ºC:
```
000+00:14:57.219 E 58: [telegram] Rx: 0C 2A 00 00 00 00 00 1B 00 00 00 FA 00 00 80 00 00 80 00 80 00 80 00 00 6B (CRC 6B != 62)
000+00:15:00.450 E 59: [telegram] Rx: 88 0B 19 00 80 00 00 DB 80 00 00 00 00 00 00 BC F7 04 3C 8E 00 00 00 04 7E 87 00 55 08 80 00 76 (CRC 76 != B8)
000+00:15:22.821 N 60: [emsesp] Unknown(0x18) -> All(0x00), ?(0x06), data: 15 04 13 10 15 00 04 00
000+00:15:23.442 E 61: [telegram] Rx: 80 E0 FF 09 (CRC 09 != 14)
000+00:15:23.698 N 62: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C1 20 00 00 00 25
000+00:16:03.751 N 63: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C0 20 00 00 00 25
000+00:16:22.717 E 64: [telegram] Rx: 98 00 06 00 15 04 13 10 16 00 04 (CRC 04 != D7)
000+00:16:23.197 N 65: [emsesp] Unknown(0x18) -> All(0x00), ?(0x06), data: 15 04 13 10 16 00 04 00
000+00:16:23.691 E 66: [telegram] Rx: 98 00 FF 00 01 A5 00 C0 20 00 00 00 20 09 (CRC 09 != 1D)
000+00:16:23.899 N 67: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C0 20 00 00 00 25
000+00:17:22.796 N 68: [emsesp] Unknown(0x18) -> All(0x00), ?(0x06), data: 15 04 13 10 17 00 04 00
000+00:17:23.059 E 69: [telegram] Rx: 98 00 FF 00 01 A5 00 C0 20 00 00 00 (CRC 00 != 49)
000+00:17:23.499 N 70: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C0 20 00 00 00 25
```

I've yet to try sending the boiler commands.

What can I do to help resolve the CRC errors and get my thermostat supported? :)

Thanks in advance!

bed31, 2021-10-30:
Hi, I also have a CRF200S thermostat called RF-NSC on ELM Leblanc branding (https://www.elmleblanc.fr/fr/fr/ocs/elm-leblanc/nsc-rf-878577-p/).
Controller (HT3) and Thermostat (CRF200S) are not recognized even with the last version of EMS-ESP32 3.2.1.
[image]

In the post it is written `We're able to read the mode, current temperature and setpoint temperature but writing is tricky.` but i nothing is decoded :(
Is it in a dev version or is there another trick ?
[log.txt](https://github.com/emsesp/EMS-ESP32/files/7446988/log.txt)

What can I proceed to at least control On/Off of heating part of the Thermostat  if I can control the thermostat ? 
Otherwise how could I replace the thermostat to control boiler ?

Thanks for your help.

proddy, 2021-10-30:
You could try the latest dev build. No promises though as technically we
haven’t supported the crf200.
https://emsesp.github.io/docs/#/Supported-EMS-Devices

bed31, 2021-10-30:
> You could try the latest dev build. No promises though as technically we haven’t supported the crf200. https://emsesp.github.io/docs/#/Supported-EMS-Devices
> […](#)

Thanks for your quick response.
Do you think it will be supported soon ?

regards

bed31, 2021-10-30:
@proddy 
I have test the last version v3.2.2b14 and no change.
Concerning Controller (HT3) is normal to have nothing also ?

I also notice some errors in the console 
`2021-10-30 18:18:39.856   ERROR 584:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 8B 98 FF 00 20 01 A8`
Is it an real issue ?

Thx

proddy, 2021-10-30:
@bed31 you should be at least able to see the details. Without details we can't help you. Have a look at https://emsesp.github.io/docs/#/Troubleshooting

MichaelDvP, 2021-10-30:
This is really a poor thermostat, not responding to hc2,3,4 telegrams. The error messages should now be fixed (only a few after startup), but it is cosmetic.

bed31, 2021-10-31:
> @bed31 you should be at least able to see the details. Without details we can't help you. Have a look at https://emsesp.github.io/docs/#/Troubleshooting

@proddy Sorry I may did not understand your comment :(
Thermostat id recognized but only Date
[image]

Controller is not recognized : 
[image]

Boiler : Ok

MichaelDvP, 2021-10-31:
> Controller is not recognized :

looks like the controller is recognized. It has no data, but controllers normally do not publish any data. This is ok.
>  Thermostat is recognized but only the Date (not very useful )

In your `log.txt` from yesterday (and this is all information from your system we have) this is correct. Your thermostat does not publish more values, we have the time publish
```
2021-10-30 17:26:34.603 TRACE 2734: [emsesp] Thermostat(0x18) -> All(0x00), RCTime(0x06), data: 15 0A 11 1E 1B 00 05 00
```
thermostat reading some boiler parameters:
```
2021-10-30 17:26:34.800 TRACE 2735: [emsesp] Thermostat(0x18) <- Boiler(0x08), Version(0x02), data: 0A
2021-10-30 17:26:34.836 TRACE 2736: [emsesp] Boiler(0x08) -> Thermostat(0x18), Version(0x02), data: 5F 14 0E 00 00 00 00 00 00 00
2021-10-30 17:26:34.855 TRACE 2737: [emsesp] Thermostat(0x18) <- Boiler(0x08), ?(0x04), data: 15
2021-10-30 17:26:34.907 TRACE 2738: [emsesp] Boiler(0x08) -> Thermostat(0x18), ?(0x04), data: 2D 58 00 40 18 1F 64 00 00 78 6C 25 00 00 00 08
2021-10-30 17:26:34.925 TRACE 2739: [emsesp] Thermostat(0x18) <- Boiler(0x08), UBAParameters(0x16), data: 1B
2021-10-30 17:26:34.964 TRACE 2740: [emsesp] Boiler(0x08) -> Thermostat(0x18), UBAParameters(0x16), data: FF 42 2D 00 00 FB 03 01 03 64 0A 04
2021-10-30 17:26:34.981 TRACE 2741: [emsesp] Thermostat(0x18) <- Boiler(0x08), UBAParameterWW(0x33), data: 17
2021-10-30 17:26:35.028 TRACE 2742: [emsesp] Boiler(0x08) -> Thermostat(0x18), UBAParameterWW(0x33), data: 08 FF 39 00 00 1C 00 02 48 00 FF FF
```
thermostat sets flow temperature setpoint:
```
2021-10-30 17:26:35.048 TRACE 2743: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00
```
thermostat sets boiler warm water temperature:
```
2021-10-30 17:27:35.051 TRACE 2830: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBAFlags(0x35), data: 3C (offset 3)
```

But the thermostat does not publish it's own values like the CRF200S from @stewiem2000. This was this telegram: `Thermostat(0x18) -> All(0x00), RC300Monitor(0x2A5), data: 00 B9 20 00 00 00 20`. Maybe it's older, i don't know, you have not told us the system information with version info.

bed31, 2021-10-31:
@MichaelDvP Thank you for all these information.

> Maybe it's older, i don't know, you have not told us the system information with version info.
Sorry but what are the information that I could give you ?
The boiler has just been install 3 months ago. It's a ELM Leblanc ODEALIS CONDENS GVT C 24-2M (same as Junkers Cerapur).
The Thermostat is a NSC RF (=CRF200S) 
[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/7449394/log.1.txt)

[image]

[image]

MichaelDvP, 2021-10-31:
Please post system info and settings, see buttons on the help page:
[image]

bed31, 2021-10-31:
> Please post system info and settings, see buttons on the help page: ![Screenshot 2021-10-31 at 19-02-55 EMS-ESP]

[emsesp_system_info.txt](https://github.com/emsesp/EMS-ESP32/files/7449443/emsesp_system_info.txt)
[emsesp_system_settings.txt](https://github.com/emsesp/EMS-ESP32/files/7449444/emsesp_system_settings.txt)