emsesp/EMS-ESP32 #140: Use wwonetime...

glitter-ball, 2021-10-04:
**Question**
Struggling to get wwonetime working on my setup. Being a cautious sort (i.e. a coward!) most of my EMS-ESP use is read-only - watching stuff happen. But there are a couple of things I'd like to be able to do controlwise. One of them is kick off a one-time hot water re-heat. I thought I'd start simple and try to trigger this from the telnet interface. When I issue "call boiler wwonetime 1", nothing appears to happen. Am I doing something silly?

**Screenshots**
```
ems-esp:/# watch on
Watching incoming telegrams, displayed in decoded format
002+20:52:00.009 N 2: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 27 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:00.314 N 3: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 02 09 00 8B (offset 13)
002+20:52:00.530 N 4: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:00.752 N 5: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 01 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:04.688 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 12 D9 2A
002+20:52:04.989 N 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 4B 64 00 04 F6 0A 01 03 64 50 04 00 00 00 00 00 00 00 00 00 00 00 00 28 00 3C
002+20:52:05.213 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 3C 00 (offset 25)
002+20:52:05.513 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 88 80 00 80 00 00 00 00 00 00 34 12 02 6A A0 00 00 00 01 D2 71 00 25 A7 80 00
002+20:52:05.712 N 10: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 80 00 (offset 25)
002+20:52:06.046 N 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 3C FB 00 28 00 00 46 00 FF FF 00
002+20:52:06.344 N 12: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B8 21 00 00 00 00 25 02 09 01 01 03 02 09 00 8B 00 00 11 01 00 07 35 00
002+20:52:06.602 N 13: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
002+20:52:06.935 N 14: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 1E 02 2A FF FF 00 25 01 E1 20 01 0F 05 00 00 02 21
002+20:52:07.058 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 25 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:07.412 N 16: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x2F5), data: 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00
002+20:52:07.700 N 17: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 28
ems-esp:/# call boiler wwonetime 1
002+20:52:09.386 N 18: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 01 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:10.009 N 19: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 25 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:10.333 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 25 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:10.610 N 21: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 88 80 00 80 00 00 00 00 00 00 34 12 02 6A A0 00 00 00 01 D2 71 00 25 A7 80 00
002+20:52:10.858 N 22: [emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 93 09 05 1D 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
002+20:52:11.129 N 23: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:11.377 N 24: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 03 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:11.571 N 25: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
002+20:52:18.522 N 26: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBADevices(0x07), data: 0F
002+20:52:18.546 N 27: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00
002+20:52:20.059 N 28: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 23 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:20.279 N 29: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:20.527 N 30: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 03 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
ems-esp:/# call boiler wwonetime 0
002+20:52:28.580 N 31: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 03 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:30.008 N 32: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
002+20:52:30.058 N 33: [emsesp] Thermostat(0x10) -> All(0x00), ?(0x267), data: 00 00
002+20:52:30.294 N 34: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 B8 21 00 00 00 00 25 02 09 01 01 03 02 09 00 8B 00 00 11 01 00 07 35 00
002+20:52:30.502 N 35: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 28 01 A8 01 A8 01 00 00 03 00 00 98 2F 00 0E 6B 00 80 00
002+20:52:30.708 N 36: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 02 20 00 00 00 02 40 C0 80 00 01 A8 80 00 00 00 FF 00 00 00 CB 00 00 00
002+20:52:31.030 N 37: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 1B 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00
002+20:52:31.328 N 38: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 05 06 FF A6 00 00 FF 00 4B 00 C6 04 FF 01 02 (offset 25)
002+20:52:31.527 N 39: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 09 07
002+20:52:31.878 N 40: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 01
002+20:52:31.924 N 41: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 28 (offset 3)
002+20:52:31.969 N 42: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 09 0A 1E 22 20 02 01 10 FF FE
ems-esp:/# watch off
```

**Device information**
```
ems-esp:/# show
EMS-ESP version 2.1.0b12

Boiler: Worcester Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.01)
  Heating active: off
  Warm water/DHW active: off
  Service code:
  Service code number: 203
  Warm water selected temperature: 60 °C
  Warm water set temperature: 40 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 5 °C
  Burner selected max power: 0 %
  Burner current power: 0 %
  Pump modulation: 0 %
[... 55 lines cut ...]
    Summer temperature: 17 °C
    Summer mode: auto °C
    Mode: auto
    Mode type: eco

Dallas temperature sensors:
  Sensor 1, ID: 28-BE2A-7791-0402, Temperature: 39.1 °C

ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Worcester Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.01)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x2A5 0x2B9 0x2AF 0x2A6 0x2BA 0x2B0 0x2A7 0x2BB 0x2B1 0x2A8 0x2BC 0x2B2 0x2F5 0x31B 0x31D 0x31E

Controller: ErP (DeviceID:0x09, ProductID:209, Version:01.03)

ems-esp:/#

```

**Additional context**
*Add any other context about the problem here.*

glitter-ball, 2021-11-19:
I still need to resolve this. I think @tp1de has a CW400 and km200 gateway, so I was going to ask him nicely to set wwonetime on the km200 and capture the telegrams between the km200 and CW400, to see how it does it.

tp1de, 2021-11-20:
@glitter-ball let me share my findings. First of all I have a Buderus KB192i with RC310 and MM100 Mixer. IP-inside (KM200).

1. wwonetime on boiler does not work since I have an EMS+ System. wwonetime telegram type is 0x35 which does not work. Any ON value is reset to OFF.
2. Most WW parameters are set by thermostat (RC310).
3. Setting one-time charge by KM200 and/or MyBuderus  App uses telegram type 0x01F5 wich is EMS+.
4. ON/OFF: Offset 0B  value FF/00 - duration offset 0A value in 15 Minutes eg. 02 = 30 mins / 04 = 60 mins
4. The settings in ems-esp are done within device thermostat by parameters: wwcharge (ON/OFF), wwchargeduration (how long in minutes) and the boiler parameter wwseltempsingle sets the onetime charge temp. These use the same telegram types as km200 and these fields are working similar to km200 on my system.

**But keep in mind. Onetime Charge only works outside of an active time-program. Switching on one-time charge during an active time program has no effect in ww temperature (I had to learn by myself !)**

glitter-ball, 2021-11-20:
@tp1de - thanks!

Yes, I don't understand wwonetime logic with the time program! I heat to 60°C twice a day and have a reduced setting of 45°C during the day to stop it getting too cold for hand-washing etc. Usually wwonetime from the thermostat will heat to 60°C if programme is set to 45°C but sometimes it won't!

My thermostat (CW400 / Sense II) is also EMS+, I think. I'll see if I can work out how to send it and see what happens.

mvjt, 2022-02-06:
I would also need this. It would give the opportunity to heat water when electricity is cheap during the night right?

mvjt, 2022-08-06:
https://domoticproject.com/ems-bus-buderus-nefit-boiler/#0x35_8211Flags_Hot_Water

Is this command used?

proddy, 2022-08-07:
> https://domoticproject.com/ems-bus-buderus-nefit-boiler/#0x35_8211Flags_Hot_Water

yes except we're not using values 39 & 7 as Dani suggested by 2 and 3. See https://github.com/emsesp/EMS-ESP/issues/341#issuecomment-596245458

You can experiment yourself by writing directly to this 0x35 telegram from the telnet Console to see what happens. Keep in mind it needs to be outside an active program

mvjt, 2022-08-07:
> > https://domoticproject.com/ems-bus-buderus-nefit-boiler/#0x35_8211Flags_Hot_Water
> 
> yes except we're not using values 39 & 7 as Dani suggested by 2 and 3. See [emsesp/EMS-ESP#341 (comment)](https://github.com/emsesp/EMS-ESP/issues/341#issuecomment-596245458)
> 
> You can experiment yourself by writing directly to this 0x35 telegram from the telnet Console to see what happens. Keep in mind it needs to be outside an active program

Thanks @proddy. I think I am/was confused how the EMS manages all the different boilers. It seems the comment above from @tp1de  is accurate. It should be sent to thermostat and not boiler. How does EMS know whether to send or RC300WWmode(0x02F5) or RC300WWmode(0x01F5)?

proddy, 2022-08-07:
> And issuing it does not seems to work on latest dev
> 
> ems-esp:# call thermostat wwonetime on

It's a *boiler* command. Look at http://ems-esp.local/api/boiler/commands

mvjt, 2022-08-07:
> > And issuing it does not seems to work on latest dev
> > ems-esp:# call thermostat wwonetime on
> 
> It's a _boiler_ command. Look at http://ems-esp.local/api/boiler/commands

I´m even more confused :-) 

- I've checked the docs and wwonetime seems to be both an boiler and a thermostat command. https://emsesp.github.io/docs/#/Command?id=thermostats-thermostat
- The docs even state vvonetime on boiler might get overridden by thermostat.
- The [dest] in the working examples above is the Thermostat (0x10)
- The [dest] in my setup is also the Thermostat, not the boiler.

I was under the impression both were possible...

MichaelDvP, 2022-08-07:
It's a boiler ommand and a RC100/300 only thermostat command. The RC300 blocks/overwrites the boiler command and uses a different logic to prepare onetime (wwextra). 
Both are listed in the docu. You can also check in telenet, api or web-settings-customisaton what commands are registerd for your system. The docu sometimes is a bit behind, if you find a missing info, don't complain here, press the `improve this article` button in the docu and add the missing infos.
> I was under the impression both were possible...

No, with RC300 the boiler command is not working, you have to use the thermostat command.

mvjt, 2022-08-07:
> No, with RC300 the boiler command is not working, you have to use the thermostat command.

So how do I get access to the Thermostat command for my HP which seems to be using "RC300" compatibility as it sends it to the thermostat? If I understand your response it is not available in my system because it is not detected properly?

MichaelDvP, 2022-08-07:
Yes, you are right, i'd remembered wrong. It was renamed and is called `wwCharge`, can't remember why.

onetimekey is a a setting for RC35 if the pushbotton on the thermostat activates onetime or its without function.

mvjt, 2022-08-07:
> Yes, you are right, i'd remembered wrong. It was renamed and is called `wwCharge`, can't remember why.
> 
> onetimekey is a a setting for RC35 if the pushbotton on the thermostat activates onetime or its without function.

Thank you for clarifying!

mvjt, 2022-08-17:
Hi @MichaelDvP ,

I noticed that with Thermostat->wwcharge the FW sends 0x01F5 command. But my HP uses 0x02F5. What is the difference and how can I get the FW to use 0x02F5 instead?

MichaelDvP, 2022-08-17:
It's a bit confusing, emsesp uses an internal offset of 0x100 to indicate ems+ addressing. the emsesp telegram 0x2F5 is raw 0x1F5.
wwCharge writes to right address:
https://github.com/emsesp/EMS-ESP32/blob/8d3bcde8d6723d1ab75dfb4504ccef38c0ffa0bb/src/devices/thermostat.cpp#L1811-L1821

mvjt, 2022-08-17:
So even this raw log means 02 F5?

```
000+00:03:12.980 I 25: [command] [readonly] Calling command 'thermostat/wwcharge' (charge) with value on
000+00:03:12.999 I 26: [telegram] [readonly] Sending write Tx telegram: 0B 10 FF 0B **01 F5** FF
```

MichaelDvP, 2022-08-17:
Yes emsesp telegram id 02F5 is send as raw 01F5.  
https://github.com/emsesp/EMS-ESP32/blob/18e9b994136268fc8af8879101ff2357ff5b90d8/src/telegram.cpp#L326-L336 
BTW: [Norberts1' docu](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf) uses a identical scheme for his message-ids, but note them decimal.

closed 2023-02-19 by proddy