emsesp/EMS-ESP32 #144: Smart control a heating system with HA?

swa72, 2021-10-04:
Hi,

now that I have EMS-ESP and HA running smoothly together, I can tune my heating circuits (adjusting the heating curve, playing around with the radiators and stuff). 

One major problem in my house are many windows, sunny days and overshooting temperatures. https://github.com/proddy/EMS-ESP/issues/524 tells me not to mess with the boiler directly as any program running on the thermostat (RC300) may override this. So basically, I was thinking along the following lines:

If weather forecast equals sunny for the next day, reduce offset temperature by X early next morning (say Y minutes before sunrise) and reset offset temperature in the evening (say, Z minutes before sunset). Given my Aqara temperature sensors in the room, I feedback that info into the system and "learn" X, Y and Z and adjust offset temperatures.

Normally, its a job for a TRV to close the valve. However, I observe the following: given that the boiler only knows outside temperature, it will fire even if the TRVs closes their valves. The consequence, boiler starts and stops. Repeat.

I'm sure that tado and others are using similar ideas. Anyone knows of something similar for HA or any other system?

glitter-ball, 2021-10-15:
My system kind of gets round that by having weather comp with room influence. If the sun shines in the room and warms it, the room temp exceeds the setpoint and the boiler flow temperature is reduced by the thermostat to prevent overheating.

Can you add room influence to your thermostat?

mvjt, 2022-08-08:
> My system kind of gets round that by having weather comp with room influence. If the sun shines in the room and warms it, the room temp exceeds the setpoint and the boiler flow temperature is reduced by the thermostat to prevent overheating.
> 
> Can you add room influence to your thermostat?

My experience is that it is too late when the room thermostat detects it. I have a light sensor in my room sensor. I've installed it where the first morning light hits the inside of the house. It helped. But for best performance weather forecasts must be used so you can pull the brakes even earlier  :-) Just my experience...

swa72, 2022-09-06:
Yup, I agree. That is why I integrated a solar forecast into my heating
control.

closed 2023-02-19 by proddy