emsesp/EMS-ESP32 #149: read maintenance messages (e.g. H6) which are not service/error codes

deejaybeam, 2021-10-04:
**Bug description**
Active Error-Message shown on Thermostat RC30 (H6) is not detected with v2.1, but was detected with v1.9.5.

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
If an error is displayed on RC30 this should also be detected with ems-esp to be able to send a notification.

**Screenshots**
*If applicable, add screenshots to help explain your problem.*
[image]

[image]

[image]

**Device information**
{
  "System": {
    "version": "2.1.0",
    "uptime": "000+04:45:31.631",
    "freemem": 40,
    "fragmem": 14
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 1,
    "mqtt_qos": 0,
    "mqtt_retain": "on",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 20115,
    "#read requests sent": 3739,
    "#write requests sent": 4,
    "#incomplete telegrams": 1,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 13,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GB125/MC10 (DeviceID:0x08, ProductID:72, Version:02.04)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC30 (DeviceID:0x10, ProductID:67, Version:02.08)",
      "handlers": "0xA3 0x06 0xA2 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37"
    },
    {
      "type": "Mixer",
      "name": "MM10 (DeviceID:0x21, ProductID:69, Version:02.00)",
      "handlers": "0xAA 0xAB 0xAC"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.00)",
      "handlers": ""
    }
  ]
}

**Additional context**
*Add any other context about the problem here.*

proddy, 2021-11-10:
I believe this has already been implemented. Otherwise we can re-open it.

closed 2021-11-10 by proddy

MichaelDvP, 2023-08-01:
Just to have it documented, here's another function in the 0x05 telegram from controller: 
Setting chimney sweeper mode:
`controller(0x09) -W-> boiler(0x08), ?(0x05), data: 64 (offset 4)`
resetting chimney sweeper mode:
`controller(0x09) -W-> boiler(0x08), ?(0x05), data: 00 (offset 4)`
I don't think we need a command for that in ems-esp.