emsesp/EMS-ESP32 #151: thermostat emulation

mglatz, 2021-10-04:
How difficult would it be to have EMS-ESP to emulate real thermostat?
Why: I dont have any thermostat at all. I have a smart home where are radiator valves are electronic and controlled independently based on current temperature in each room. So I have one circuit and boiler will only heat the heating water if at least one of the vales is open. But it seems that without a thermostat, very few "things" can be directly controlled on the boiler (jukers cerapur).
So... could technically EMS-ESP pretend to be a thermostat to make the boiler more controllable?

closed 2023-02-19 by proddy

mcarbonneaux, 2023-10-15:
i have ems compatible gas heater (elm leblanc), and i have zigbee TRV on each of my radiator, each have temperatur sensor, plus i have also external zigbee temperature sensor, and i want manage all this globaly... all are connecter to home assitant...

actualy i have thermostat using lr/ls (on/off) of my heater.... by i whant to control on/off by my home assistant... ideally the warming power ...

is possible to control the warming of my elm leblanc heater with ems-esp gateway ? idaly by setting the boiler temperature (i've seen @Oderik speaking about flowtemp) ?

MichaelDvP, 2023-10-16:
@mcarbonneaux 
>  is possible to control the warming of my elm leblanc heater with ems-esp gateway ? idaly by setting the boiler temperature (i've seen @Oderik speaking about flowtemp) ?

Yes, you can send the desired boiler temperature (selflowtemp). The boiler accepts only temperatures that are lower than the selected temperature on the controlpanel and the remote command has to be repeated every minute. After ~2 minutes without command the boiler resets to the setting on the controlpanel. 
We have added a `forceheatnigoff` command, that sets the boiler temperature to off and repeats this every minute, so a simple on/off control can be done without any scheduling/repeating. 
If you want to set variable temperatures you have to do the repeating by your own logic.

mcarbonneaux, 2023-10-16:
the posibility to set boiler temp permit to make modulated pid regulation. it's cool, it's more efficace than on/off!

> The boiler accepts only temperatures that are lower than the selected temperature on the controlpanel
ok it's logic (the Elm leblanc CR100 modulation thermostat, modulates down the power of the boilers when they are close to the set point!)

> After ~2 minutes without command the boiler resets to the setting on the controlpanel.
hummm, ok i need the schedule the value each minutes (<2 minutes) on the home assistant...

> We have added a forceheatingoff command, that sets the boiler temperature to off and repeats this every minute, so a simple on/off control can be done without any scheduling/repeating.
ok, with that a can reproduct actual way of doing of my thermostat by setting this off/on.

FireFury0, 2025-04-30:
Sorry to revive an old topic.  I too am looking at using Home Assistant to emulate a room thermostat (replacing a dead Tado system).  The boiler is a Worcester Greenstar.

The old Tado controller is wired as a simple relay across the boiler's Lr/Ls terminals, and from what I can see these terminals need to be bridged, which puts the boiler into heating mode, which can then be controlled by a thermostat (or in my case, the EMS-ESP) over the EMS bus.

This means that the boiler's "default" mode will be heating, and it will only stop heating if told to do so over the EMS bus.  If the bus breaks, it would be preferable for the boiler to not default to non-stop heating.  The service manual for the boiler says that it will stop with error A8 if there is a break in communication with a "Sense" thermostat, but having read both the boiler's service manual and the service manual for Worcester's "Sense" thermostats, none of the instructions seem to indicate any kind of configuration on the boiler itself to tell it to expect to see a thermostat (and therefore stop with error A8 if it can't talk to one).

Am I missing a configuration step to put the boiler into a "thermostat on the EMS bus" mode?