emsesp/EMS-ESP32 #152: HA statistics - share ideas

S-Przybylski, 2021-10-04:
Dear all
i am interested in statistics about my heating. I just started to learn a little bit about the historical statisics in HA (and yes i am aware that functions like max, min, mean, integrate and differentiate could easily be done with Grafana - i also use it). But i am also interested in simple functions within HA.

So my first question was, how long does the heating heats the warm water and heating circuit itself?
Please note: The yesterday value will be calculated

````
sensors.yaml:
##########################
#Heating Test historical statistics
##########################
- platform: history_stats
  name: heating_warmwater_hst_runtime_per_day
  entity_id: binary_sensor.heating_warmwater
  state: 'on'
  type: time
  end: '{{ now().replace(hour=0).replace(minute=0).replace(second=0) }}'
  duration:
    hours: 24

- platform: history_stats
  name: heating_warmwater_hst_ratio_per_day
  entity_id: binary_sensor.heating_warmwater
  state: 'on'
  type: ratio
  end: '{{ now().replace(hour=0).replace(minute=0).replace(second=0) }}'
  duration:
    hours: 24

- platform: history_stats
  name: heating_warmwater_hst_ratio_between_5_and_23
  entity_id: binary_sensor.heating_warmwater
  state: 'on'
  type: ratio
# between 5 -23h
  end: '{{ ((as_timestamp(now().replace(hour=0).replace(minute=0).replace(second=0)) | int) - 3600)  }}'
  duration:
    hours: 18

- platform: history_stats
  name: heating_heating_hst_runtime_per_day
  entity_id: binary_sensor.heating_heating
  state: 'on'
  type: time
  end: '{{ now().replace(hour=0).replace(minute=0).replace(second=0) }}'
  duration:
    hours: 24

- platform: history_stats
  name: heating_heating_hst_ratio_per_day
  entity_id: binary_sensor.heating_heating
  state: 'on'
  type: ratio
  end: '{{ now().replace(hour=0).replace(minute=0).replace(second=0) }}'
  duration:
    hours: 24
````

and the template sensors to identify if the heating works in heating mode or warm water mode:
````
binary_sensors.yaml:
##########################
# for Heating Test historical statistics
##########################
- platform: template
  sensors:
    heating_warmwater:
      friendly_name: "Heating Warmwater"
      value_template: >-
       {% if is_state("binary_sensor.heating", "on") and is_state("sensor.Warm_Water_3_way_valve", "on") and is_state("sensor.boiler_pump", "on") %} {{ true }}
       {% else %} {{ false }}
       {% endif %}
# used to get frequent updates!!!
      entity_id:
        - binary_sensor.heating
        - sensor.Warm_Water_3_way_valve
        - sensor.current_flow_temperature

- platform: template
  sensors:
    heating_heating:
      friendly_name: "Heating circuit"
      value_template: >-
       {% if is_state("binary_sensor.heating", "on") and is_state("sensor.Warm_Water_3_way_valve", "off") and is_state("sensor.boiler_pump", "on") %} {{ true }}
       {% else %} {{ false }}
       {% endif %}
# used to get frequent updates!!!
      entity_id:
        - binary_sensor.heating
        - sensor.Warm_Water_3_way_valve
        - sensor.current_flow_temperature
````
What do you think about that?

Do you have other interessing ideas?

closed 2021-11-10 by proddy

toleabivol, 2024-01-04:
For GB172-20 now I get a sensor natively with total/dhw/heating energy . This can then be added to the energy dashboard and track costs of gas.

proddy, 2024-01-04:
> For GB172-20 now I get a sensor natively with total/dhw/heating energy . This can then be added to the energy dashboard and track costs of gas.

if you have time it would be great if you could write this up, with screenshots so we can post on the Wiki for others to learn. 🥇

toleabivol, 2024-01-04:
Sure thing.
Since the sensor comes directly from the ems there's not much one must do, just go to the energy dashboard and select it for the gas consumption.

[image]
[image]

I am now looking into calculating the cost/kW for liquified gas so I can insert it. I have a tank burried in the ground and fill it up from time to time. The cost depends on the market value when I am carrying out the tank fill-up order. The complicated part is that I am charged with L (liters). 
If I am right we can calculate how many kW can a Liter of liquified gas produce and thus get a static number for the cost of gas.

toleabivol, 2024-01-04:
If one wants to see separate values for water and heating then just use the other 2 sensors:

[image]

[image]

toleabivol, 2024-01-04:
Hmm, not sure why the 2 grapsh don't add up, the 3 sensors look to be adding up in history .
Could be due to the annomaly at ~11:30 when the dhw energy went down (could be due to me restarting ems?).

[image]

proddy, 2024-01-09:
I think it's better if you open a separate GitHub issue for this so we can track and comment. the values should be stable and linear