emsesp/EMS-ESP32 #157: Sensors DS18B20

C-arsten, 2021-10-13:
I connected 11 Dallas DS18B20 Sensors to the Board via JST.
10 Sensors show the right ID and Temp. The 11 doesnt show ID and Temp.
Telnet -> Show -> all 11 Sensors are named with right ID and show the right Temp.
The Sensor are all working fine also in ESP_Easy.
[image]
[image]

C-arsten, 2021-10-14:
i can uplaod the stable v3.2.1 but not the v3.2.2b11

proddy, 2021-10-14:
> i can uplaod the stable v3.2.1 but not the v3.2.2b11

try the old fashioned way using a USB cable and https://github.com/Jason2866/ESP_Flasher/releases/tag/v.1.3

commit c029cf79 (2021-10-14): use smaller json key names, increase buffer size. fixes #157