emsesp/EMS-ESP32 #174: Support for Tado thermostat

Ruben9477, 2021-10-31:
Hi, 

Would be asome if my tado thermostat (multizone) is visible for the EMS gateway. 
Is there anyone willing to investigate the log? 

Thanks!

MichaelDvP, 2021-11-01:
@Ruben9477 Send a description of your system, the `system info`(see help page on your emsesp) and the full log while changing some values on the thermostat. 
I think the tados are simple room-thermostats with only a few values implemented, compatible to a known buderus/junkers-type. I don't expect that Bosch has implemented special telegrams for tado.

Ruben9477, 2021-11-01:
@MichaelDvP Hereby my system info: 

{
  "System": {
    "version": "3.2.2b14",
    "uptime": "000+20:13:36.123",
    "uptime_sec": 72816,
    "freemem": 122,
    "reset_reason": "Software reset CPU / Software reset CPU"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 78690,
    "read requests sent": 4849,
    "write requests sent": 0,
    "incomplete telegrams": 174,
    "tx fails": 4982,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT publishes": 30482,
    "MQTT publish fails": 0,
    "dallas sensors": 0,
    "dallas reads": 0,
    "dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Nefit GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.01)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:190, Version:01.03)"
    }
  ]
}

And hereby the log, changing Tado setpoint's: 

[log-2.txt](https://github.com/emsesp/EMS-ESP32/files/7455437/log-2.txt)

MichaelDvP, 2021-11-02:
The Tado emulates a RC10, but does not responde to version request. That's a problem, emsesp uses the version to register a device. I'm not sure if the Tado is readable/writeable from the bus, or only sends out a few commands:
```
Thermostat temperature/setpoint: 17(0x17) -> All(0x00), ?(0xB1), data: 04 14 00 7B 00 00 00 00
Warm water on to boiler:         17(0x17) -> Boiler(0x08), UBAFlags(0x35), data: 11 11
Warm water temperature:          17(0x17) -> Boiler(0x08), UBAParameterWW(0x33), data: 37 (offset 2)
Setpoint flowtemperature:        17(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 4B 64 64 00
``` 
can you check from terminal `read 17 B0`?
I'm thinking about the best way to handle the missing response to version-request. One solution is to add a web setting for the Tado, or a "hack" to register a generic thermostat for device-ID 0x17 with no version. 
I made a test for this hack in my [experimental version](https://github.com/MichaelDvP/EMS-ESP32/releases), could you check if the tado is registered. This includes also a write of the setpoint in telegram B0, check if it works.
Also check if the Tado is reading the warm water temp from boiler: change boiler wwseltemp (ww selected temperature) from emsesp and check if Tado reads it or overwrites it.

proddy, 2021-11-02:
there must be some handshaking between boiler & the tado otherwise the UBA would discard it, I would expect. Perhaps power-off and on the Tado and watch the first telegrams that happen between 0x17 and 0x08?

Ruben9477, 2021-11-02:
@MichaelDvP I have installed your experimental version, thanks!
EMS Gateway now makes thermostat visible. With only one zone activated the setpoint and temp reading are correctly viewed in EMS Gateway. 
[log-only-one-zone-active.txt](https://github.com/emsesp/EMS-ESP32/files/7462922/log-only-one-zone-active.txt)
[emsesp_system_info-3.txt](https://github.com/emsesp/EMS-ESP32/files/7463109/emsesp_system_info-3.txt)

after activating the second zone, the highest setpoint follows in EMS gateway with corresponding zone temperature. 
[log-two-zones-active.txt](https://github.com/emsesp/EMS-ESP32/files/7462906/log-two-zones-active.txt)

With multiple zones activated it's not completely clear what Tado is choosing for setpoint and measurement. Looks like it's choosing the highest setpoint and corresponding value from this zone. 
[log-alle-zones-highest-sp-wins.txt](https://github.com/emsesp/EMS-ESP32/files/7462921/log-alle-zones-highest-sp-wins.txt)

I have the hot tapwater function disabled in Tado, there is only a small boiler inside (CW5). Tado is not reading temperature, I can only give a setpoint in Tado and if it's matches the range of possibilities it changes over, but if I give an setpoint of 65 for example, the boiler doesn't change. It is changing directly the tapwater temperature and not only the temperature from the small boiler. So for me this function is useless. see log changing temperatures: 
[log-changing-tapwater.txt](https://github.com/emsesp/EMS-ESP32/files/7463069/log-changing-tapwater.txt)

>can you check from terminal read 17 B0?
don't know how? 

Tado isn't accepting write request for heating, setpoint is not changing and gives error report's in log: 
[changing-thermostat-setpoint.txt](https://github.com/emsesp/EMS-ESP32/files/7462955/changing-thermostat-setpoint.txt)

@proddy see log after connecting Tado thermostat:
[log-reconneting Tado.txt](https://github.com/emsesp/EMS-ESP32/files/7462870/log-reconneting.Tado.txt)

MichaelDvP, 2021-11-03:
Interresting, when using multible zones the Tado not only changes the flowtemp on boiler, it changes the setpoint too.
And the setpoint is not writable from ems-bus, i'll remove the setting.

Is this CW5 an extra boiler? I see in the log that tado changes the setpoint for warm water in ems boiler and warm water function is enabled.

The reconnect log is very interresting. 

@proddy 
What do you think about this "hack"? 
We have seen before that some devices do not reply to requests (modem 0x0D, CRF200S, etc.), normal devices send empty telegram if the type-id is not implemented. This empty telegram is also used by emsesp to toggle the fetch.
To reduce TX-errors i've generate a empty telegram in rx-queue, [send after 3 read retries](https://github.com/MichaelDvP/EMS-ESP32/blob/0b41e0ce9304083e8d7df88d3c6e350bb41a194b/src/telegram.cpp#L589-L591) and [here](https://github.com/MichaelDvP/EMS-ESP32/blob/0b41e0ce9304083e8d7df88d3c6e350bb41a194b/src/telegram.cpp#L237-L244). This works as we see here for the setting telegram, T235 is the generated telegram:
```
2021-11-02 18:13:40.059 T 233: [emsesp] Me(0x0B) <- Thermostat(0x17), RC10Set(0xB0), data: 20
2021-11-02 18:13:40.309 E 234: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 97 B0 00 20
2021-11-02 18:13:40.309 T 235: [emsesp] Thermostat(0x17) -> Me(0x0B), RC10Set(0xB0), data: <empty>
2021-11-02 18:13:40.309 D 236: [emsesp] This telegram (RC10Set) is not recognized by the EMS bus
2021-11-02 18:13:40.309 D 237: [emsesp] Toggling fetch for device ID 0x17, telegram ID 0xB0 to 0
``` 
Now we also get a empty version telegram, normally sorted out, [but now set to product id 0](https://github.com/MichaelDvP/EMS-ESP32/blob/0b41e0ce9304083e8d7df88d3c6e350bb41a194b/src/emsesp.cpp#L771-L773), added to [device-library](https://github.com/MichaelDvP/EMS-ESP32/blob/0b41e0ce9304083e8d7df88d3c6e350bb41a194b/src/device_library.h#L141). Now we can [add known devices](https://github.com/MichaelDvP/EMS-ESP32/blob/0b41e0ce9304083e8d7df88d3c6e350bb41a194b/src/emsesp.cpp#L1024-L1051) that does not have version info by it's device-ID. 
This happens here:
```
2021-11-02 18:13:32.072 D 194: [telegram] Sending read Tx [#26], telegram: 0B 97 02 00 20
2021-11-02 18:13:32.107 T 195: [emsesp] Me(0x0B) <- 17(0x17), Version(0x02), data: 20
2021-11-02 18:13:32.345 E 196: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 97 02 00 20
2021-11-02 18:13:32.345 T 197: [emsesp] 17(0x17) -> Me(0x0B), Version(0x02), data: <empty>
2021-11-02 18:13:32.345 D 198: [emsesp] Adding new device generic thermostat (device ID 0x17, product ID 0, version 00.00)
2021-11-02 18:13:32.345 D 199: [thermostat] Adding new thermostat with device ID 0x17 (as master)
```
What do you think, merge it to dev, or keep official emsesp clean, reading the bus as it is without spoofing telegrams?

proddy, 2021-11-03:
> What do you think, merge it to dev, or keep official emsesp clean, reading the bus as it is without spoofing telegrams?

I don't think it's a hack, I actually think it is quite clever and a perfectly acceptable solution. And we get to finally use the Generic class. Maybe some text changes we can look at later but its good - merge it in

MichaelDvP, 2021-11-03:
Ok, i've merged, @Ruben9477 please check if i've missed something. Your Tado should now also work with[ official build](https://github.com/emsesp/EMS-ESP32/releases). The non-working write to setpoint is removed.

Ruben9477, 2021-11-03:
@MichaelDvP 
Thanks a lot for u time!

> Is this CW5 an extra boiler? I see in the log that tado changes the setpoint for warm water in ems boiler and warm water function is enabled.

There is no extra boiler installed, I can choose to keep the hot tap water preheated (only a couple liters I think?).
It's a setting to choose tapwater comfort.  Keep it always on temperature (hot), Smart learning (intelligent) or only when asked. (eco) Tado has a function to adjust the tapwater temperature, but it actually changes the set temperature and not the pre heating temperature. I think Tado designed this function for external boilers, not really functioning for this setup. 

For testing ur previous build I enabled the function in Tado, now I disabled it and setting on boiler is eco. 

> Ok, i've merged, @Ruben9477 please check if i've missed something. Your Tado should now also work with official build. The non-working write to setpoint is removed.

See log after your last build here: 

[log.txt](https://github.com/emsesp/EMS-ESP32/files/7470753/log.txt)

Setting the thermostat is disabled, but I still see some error for reading telegrams.

Ruben9477, 2021-11-03:
@MichaelDvP 
I noticed something strange with this build, values are mixed up?
[image]

MichaelDvP, 2021-11-04:
After updating the browser holds some cached parts of the old page with data for the new, This shows often strange things. You have to reload page in browser after the update.

proddy, 2021-11-04:
> After updating the browser holds some cached parts of the old page with data for the new, This shows often strange things. You have to reload page in browser after the update.

when I get to implementing https://github.com/emsesp/EMS-ESP32/issues/178 it should be easier

bbqkees, 2021-11-04:
Another great addition to EMS-ESP. Thanks guys.

MichaelDvP, 2021-11-04:
@Ruben9477 
> but I still see some error for reading telegrams.

yes, mainly after startup, but this is part of the plan. We use the errors to identify the device.

Can you please try to log with the tado set to ems+ mode. I think the Tado will also send out some telegrams and we are prepared if someone will use it. ems+ is downward compatible, maybe it works. I think the HT3-mode will not send to the bus, but if you like to try i'll take a look in the log.

Ruben9477, 2021-11-05:
@MichaelDvP 

See log after changing tado to EMS+, Now it sends its device ID, but no setpoint or temperature visible in EMS gateway. 

[Log-Tado-Connect-EMS+.txt](https://github.com/emsesp/EMS-ESP32/files/7481605/Log-Tado-Connect-EMS%2B.txt)

Boiler is not responding on EMS+ mode, so I changed back EMS
[Log-changing-setpoints-tado-EMS+.txt](https://github.com/emsesp/EMS-ESP32/files/7481608/Log-changing-setpoints-tado-EMS%2B.txt)

MichaelDvP, 2021-11-05:
Thanks, 
seems in ems+ mode the Tado only works if there is a master-thermostat (RC300) in system. It uses a remote thermostat id (0x38) and sends all values to the master at 0x10 (which is not present in your system). 
If there is a master we will read setpoint and roomtemp from the master, i think there is no need to read direct from Tado, but setpoint is send here `Thermostat(0x38) -> 10(0x10), RC300Set(0x2B9), data: 24 (offset 10)` and i think the roomtemp are in first 2 bytes of this unknown telegram: `Thermostat(0x38) -> 10(0x10), ?(0x42B), data: 00 CA 07 EB 01` 
Funny that the Tado sends a version info with brand ' Buderus' and that the version request from emsesp is not replied direct (causes emsesp to repeat request). The reply is on next poll as normal message to emsesp. 

I think for now we dont need extra code to handle tado-ems+. 
BTW: writing the setpoint from ems is not useful for this implementation, we've seen that tado in multizone not only changes the boiler flowtemp, it also adapts the setpoint if another zone needs heat. So a write from ems will be internally overwritten by the multizone calculation and has no effect.

bbqkees, 2021-11-05:
So it looks like they reverse engineered a simple Buderus thermostat and just emulated it in the Tado firmware.
So with no assistance from Bosch at all.

proddy, 2021-11-05:
thats what we did ;-)

Ruben9477, 2021-11-05:
Thanks for your time guys, looks like this issue is solved now :)

MichaelDvP, 2021-11-09:
I think we can close this and reopen if someone uses tado ems+ or ht3 mode.

proddy, 2021-11-10:
@MichaelDvP shall we mention something about Tado support in the documentation?

MichaelDvP, 2021-11-10:
Yes, mention that Tado ems mode is supported read-only as "generic thermostat", but ems+ mode and ht3 mode needs help from owner and is not yet supported..

berndh1979, 2021-12-03:
ALl, really impressed with the work done. Wel done. I have a BBQKees device attached to my Tado icw Trendline HRC30 CW5. Is there anything I can provide/help you with to extend the the Tado support?

proddy, 2021-12-03:
Thanks for the support. For Tado additions, I'm not sure..if you have any suggestions I'll see if we can implement them

closed 2023-02-19 by proddy