emsesp/EMS-ESP32 #176: New version 3.3.0b0 crashes on api calls

tp1de, 2021-11-02:
@proddy @MichaelDvP 

I tried loading 3.3.0b0 version. While using the api calls from my ioBroker adapter the ems-esp crashes by issuing the api call
`ems-esp/api/boiler`. The ems-esp device is not reachable anymore by web-ui or telnet. After a hard reboot web-ui and dashboard is working again but on next api call the system crashes again.

I went back to 3.2.2b14 from a week ago ....

proddy, 2021-11-02:
which API call is causing the crash?

tp1de, 2021-11-02:
ems-esp/api/boiler

tp1de, 2021-11-02:
I will test again ... if this API call will to the crash or the number of following API calls for detailed fields.
Could it be a memory allocation problem?

proddy, 2021-11-02:
no, probably a parsing error but I did try and test for every scenario. a GETto http://ems-esp.local/api/boiler works fine for me.  if you can help drill down to the exact call that causes the crash it would help.

tp1de, 2021-11-02:
> no, probably a parsing error

I agree but I need to adjust my adapter slightly to get the respective API call which gives the crash.
I think it will be one of the API call per field to get the attributes ...

tp1de, 2021-11-02:
I reloaded 3.3.0b0 again. telnet shows an error on startup:

[image]

I just recognized that API calls per device now return long names and not short names anymore. (rettemp is now return temperature)
The API calls per field does not seam to work anymore: ems-esp/api/boiler/gas makes the ems-esp to reboot !

proddy, 2021-11-02:
ok, the latest build from 30mins ago should have improved error reporting so try that if you can. The short-names I was wondering about that myself, and will fix. I'll also try locate the cause of the crash after a ems-esp/api/boiler/gas

tp1de, 2021-11-02:
I will wait until short names will be available again. Otherwise I will create new duplicate data fields with long names and my ioBroker adapter and the respective datafiles / diagrams will not work anymore.
Just tell me when short names will be fixed ...

> I'll also try locate the cause of the crash after a ems-esp/api/boiler/gas

The original data field was burngas and gas the long name !

proddy, 2021-11-02:
ok - found the issue and working on an improvement. I'll let you know, thanks again for testing

proddy, 2021-11-02:
ok firmware is building, perhaps wait until 16:10 and then grab it from the releases page

tp1de, 2021-11-02:
@proddy 
devices work now but `ems-esp/api/system`  returns no info: 
```
{
  "message": "invalid command 'list'"
}
```

tp1de, 2021-11-02:
BTW: was working before ...

proddy, 2021-11-02:
ok, new build ready for testing

tp1de, 2021-11-02:
Everything seems to be ok now. But I will observe until tomorrow....

closed 2021-11-03 by tp1de