emsesp/EMS-ESP32 #177: Launching the WebUI from Linux cashes havok

proddy, 2021-11-02:
From the Discord chat both @glitter-ball and @JakeOfOz mentioned issues getting the WebUI working on Linux. I install Firefox 84.02 on Ubuntu 21.10 (headless server) and can confirm when going to http://ems-esp.local/ it causes the EMS-ESP to reboot! Time to get the debugger out.

proddy, 2021-11-02:
oh, that is odd. I just tried again after cold rebooting the EMS-ESPs and it works. Both the Ethernet connected EMS Gateway and the old S32 Wifi. hmmm

glitter-ball, 2021-11-02:
I wasn't using Linux! Firefox 93.0 on Windows 10 and Edge on Windows 10 are the worst-behaved. Chrome also does odd things. Best-behaved browser is Safari on iPhone.

Don't really know where to start with this! I've been having all sorts of weird behaviour for months...
* Hanging showing "Loading..." at the logon prompt. Hitting Refresh causes the login dialog to flash up and then disappear and be replaced by "Loading..." again. Only fix is to clear the browser cache and start again.
* Hanging during WiFi scan. Tried multiple times with Firefox. Tried again with iPhone and it worked straight off.
* Firmware upgrades never work properly. I drag and drop the .bin file on the firmware update page and the progress bar whizzes to 100% and nothing else happens. When I navigate away from the page I get "Update cancelled" or similar. Often a reboot will show that the update has actually happened, but I'm always nervous of it.
* Restart button on the System page doesn't work. Click the button on the Confirm Restart page and it goes from blue to grey. Nothing visible happens and the page just hangs. Usually the device will restart but it's just messy.

Don't want to sound like a whinger, but it's a grim user experience! Worst still, not really sure why it's happening. I was ready to blame my home network, but when it does it on a wiped device in access point mode, that rules that out. The fact that I can get working pages on iPhone via home network also suggests the network isn't the problem. Safari on iPhone does seem to work pretty well. Just tried the restart on Edge for iPhone and I did get the "System is restarting" popup that I just don't see on laptop. Scan WiFI Networks also works on Edge for iPhone. On Firefox on laptop, with a clear cache, it will work once. Thereafter, it will just hang at the "Scanning..." dialog with progress bar. If I clear the cache, it will work again - just once. Thereafter, if I leave the page or close the session and log in again, the Scan WiFi page just hangs at the progress bar again. Weird, weird, weird.

So, there's something odd going on around animations / popups and the browser cache but I can't pinpoint it. I could try and capture some video if GitHub will let me post. It's sad - we have a good-looking, slick UI but it's just really flaky when I use it. I also wonder why more people don't report issues if it's something big/widespread. I can replicate the issues on at least three different machines.

proddy, 2021-11-03:
it is odd, but I'd like to know what is causing it. There are probably more than 1500 users with EMS-ESP out there, maybe more (@BBQKees knows the numbers) and I haven't heard of issues before.

Do you have any anti-virus software on your Windows 10 machine? If so try disabling them. EMS-ESP does support SSL, so only HTTP and maybe something doesn't like it. Is there another PC or laptop you can try out in your network (other than a mobile device) that will eliminate any network-related interference?

As for the Restart button, I'll see if there is something I can do with the UI like we had back in version 2 which shows a blank screen and tries to refresh in 10 seconds. The difficulty is that the web session is gone since the web server instance running on the little ESP32 is destroyed so there's no way for EMS-ESP to interact back with the browser page. I'll create a new GitHub issue for this.

bbqkees, 2021-11-03:
@glitter-ball here it works fine. You are also the one with the most hardware issues it seems. Maybe that's related. Try another ESP32 board. A year or two back we had very strange connection issues that turned out to be a bad batch of ESP boards.
And perhaps an ad blockers on your system adds problems as well.

I did get a few emails about the web interface not loading over Ethernet while it worked fine over WiFi but that's it lately.

glitter-ball, 2021-11-03:
Only anti-virus is Windows Defender. Lots of other people will be using that so I don't think that's the problem. I'm using some web filtering / parental controls stuff but I disabled that and it still didn't work. I have another client without this so I'll try that too.

Networking is tricky. I get the problems even with the EMS-ESP in access point mode, so there's nothing else in the way between the client and EMS-ESP. I also don't get issues when using the home network from an iphone. So, there's something about my PCs and/or their browsers which is breaking it.

I was having a lot of the UI issues with the ESP8266 too, which is a useful clue. I need to get another ESP32. The one which arrived recently is proving tricky to reflash so may be duff.

So, I think we probably have to blame my PC-based clients...

proddy, 2021-11-04:
Pokito from Discord also is seeing issues. The EMS-ESP gets stuck and crashes with a async_tcp error when retrieving a large amount of data (like some of the JavsScript frontend). We're seeing if its only Client Wifi to Ethernet EMS-ESP or only Ethernet-to-Ethernet. Since I'm unable to reproduce I'll go ahead and make a few special branches with some tweaks to the TCP library to see what helps

glitter-ball, 2021-11-04:
As I said in Discord, I could hook a stand-alone ESP32 to a PC and monitor the USB console when I try some stuff, in case it spits out any useful data.

proddy, 2021-11-05:
One scenario we know that causes EMS-ESP to crash when loading the WebUI is on a (too) fast network. Pokito (from Discord) did some tests from an Ethernet-connected EMS-ESP to an Ethernet-connected laptop running Linux and was able to reproduce the crash with a `wget` command (no browser). By adding a 1ms latency (using `sudo tc qdisc add dev $INTERFACE root netem delay 1ms`) it solved the issue. And it looked like Ubuntu 18.04 is more stable than Ubuntu 20.04. So in this its seems the TCP async trips up. I'll make the special branch with some tweaks for experimentation.

@glitter-ball anything you could play with would help too, but I'm not sure right now where to begin

cyberfox01, 2021-11-05:
I am Pokito from Discord.

Here is the script I use for testing. It is a little overengineered, but testing can be boring sometimes :-)
When started as root or with sudo, an artificial delay is added to the network interface (currently set to 1ms) and the whole communication is recorded with tcpdump.
Change to DELAY="" and/or RECORD_PCAP="" to disable that.

[crash-test.txt](https://github.com/emsesp/EMS-ESP32/files/7481427/crash-test.txt)

VDRainer, 2021-11-05:
I'm also having this issue with EMS-ESP32 crashes, but only if i open the WebUI over a wired connection.

I'm using the Gateway E32 with Ethernet.

Pulling the cable on my Notebook and access the WebUI over WLan works without problems.
Latest Firefox and Chrome on Debian 11 (Ubuntu 18.04 until a month ago).

From a Windows 10 PC and a Virtualbox Windows 10 VM with bridged Network, there are no problems over wired connection.

Here's a  debug log that i made a while ago, but it didn't show anything relevant to me and i was to lazy to report. :smile:
```
Sep 13 15:22:05 ems-esp mqtt 000+00:42:20.298 D 215: Publishing topic ems-esp/thermostat_data (#1075, retain=1, retry=1, size=1023, pid=1)
Sep 13 15:22:05 ems-esp mqtt 000+00:42:20.399 D 216: Publishing topic ems-esp/mixer_data (#1076, retain=1, retry=1, size=89, pid=1)
Sep 13 15:22:09 ems-esp telegram 000+00:42:24.503 W 217: Incomplete Rx: 21 E8 00
Sep 13 15:22:14 ems-esp mqtt 000+00:42:30.062 D 218: Publishing topic ems-esp/boiler_data (#1077, retain=1, retry=1, size=669, pid=1)
Sep 13 15:22:14 ems-esp mqtt 000+00:42:30.163 D 219: Publishing topic ems-esp/boiler_data_ww (#1078, retain=1, retry=1, size=461, pid=1)
Sep 13 15:22:15 ems-esp mqtt 000+00:42:30.265 D 220: Publishing topic ems-esp/thermostat_data (#1079, retain=1, retry=1, size=1023, pid=1)
Sep 13 15:22:15 ems-esp mqtt 000+00:42:30.367 D 221: Publishing topic ems-esp/mixer_data (#1080, retain=1, retry=1, size=89, pid=1)
------------- CRASH ---------------------------------------------------------------
Sep 13 15:22:25 ems-esp emsesp 000+00:00:00.000 I 1: Ethernet initialized
Sep 13 15:22:25 ems-esp emsesp 000+00:00:00.000 I 0: Starting Syslog
Sep 13 15:22:25 ems-esp syslog 000+00:00:00.000 I 2: Log level set to DEBUG
Sep 13 15:22:26 ems-esp telegram 000+00:00:00.000 D 3: Tx read request to device 0x08 for type ID 0x07
Sep 13 15:22:26 ems-esp emsesp 000+00:00:00.000 I 4: EMS Device library loaded with 79 records
Sep 13 15:22:26 ems-esp emsesp 000+00:00:00.000 I 5: Starting Access Point with captive portal on 192.168.4.1
Sep 13 15:22:26 ems-esp telegram 000+00:00:01.411 W 6: Incomplete Rx: 21 08 00 07 00 03 01 00 02 00 00 00 00 00 00 00 00 00 00 00
Sep 13 15:22:26 ems-esp emsesp 000+00:00:01.720 D 7: No telegram type handler found for ID 0x19 (src 0x08)
Sep 13 15:22:26 ems-esp emsesp 000+00:00:02.402 D 8: No telegram type handler found for ID 0x18 (src 0x08)
Sep 13 15:22:26 ems-esp telegram 000+00:00:02.809 W 9: Incomplete Rx: 21 08 00 34 00 38 02 35 83 00 21 00 01 03 00 03 8C EA 00 6C D3
Sep 13 15:22:26 ems-esp emsesp 000+00:00:03.096 D 10: No telegram type handler found for ID 0x14 (src 0x08)
Sep 13 15:22:26 ems-esp emsesp 000+00:00:04.281 I 11: Starting NTP
Sep 13 15:22:26 ems-esp emsesp 000+00:00:04.305 I 12: Ethernet Connected with IP=192.168.77.14, speed 100 Mbps
Sep 13 15:22:30 ems-esp mqtt 000+00:00:04.305 I 13: MQTT connected
Sep 13 15:22:30 ems-esp mqtt 000+00:00:04.305 D 14: Subscribing to topic: ems-esp/system
Sep 13 15:22:30 ems-esp mqtt 000+00:00:04.409 D 15: Publishing topic ems-esp/info (#01, retain=1, retry=1, size=58, pid=1)
Sep 13 15:22:30 ems-esp mqtt 000+00:00:04.510 D 16: Publishing topic ems-esp/heartbeat (#02, retain=1, retry=1, size=171, pid=16408)
Sep 13 15:22:30 ems-esp mqtt 000+00:00:04.611 D 17: Subscribing to topic: ems-esp/system
Sep 13 15:22:31 ems-esp mqtt 000+00:00:04.712 D 18: Publishing topic ems-esp/status (#04, retain=1, retry=1, size=6, pid=1)
Sep 13 15:22:27 ems-esp mqtt 000+00:00:06.234 D 19: Publishing topic ems-esp/heartbeat (#05, retain=1, retry=1, size=171, pid=1)
Sep 13 15:22:28 ems-esp telegram 000+00:00:07.017 W 20: Incomplete Rx: 21 89 00
Sep 13 15:22:30 ems-esp telegram 000+00:00:08.740 W 21: Incomplete Rx: 10 21 00
Sep 13 15:22:30 ems-esp telegram 000+00:00:09.111 W 22: Incomplete Rx: 10 21 00
Sep 13 15:22:30 ems-esp emsesp 000+00:00:09.571 D 23: No telegram type handler found for ID 0x48 (src 0x10)
Sep 13 15:22:31 ems-esp emsesp 000+00:00:10.001 I 24: Stopping Access Point
Sep 13 15:22:32 ems-esp telegram 000+00:00:10.997 W 25: Incomplete Rx: 21 00 AB 00 05 01
Sep 13 15:22:33 ems-esp telegram 000+00:00:11.804 W 26: Incomplete Rx: 21 00 AB 00 05 01 26 00 88
Sep 13 15:22:33 ems-esp emsesp 000+00:00:11.914 D 27: No telegram type handler found for ID 0x18 (src 0x08)
Sep 13 15:22:33 ems-esp emsesp 000+00:00:12.195 D 28: No telegram type handler found for ID 0x18 (src 0x08)
Sep 13 15:22:34 ems-esp telegram 000+00:00:12.702 W 29: Incomplete Rx: 21 00 AB 00 05 01 26 00 9C 00
Sep 13 15:22:34 ems-esp emsesp 000+00:00:12.802 D 30: No telegram type handler found for ID 0x34 (src 0x08)
```
Please contact me if you need further informations.

proddy, 2021-11-05:
@cyberfox01 could you try the build from the `dev_webui_crash` branch? I increased the queue size

proddy, 2021-11-05:
and thanks @VDRainer for your input...looks like there is a common thread here. Linux & Ethernet which is the same as Pokito

cyberfox01, 2021-11-05:
I have built a new binary from the dev_webui_crash branch and flashed it to the device, but sadly that did not change anything noticeably. Still most of the requests fail:
1: failed! (142019/161421 bytes)
2: failed! (153507/161421 bytes)
3: failed! (150635/161421 bytes)
4: failed! (144891/161421 bytes)
5: failed! (133403/161421 bytes)
6: failed! (152071/161421 bytes)

The new binary has the same version 3.2.1 but I have checked and the binaries differ to the original 3.2.1. I was afraid that I built something wrong or flashing did not work.

proddy, 2021-11-05:
damn, was hoping this would be the silver bullet. The WebUI needs & cache refresh if you're looking for a version in the WebUI. Or via `curl -s http://ems-esp.local/api/system | grep version`

cyberfox01, 2021-11-05:
$ curl -s http://192.168.12.123/api/system | grep version
    "version": "3.2.1",

Should it show something different than 3.2.1?

proddy, 2021-11-05:
yes, `"version": "3.3.0b1"`. I'll make a binary for you

proddy, 2021-11-05:
> yes, `"version": "3.3.0b1"`. I'll make a binary for you

https://github.com/proddy/EMS-ESP32/releases/download/test/EMS-ESP-3_3_0b2-ESP32.bin

cyberfox01, 2021-11-05:
Thanks, I have downloaded and flashed the binary.
Had to change the test-script because the large js file is now named 2.1b9b.js and has size 164165, but amazingly it works now!
No more reboots on 100/100 tries. Congratulations!

proddy, 2021-11-05:
great! I'll make one more change if you could try that too, and then that should be it. Each build will have a different filename for the JS chunk. It's by design (called cache busting) and forces the WebUI to refresh every time new firmware is uploaded (it invalidates the cache).

cyberfox01, 2021-11-05:
No problem, chuck it over

proddy, 2021-11-05:
https://github.com/proddy/EMS-ESP32/releases/download/test/EMS-ESP-3_3_0b2b-ESP32.bin

proddy, 2021-11-05:
this uses half the cache size in the tcp buffer

cyberfox01, 2021-11-05:
Works as well, 100/100 requests completed without reboot

proddy, 2021-11-05:
great, thanks a million for testing. I'll apply to the fix to the dev branch.

cyberfox01, 2021-11-05:
I just double-checked: flashed again v3.2.1 => errors, then 3.3.0b2b => no more errors

cyberfox01, 2021-11-05:
> great, thanks a million for testing. I'll apply to the fix to the dev branch.
You're welcome, it was also fun to investigate :-)

VDRainer, 2021-11-05:
I can confirm, no more issues over ethernet with 3.3.0b2b.
Thanks so much! :+1:

proddy, 2021-11-05:
@cyberfox01 @VDRainer you guys, thanks! you rock.

glitter-ball, 2021-11-06:
I've now built a Lubuntu machine  to investigate the long-standing UI issues I've been having. I've connected this to a new ESP32 not in a gateway and in AP mode to eliminate as much of my home system as possible.

The single biggest change has been the firmware update page. When I've had problems, I drag the file into the upload box and the progress runs to 100% almost immediately. The session then appears to hang. If I refresh the page I get "Upload cancelled by user" or similar.

With the barebones Lubuntu client running Firefox, when I drag the firmware into the box, the progress bar moves gradually up to 100% over, perhaps, 45 seconds. This probably accurately represents the time to dump the code over the WiFi into the device and process it. Shortly after, I get "Activating new firmware." If I refresh the page shortly after, the upgrade's happened.

So, I guess the question for is... what in my setup could be breaking the progress display when uploading the .bin file? How is the progress bar updated? My guess is that it's running up to 100% on the UI before the file has actually finished transfer to the EMS-ESP. Either the session then hangs or I'm not patient enough to wait for the firmware activation message.

Checked again on the home network instead of AP mode. Still works. So the issues arise from either the config of my Windows clients or the original ESP32 which is suspect. I'll attack the second possibility by swapping over the ESP32s and trying again to see what happens.

proddy, 2021-11-06:
30-60 seconds sounds about right. If you're able to get a Serial connection from your ESP32 then open it up and do the Firmware upload again. You'll see a bunch of information in the terminal window showing the % progress straight from the internal Arduino upload code. It should give some pointers where its breaking.

So you're saying on your Ubuntu with Firefox it works over WiFi. In what scenario does it *not* work?

glitter-ball, 2021-11-06:
Issues seem to occur on my Windows clients with almost any browser. Lubuntu and iOS work flawlessly. I wondered if it was some web filtering software on the Windows builds but I disabled that and it still happened. I'll do some experiments with the console connected and see what this chucks out.

proddy, 2021-11-06:
and you're using the latest 3.3.0b2? Can't think of what else is causing it. You could try Windows safe mode but now its getting a little silly

glitter-ball, 2021-11-06:
Yep 3.3.0b2. Did two firmware updates with the console monitored - dump attached.

First from the Lubuntu client - all went without a hitch.

Second from the Windows client. Progress bar whizzed to 100% then nothing else happened. 'Show system' revealed that the ESP32 hadn't rebooted and would probably have sat like that indefinitely. As soon as I refreshed the Firmware Update web page, I got "Firmware update cancelled" pop-up and the console showed the device restarting with the new firmware. Just weird. How does the progress bar show 100%  way before the file could have been transferred?... 

[2021 11 06 180535.txt](https://github.com/emsesp/EMS-ESP32/files/7491037/2021.11.06.180535.txt)

I think this could turn into a big time-waster, especially if I'm the only person having this precise issue, and efforts are better-used elsewhere. I'll stick to using Firefox on Lubuntu for now!

proddy, 2021-11-06:
probably not worth chasing, I agree, but its annoying!

greg-de, 2021-11-06:
Hi proddy,
you are not alone with problems with WebUI. I'm new to the topic having gateway with Wifi and Ethernet and I'm not able to connect to WebUI using LAN connection. It works only with Wifi. If it would help investigating the topic I can provide logs or test some versions.

proddy, 2021-11-06:
> Hi proddy, you are not alone with problems with WebUI. I'm new to the topic having gateway with Wifi and Ethernet and I'm not able to connect to WebUI using LAN connection. It works only with Wifi. If it would help investigating the topic I can provide logs or test some versions.

Ok, great that you'll able to help. First step is to take the latest dev build 3.3.0b2 and see if that helps.

greg-de, 2021-11-06:
No problem I will make a test but have to wait till Tuesday due to a business trip.

greg-de, 2021-11-09:
Hi proddy,

success! the latest version of ems-esp (EMS-ESP version 3.3.0b3) shows WebUI on LAN network (Setup: Linux Mint 20 with 5.4.0-89 + Chrome/Firefox).

Thanks a lot!

proddy, 2021-11-09:
great! thanks for testing it out. I'll close this issue for now.

closed 2021-11-09 by proddy

proddy, 2024-02-04:
I've moved the async queue length from 64 back to 32 (in AsyncTCP) as this is default for the new library. Making a note here in case users have problems again.