emsesp/EMS-ESP32 #179: make ems-esp32 compatible with the new "Entity categorization" in Home Assistant V2021.11

bartplessers, 2021-11-03:
Hi,

Home Assistant v2021.11 introduces a new feature "Entity categorization"
https://www.home-assistant.io/blog/2021/11/03/release-202111/#entity-categorization

Is it somehow possible to have ems-esp32 this capability adapted?
It would be great to have only the "important" EMS-entities exposed to my lovelace dashboard. Right now, I have so many (binary)sensors exposed to my autogenerated Lovelace dashboard that it sometime looks like a control room of an industrial powerplant :-)

kind regards,
Bart Plessers

proddy, 2021-11-03:
yes, I could add, but after a quick look, I'm not sure of exactly how. If I disable an entity in HA via the UI I noticed it also disappears as an actual entity and is also unreadable? Then there's the question of what are the important entities to show and which ones to hide. Maybe an easier solution is to use the auto-generate lovelace dashboard and then customize it. Dunno. Requires some more thinking....

bartplessers, 2021-11-04:
Hi @proddy 
I suppose it's an extra property of the sensors. So not only 'disabled' but also 'configuration' or 'diagnostic', etc

True, to select the "important" properties is subjective. Fir my installation, this would be
- inside temp
- outside temp
- set temp
- current power

Grtz
B

proddy, 2021-11-05:
Right. We'll need to wait until this becomes supported in [MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/) first to be able to set the flag. Then we can start the debate on what to include/not to include

proddy, 2022-01-05:
until then the new customization service in v3.4 should tidy up the Lovelace UI. You can choose which sensors to show

bartplessers, 2022-01-07:
Any idea when v3.4 will be released?

proddy, 2022-01-07:
There are still some small quirks to iron out with @MichaelDvP and then I'd like to push it to the dev branch.

I'm doing daily builds, with another one today with some more HA improvements. I'm posting the firmware builds Discord too if you want an early peek. Or you can build it yourself from the source code in the v3.4 branch.

closed 2023-07-03 by proddy