emsesp/EMS-ESP32 #182: RC35 and controlling two heat circuits with custom time programs.

greg-de, 2021-11-06:
Hi all,

I have a Buderus Logano oil boiler with RC35 connected to it. It controls two heat circuits. 
According to the documentation this thermostat provides two user specific programs for every of each four heat circuits called Eigen 1 and Eigen 2 (German names). In my current configuration I'm using:
- for heat circuit 1 Eigen 1 program  and 
- for heat circuit 2 Eigen 2 program. 

Eigen 1 program shall be controlled by 0x3F telegram according to https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme, Eigen 2 shall be controlled by 0x42. 

Based on the code of thermostat.cpp (line 66) for RC35 only the first telegram (0x3F) is supported. As these both telegrams are used for example for switching on/off party or pause function I think that the current implementation of EMS-ESP app can only control the Eigen 1 program.  

Could you please check the implementation and if necessary extend it to support both programs. 

BR,
Greg

proddy, 2021-11-06:
sure, @MichaelDvP this is your expertise...

MichaelDvP, 2021-11-07:
The "Eigen 1" contains the used program, holiday, pause, party functions, "Eigen 2" only the switchprogram, Thats why we read only the first telegram. But for[ setting the switchtime](https://github.com/emsesp/EMS-ESP32/blob/91b7fd59d11b9d5a117c356dca4787c59d8dccb2/src/devices/thermostat.cpp#L1981) we should add an optional program-no and support writing a entry of second telegram. 

But i think each hc has its own 2 programs, You can use hc2 ownprog 1 with different values as hc1 ownprog 1.
Can you do in terminal `read 10 3f`, `read 10 42`, `read 10 49`, `read 10 4c` and post the result.

greg-de, 2021-11-07:
thanks for the info, I will send you the results on Tuesday/Wednesday as I'm currently on a business trip.

greg-de, 2021-11-09:
Hi MichaelDvP,

here are the results:

┌──────────────────────────────────────┐
│ EMS-ESP version 3.3.0b3              │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└──────────────────────────────────────┘

ems-esp:$ read 10 3f
000+00:00:58.582 N 0: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: 01 18 00 72 21 18 20 72 41 18 40 72 61 18 60 72 81 18 80 72 A1 18 A0 72 C1 18 C0
000+00:00:58.962 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: 18 C0 72 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 25)
000+00:00:59.209 N 2: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 50)
000+00:00:59.573 N 3: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: 90 E7 90 E7 90 E7 90 E7 90 00 00 00 01 01 00 01 01 00 01 01 00 01 01 00 (offset 75)
ems-esp:$ read 10 42
000+00:01:28.081 N 4: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x42), data: 01 21 00 84 21 21 20 84 41 21 40 84 61 21 60 84 81 21 80 8A A1 27 A0 8D C1 2A C0
000+00:01:28.335 N 5: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x42), data: 2A C0 84 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 25)
000+00:01:28.709 N 6: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x42), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 50)
000+00:01:29.036 N 7: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x42), data: 90 E7 90 E7 90 E7 90 E7 90 (offset 75)
ems-esp:$ read 10 49
000+00:01:35.330 N 8: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x49), data: 00 18 00 72 20 18 20 72 40 18 40 72 60 18 60 72 80 18 80 72 A0 18 A0 72 C0 18 C0
000+00:01:35.709 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x49), data: 18 C0 72 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 25)
000+00:01:36.085 N 10: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x49), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 50)
000+00:01:36.324 N 11: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x49), data: 90 E7 90 E7 90 E7 90 E7 90 0A 00 00 01 01 00 01 01 00 01 01 00 01 01 00 (offset 75)
ems-esp:$ read 10 4c
000+00:01:42.205 N 12: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x4C), data: 01 18 00 72 21 18 20 72 41 18 40 72 61 18 60 72 81 18 80 72 A1 18 A0 72 C1 18 C0
000+00:01:42.459 N 13: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x4C), data: 18 C0 72 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 25)
000+00:01:42.835 N 14: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x4C), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 50)
000+00:01:43.660 N 15: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x4C), data: 90 E7 90 E7 90 E7 90 E7 90 (offset 75)
ems-esp:$ `
`

greg-de, 2021-11-09:
Additional info:
I see in WebUI that for hc1 there is an entry hc1 program set to 'own_1' but for hc2 such entry does not exists. hc2 is using own_2 right now. 
Maybe it will help you.

MichaelDvP, 2021-11-09:
There is a entry missing in the list of possible programs, i'll add it.
Your program for hc2 looks a bit strange. All setpoints are set to off, but same times/dates as program for hc1.
What are the switchtimes for your programs, hc1 first starts with `01 18` , Mo, on, 4:00, second program in 0x42 starts with `01 21`, Mo on 5:30. hc2 first program is `00 18` Mo, off 4:00, and second starts with `01 18`.
The settings for program, holiday, vacations, party, pause are stored in the fisrt telegram, but i dont know which switchpoints are active.

MichaelDvP, 2021-11-09:
I've tested a bit, it seems that RC35_timer: 0x3F, 9x49, .. holds own_1 program and the additional settings to prog, etc. 
0x42, 0x4C, etc holds own_2. When changing the program only the program index changes. 

Then the new dev should work with command `switchtime` and parameters:
`<p.nn.d.o.hh:mm>` (program (optional), no of switchpoint, on, time) 
or `<p.nn>` to query a switchtime
```
ems-esp32:# call thermostat switchtime
{
  "name": "switchtime",
  "fullname": "hc2 single program switchtime",
  "circuit": "hc2",
  "type": "command",
  "writeable": true,
  "value": "not set"
}
ems-esp32:# call thermostat switchtime 1:01
000+00:24:07.467 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x49), data: 00 81 (offset 2)
ems-esp32:# call thermostat switchtime 2:01
000+00:24:12.717 N 10: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x4C), data: 00 84 (offset 2)
```

greg-de, 2021-11-09:
Both programs for HC1 and HC2 are the same but stored under different names:
For HC1 it is Own_1 starting a heating from 4:00 till 19:00 and
for HC2 it is Own_2 starting a heating from 4:00 till 19:00.

The reason for this unusual configuration is simple. I was not sure if each HC has two own programs or there are only two global programs for all HCs. 

Do you need any more info?

greg-de, 2021-11-09:
trying to test the commands you have written I get:

ems-esp:# call thermostat program
{
  "name": "program",
  "fullname": "hc1 program",
  "circuit": "hc1",
  "value": "own_1",
  "type": "enum",
  "enum": [
    "own_1",
    "family",
    "morning",
    "evening",
    "am",
    "pm",
    "midday",
    "singles",
    "seniors",
    "own_2"
  ],
  "writeable": true
}
ems-esp:# call thermostat switchtime
{
  "name": "switchtime",
  "fullname": "hc1 single program switchtime",
  "circuit": "hc1",
  "type": "command",
  "writeable": true,
  "value": "not set"
}

How it was possible for you to display info about HC2 using these commands?

MichaelDvP, 2021-11-09:
You have to update to the new dev i've pushed today.

greg-de, 2021-11-09:
OK, I will do it.

greg-de, 2021-11-09:
Ok, here are the results:

ems-esp:# call thermostat program
{
  "name": "program",
  "fullname": "hc1 program",
  "circuit": "hc1",
  "type": "enum",
  "enum": [
    "own_1",
    "family",
    "morning",
    "evening",
    "am",
    "pm",
    "midday",
    "singles",
    "seniors",
    "new",
    "own_2"
  ],
  "writeable": true,
  "value": "not set"
}
ems-esp:# call thermostat switchtime
{
  "name": "switchtime",
  "fullname": "hc1 single program switchtime",
  "circuit": "hc1",
  "type": "command",
  "writeable": true,
  "value": "not set"
}
ems-esp:#

greg-de, 2021-11-09:
But my question remains: How one can select for example hc2 in call thermostat switchtime? (how to select a circuit 2 or warm water)?

greg-de, 2021-11-09:
And are there any switches to see the switchtimes in a formatted way? e.g. Mo 4:00? 
In my understanding if I will set up switchtimes for Mo till So there will be seven entries in the switch time telegram. 
Is there an API to get a number of entered/active switchtimes in order to get single values? 

It would be perfect to have some API to get all switchtimes for every circuit (including warm water) at once. 
Do you think it would be possible to implement it? This way one could really control and trace remotely if a heating is working as expected.

MichaelDvP, 2021-11-10:
> But my question remains: How one can select for example hc2 in call thermostat switchtime? (how to select a circuit 2 or warm water)?

In console it was `call thermostat /hc2/switchtime` or `call thermostat switchtime ? 2` in v3.2.x, but these commands are removed in v3.3 (i think not intended). I'll add back these options.

ww-program settings are actual not supported. Maybe in future.
> And are there any switches to see the switchtimes in a formatted way? e.g. Mo 4:00?

No, i'm working on it, takes a while. A program can hold up to 42 switchtimes, thats a long string to publish and blows mqtt. I think changing program-switchtimes is done very rare and can be done on thermostat directly. Reading/writing a single timer-entry should be enough, users can use a script to read all values to their HA/domotics/ioBroker/whatever and process the there.

greg-de, 2021-11-10:
OK thanks for your replay. Yes, probably reading/writing a singe entry would be enough in order to keep strings short. For that it would be very helpful if there will be a counter showing how many real values are there. 

Fun fact about 42 switchtimes: My program is set from Mo to So which produces 7 of 42 possible entries. I was thinking at the beginning that in such case some kind of binary flag would be set and I was wondering who would need in such case all 42 entries :).

MichaelDvP, 2021-11-10:
> Fun fact about 42 switchtimes: My program is set from Mo to So which produces 7 of 42 possible entries.

No, you have every day a time with on in the morning and off in the evening, it's 14 switchtimes. 
For people out to work am and pm, but have lunch at home there are 6 switchtimes per day (morning on and off, noon on and off, evening on and off), 42 a week (if working on weekend).

Fun fact, your own_1 for hc2 has 14 switchtimes, all setting the heating off(night), but not a single turn on. You can fill all 42 times with switchtimes not changing the boiler state, useless, but possible.

MichaelDvP, 2021-11-11:
I'm thinking about how to show/set the switchtime. 
Human readable as `own_1 00 mo 08:00 on` (program, number, day, time,  state) or only use numbers making it easier to process in systems, as is is dokumented now: `1:00.0.1.08:00` (program, number, day, state, time)?
For warm water and circulation we should use the same scheme, but `ww | circ` as prog in human readable 
or for computers:  `0 | 1` or `w | c`.
Any suggestions for the delimiter?
For a not set value `own_1 32 not_set` (or `na` instead of `not_set`) or `1:32:7:0:24:00` (the Bosch code is day 7, 24:00) 
@proddy What's your opinion?

> For that it would be very helpful if there will be a counter showing how many real values are there.

Sorry, not possible, the data don't has to be in row. You can have switchpoints 1, 5, 32 set and all others unset. Also the switchtimes does have to be in timeline, 00 can be friday, 01 monday, 02 we, etc. If you create a program and send it to the thermostat it is best to send all 42 values.

proddy, 2021-11-11:
> @proddy What's your opinion?

Firstly I wouldn't worry too much about the output formatting as long as it's in JSON and can be easily parsed or traversed by other programs. Maybe ask Thomas to see how the KM200 describes this? For the input, again keep it simple and don't bother too much on the console syntax (no one will really use it). In the API the <value> can also be a JSON object so maybe use that instead of trying to parse a complicated string.

MichaelDvP, 2021-11-11:
?
The command functions only accepts strings as value, no json. Also a json looks strange on the webpage.
>  don't bother too much on the console syntax

I ask for the command/output, there is no change in console handling, please read the thread before reply and mixing strange things.

proddy, 2021-11-11:
ok, sorry for not reading. Go ahead and implement this as a PR. If it has a mojo witch-craft hacks in it, I'll review, comment and send it back.

MichaelDvP, 2021-11-12:
Ok, to finish this: Since there are no usefull sugestions, i've decided to use a readable text and publish an entry for each program.  This is shown in web and mqtt.
- wwswitchtime
- wwcircswitchtime
- hcx/switchtime1
- hcx/switchtime 2

Format is `00 mo 07:00 on` index, day, time, state, the seperators are published as blank, for setting it may be another value to identify a string, or put string in quotes, e.g.  in console `call thermostat hc2/switchtime1 01-mo-22:00-on` or `call thermostat hc2/switchtime1 "01 mo 22:00 on"`. Mqtt/api/web is always a string, no need for quotes. A invalid switchtime shows up as `32 not_set` and sending this clear a switchtime. Sending only the index number (e.g. `14`) returns the value to mqtt/web.
[image]
[image]
[image]

Software can be found [here](https://github.com/MichaelDvP/EMS-ESP32/tree/dev), and bin [here](https://github.com/MichaelDvP/EMS-ESP32/releases/download/latest/EMS-ESP-3_3_0b4-ESP32.bin).

I was kickt out of this project, without a message before. Also the `improve this article` in documentation leads to 404, seems that contributing is unwanted now.

proddy, 2021-11-12:
> I was kickt out of this project, without a message before. Also the improve this article in documentation leads to 404, seems that contributing is unwanted now.

not sure how you can get kicked out of an Open Source project?! Anyone can contribute. See CONTRIBUTE.md and https://emsesp.github.io/docs/#/Contributing.

greg-de, 2021-11-12:
Hi MichaelDvP, thanks for your hard work. 

I have started to test your implementation and here are the very first results:
- WebUI - I only see switchtimes for warm water, no info about hc1/hc2
- Telnet - here I see both ww and hc1 and hc2
Format of the strings: cool
Unfortunately I don't see the complete program. I would expect that in all cases ww,hc1,hc2 I will see 14 entries.
As you have calculated: two per day. I only see one entry with index 00. 

Have I misunderstood something?

proddy, 2022-01-26:
I believe this has been resolved in v3.4. If not, please re-open.

closed 2022-01-26 by proddy