emsesp/EMS-ESP32 #183: Improve moduline 200 functionality

kpschaper, 2021-11-07:
New issue to improve moduline 200. Ref: #181 

**B1:** 

- first byte 0 (mode)
> Maybe better to store it in modetype as it is the reported state, but since there is no mode-setting it does not matter. hamode can be set independent of the naming. As there is a off-temperature of 7 degrees i suggest to use mode(types) : "nofrost, night, day" for emsesp publish (hamode has fixed names)
- 1,2 is the actual temperature
- 4,5 remaining time in minutes for return to day function -> value decreases with time

> For example return to day time set to 8 hours:
day 04 27 00 C9 00 00 00 C9
night 02 1C 00 C9 01 E0 00 C9
000+00:05:01.454 N 9: [emsesp] Thermostat(0x17) -> Me(0x0B), RC10Monitor(0xB1), data: 02 1C 00 C9 01 E0 00 C9
000+00:07:01.526 N 12: [emsesp] Thermostat(0x17) -> Me(0x0B), RC10Monitor(0xB1), data: 02 1C 00 C9 01 DE 00 C9

**B0:**
- call system send "0B 17 B0 00 01" -> calibrate room temperature 0.1 = 01 , -0.1 = FF and -0,2 = FE (steps 0,1 with range -5 - 5 deg according to manual)
- call system send "0B 17 B0 01 00" -> set key backlight FF = on, 00 = off
- call system send "0B 17 B0 02 00" -> set hot water preparation 0 = on, 1 = off, 2 = auto 
- call system send "0B 17 B0 03 00" -> set night temperature
- call system send "0B 17 B0 04 00" -> set day temperature
- call system send "0B 17 B0 05 08" -> set night timer start to 8 hours (value between 1 and 99 according to manual)
- call system send "0B 17 B0 06 00" -> set building type valid options = 0,1,2 (valid options +1 map to options 1,2,3 according to manual)

**Unfortunately, it does not seem to be possible to set the mode remotely through RC10Set**

MichaelDvP, 2021-11-07:
Oh, you have been busy, great  👍  Some ideas to test:
Setting the timer only changes the value, or does it start night-mode on writing?
Is B1 writable, the modes looks like a bit field, maybe the high nibble is a mask (like 0x35 telegram on other thermostats). 
Try to write `call system send "0B 17 B1 00 22"` for nightmode, 44 for day.

For most of the values we have datastructure implemented, it's mostly copy&paste, would you like to add them, or should i do?

kpschaper, 2021-11-07:
> Setting the timer only changes the value, or does it start night-mode on writing?

It only changes the value, same is holds for set temperature during night mode

>Is B1 writable, the modes looks like a bit field, maybe the high nibble is a mask (like 0x35 telegram on other thermostats).
Try to write call system send "0B 17 B1 00 22" for nightmode, 44 for day.

Unfortunately, this does not seem to work either

> For most of the values we have datastructure implemented, it's mostly copy&paste, would you like to add them, or should i do?

I think i would need two or more iterations to get it right as i'm not yet familiar with this code.
Could you add the values?

MichaelDvP, 2021-11-09:
Please check values and namings for the new values in dev. 

One last idea to set the mode: Bosch sometimes use telegram fields for settings, that are not published on read, e.g. my boiler have circulation start in telegram 0x35 pos 1, but `read 8 35` returns a empty message. I think thats because it's only a button-push-function and does not change a state (writing 0x22 starts the circulation for 3 min, but writing 0x00 do nothing).
As the mode in your thermostat is also a button-push, check writing to B0 on pos 7 or 8, try with values 0x44 (day), 0x22 (night), 0x11 (off/nofrost), eg.  `call system send "0B 17 B0 07 44"`

kpschaper, 2021-11-09:
> Please check values and namings for the new values in dev.

Looking good, everything works as expected:
[image]
What is the reason that building, key backlight en ww mode are not part of hc1?
Because it are more general options not releated to the heating circuit?
If that's the case one could argue that building is part of hc1, because it tells something about the boiler size in relation to the heating circuit and therefore the expected heating rate when switched on.
I'm also not sure about the option naming for building: light, medium and heavy 

> One last idea to set the mode: Bosch sometimes use telegram fields for settings, that are not published on read, e.g. my boiler have circulation start in telegram 0x35 pos 1, but read 8 35 returns a [...]
As the mode in your thermostat is also a button-push, check writing to B0 on pos 7 or 8, try with values 0x44 (day), 0x22 (night), 0x11 (off/nofrost), eg. call system send "0B 17 B0 07 44"

Tried 4 options:
- call system send "0B 17 B0 07 11"
- call system send "0B 17 B0 07 01"
- call system send "0B 17 B0 08 11"
- call system send "0B 17 B0 08 01"
Non of them resulted in a mode change on the thermostat

I think that you're on the right track: mode change should be remote possible but we don't know what to send....

MichaelDvP, 2021-11-10:
> What is the reason that building, key backlight en ww mode are not part of hc1?

Thermostats with more than one hc have some global parameters like clock, building, internal sensor, etc. They do not depend on hc. Your thermostat have only a single telegram and a single hc, the parameters are mixed in this telegram.

The building parameter is normally only for outdoor temperature control and describes the damping (heat capacity) of the building, the values are defined by RC35 and RC300 thermostats. I dont know what is does in this room thermostat. If it's something different we should change name and the values(names/numerical). Suggestions?

> I think that you're on the right track: mode change should be remote possible but we don't know what to send....

Maybe there is another telgram for this, Try a bit with `read 17 B2` B3, B4, etc. if there is a response.

kpschaper, 2021-11-10:
> The building parameter is normally only for outdoor temperature control and describes the damping (heat capacity) of the building, the values are defined by RC35 and RC300 thermostats. I dont know [...]

The manual states the following: 
When the thermostat is set to automatic temprature control, the heating rate for your boiler / heating system can be adjusted to the house. Change the PID setting only when the set heating temprature is exceeded with a large amount during heating. 
The three options are: 
- 1 The heating system heats as fast as possible (factory setting)
- 2 The heating system heats slower
- 3 The heating system heats even slower

> Maybe there is another telgram for this, Try a bit with read 17 B2 B3, B4, etc. if there is a response.

**SUCCESS** I've tried to read B2 tm B8 and only B2 did not return <empty> but instead returned 00 00
Then I tried to write 01 to position 00 and the state changed to nofrost!!

- system send "0B 17 B2 00 04" -> Change to day
- system send "0B 17 B2 00 02" -> Change to night
- system send "0B 17 B2 00 01" -> Change to nofrost

MichaelDvP, 2021-11-11:
> Change the PID setting 

Should we name it `heating PID` with options `fast | medium | slow`?
> SUCCESS 

👍  I'll add the mode change. What does the second byte do? Maybe temporary temperature change without storing in night/day temperature?

kpschaper, 2021-11-11:
> Should we name it heating PID with options fast | medium | slow?

👍Sounds good!

> 👍 I'll add the mode change. What does the second byte do? Maybe temporary temperature change without storing in night/day temperature?

Good question, I don't think that the thermostat has a temporary temperature change option
I think this second byte might be used of the last button used to display info (forth button) 
This button seems to be unrelated to mode, showing info keeps the current mode active
I've tried writing 00, FF and 01 but did no see anything change

closed 2021-11-14 by kpschaper