emsesp/EMS-ESP32 #184: Read command always returns <empty> starting 3.3.0b1

kpschaper, 2021-11-07:
**Bug description**
Read command always returns \<empty\> starting from 3.3.0b1

**Steps to reproduce**

- Install 3.3.0b1
- Open telnet console -> su
- read thermostat b0 
> 000+00:01:45.932 E 1: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 80 B0 00 20
000+00:01:45.932 N 2: [emsesp] All(0x00) -> Me(0x0B), RC10Set(0xB0), data: \<empty\>

**Expected behavior**

- Install 3.3.0b0 (version still seems to work fine, not sure which commit after this version broke this functionallity)
- Open telnet console -> su
- read thermostat b0 
> 000+00:01:56.666 E 5: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 80 B0 00 20
000+00:02:01.194 N 6: [emsesp] Thermostat(0x17) -> Me(0x0B), RC10Set(0xB0), data: 02 FF 00 1C 28 08 00

**Additional remark**
Read function always returns "Last Tx Read operation failed after 3 retries. Ignoring request: 0B 80 B0 00 20" (version independent), seems strange behaviour or is this a acknoledge message?

proddy, 2021-11-07:
format is `read <deviceid> ...`. Can you try that?

Also I think read doesn't need admin privs in the Console so suggest we change that. It's just a harmless read command

MichaelDvP, 2021-11-07:
@kpschaper This is a bit complicated, try with `watch B0` before sending the read command. 
Your thermostat does not reply direct to a read request, it waits for the next poll (as we've seen with Tado), but emsesp publishes only the direct reply, and retries 3 times. After 3. try a empty reply is created inside emsesp (not on the bus), which fullfills the expected read.

With log debug you should see
- send request
- send request
- send request
- empty reply
- reply from thermostat

kpschaper, 2021-11-07:
> format is read <deviceid> .... Can you try that?

@proddy This does work and returns:

> ems-esp:# read 17 b0
000+01:13:33.744 N 53: [emsesp] Me(0x0B) <- Thermostat(0x17), RC10Set(0xB0), data: 20
000+01:13:33.801 N 54: [emsesp] Thermostat(0x17) -> Me(0x0B), RC10Set(0xB0), data: 02 FF 00 1C 28 08 00

@MichaelDvP It seems that the reply from the thermostat is missing.

> ems-esp:# watch b0
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xB0
ems-esp:# read thermostat b0
000+01:08:13.103 N 27: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:13.526 N 28: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:14.000 N 29: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:14.423 N 30: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:14.656 E 31: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 80 B0 00 20
000+01:08:14.656 N 32: [emsesp] All(0x00) -> Me(0x0B), RC10Set(0xB0), data: \<empty\>
ems-esp:# read thermostat b0
000+01:08:19.354 N 33: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:19.827 N 34: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:20.251 N 35: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:20.724 N 36: [emsesp] Me(0x0B) <- All(0x00), RC10Set(0xB0), data: 20
000+01:08:20.956 E 37: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 80 B0 00 20
000+01:08:20.956 N 38: [emsesp] All(0x00) -> Me(0x0B), RC10Set(0xB0), data: \<empty\>

MichaelDvP, 2021-11-07:
Oh, i was blindet by the B0, which had some issues with Tado and did not realize the "thermostat". If `read 17 xx`works, all is good. proddys eyes are better.

closed 2021-11-07 by kpschaper