emsesp/EMS-ESP32 #186: MQTT isn't connecting sometimes

splitbrain, 2021-11-07:
I'm having trouble making EMS-ESP to connect to my MQTT broker and have no idea how to debug it.

I'm using EMS-ESP v3.2.1 on ESP32 / v3.3.5-1-g85c43024c on a Gateway S32 (Standard Edition). 

For MQTT I am running mosquitto version 1.6.12 in Homeassistant OS' supervisor.

The device is running correctly and is reading the boiler values just fine. The WiFi connection seems to be fine, too - the webinterface is responsive and without problems.

But MQTT will just not connect. The Webinterface says Status: disconnected, Disconnect Reason: TCP disconnected.

My settings are as follows:

[image]

Now the thing is that it sometimes work. But at some point the connection is lost. I then start mucking around in the ESP webinterface (enabling disabling MQTT) and try restarting mosquitto and at some point it randomly starts working again for a while. But I still couldn't figure out when and why.

I am wondering if there are any ways I can figure out what's going on? Eg. is there any network diagnostics possible on the device itself? Eg. it would be interesting if the EMS "sees" the homeassistant/mosquitto instance at all - is there a ping-like command available?

Note: when the EMS can't connect to the MQTT server, I can still connect without problems using mqtt-explorer, so mosquitto is running in general.

Is there any logging I can provide that would be helpful? Anything I can do to provide additional info? Any other tips on debugging this?

proddy, 2021-11-07:
I just tried to re-create your environment, by installing Mosquitto broker in HA's Supervisor (version is 6.0.1) and using these options:
```
logins:
  - username: mqtt-ems
    password: mqtt-ems
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
```

and the same EMS-ESP settings (username/password, QOS 1, Retain Flag on). And it seems to work fine.

The only suggestion I would have right now is to upgrade to the latest Dev build. You can get the firmware from https://github.com/emsesp/EMS-ESP32/releases/tag/latest. It shouldn't make a difference but at least we're in sync. And secondly, you need to look at the Mosquitto logs in the HA - if the connection fails you will see the error and reason show up there.

splitbrain, 2021-11-08:
Thank you. I will try the devel release later. I am not really convinced that it is actually a bug in ems-esp but suspect something weird in my network. Especially since I see no failing connect attempts in the mosquito logs. So any network diagnostics (like a simple ping) that I could run from the ems-esp would be helpful.

proddy, 2021-11-08:
EMS-ESP does a constant ping to the MQTT. If the connection drops it will show the reason and try and reconnect. Try and correlate the EMS-ESP warnings you see in the WebUI's System Log against HA's Mosquitto logs. Also, bring it back to the bare minimum, i.e. remove the username/password authentication, switch back to the default QoS0 and definitely turn the Retain flag off.

MichaelDvP, 2021-11-08:
I think this is v3.2.1 and issue #99. It is fixed in the dev.

proddy, 2021-11-08:
hope so. We really need to get 3.3 onto the streets asap

MichaelDvP, 2021-11-08:
> We really need to get 3.3 onto the streets asap

There were a lot of enhancements and changes the last days, maybe update the main to 3.2.2 in a state of end oktober before the command/mqtt-refactor and tado enhancements. Then there is some time to test these new changes and add the solar additions and renamings (breaking changes) to the 3.3 version update.

proddy, 2021-11-08:
> There were a lot of enhancements and changes the last days, maybe update the main to 3.2.2 in a state of end oktober before the command/mqtt-refactor and tado enhancements. Then there is some time [...]

That sounds really complicated as there have been many small fixed in between the larger factorings. Maybe just wait a while and then do the 3.3 push and until then if people have issues and come to Discord/GitHub we'll recommend them to upgrade. BBQKees can start pre-loading the latest dev build on his gateways if he wants.

glitter-ball, 2021-11-08:
@splitbrain - does the log from the MQTT add-on in HA tell you anything? Supervisor, Dashboard, Mosquitto broker, log...

splitbrain, 2021-11-08:
I updated to v3.3.0b3 and MQTT immediately worked. Now I don't know yet if it's just a happy accident or if the problem really is fixed. I will give it a couple of days to see if it stays connected.

proddy, 2021-11-08:
Then I think it was the fix we made earlier as Michael mentioned. I'll close it for now. Feel free to re-open if it happens again.

closed 2023-02-19 by proddy