emsesp/EMS-ESP32 #187: Can't turn heating on/off on Cerapur Compact

erykwieliczko, 2021-11-08:
EMS-ESP: v3.3.0b3
Boiler: DeviceID: 0x08 ProductID: 95 Version: 43.10 
Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3

"heating activated" and "ww activated" (and I suspect all binary options) can't be set. Which means that turning the boiler on and off is impossible.
Numerical options can be set correctly.

[image]
[image]

proddy, 2021-11-08:
So via web which uses "on" and "off" it works but from the console it doesn't?

erykwieliczko, 2021-11-08:
`call boiler heatingactivated off` always resulted in `command not found`. I just assumed it's a problem with console command parsing or something...

And so, I tried only from web. I assume it ultimately does the same thing. The problem seems to be either somewhere in the protocol (boiler doesn't understand the command?) or boiler itself (it ignores on/off commands?).

As I said, numerical values work fine (ex. heating temperature is set correctly every time from web).

MichaelDvP, 2021-11-08:
Can you make a log from that (maybe a typo), with downloaded v3.3.0.b3.bin i have:
```
ems-esp32:# call boiler heatingactivated
{
  "name": "heatingactivated",
  "fullname": "heating activated",
  "value": "on",
  "type": "boolean",
  "writeable": true
}
ems-esp32:# watch 16
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x16
ems-esp32:# call boiler heatingactivated off
000+00:00:52.754 I 5: [command] Calling boiler command 'heatingactivated', value off, id is default
000+00:00:52.754 I 6: [boiler] Setting boiler heating off
000+00:00:52.803 N 7: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 00
000+00:00:53.078 N 8: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
000+00:00:53.131 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 5A 64 00 0A FA 0F 02 06 64 64 02 08 F8 0F 0F 0F 0F 1E 05 04 09 09 00 28 00 3C
000+00:00:53.358 N 10: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20 (offset 25)
000+00:00:53.396 N 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 3C (offset 25)
000+00:01:01.859 N 12: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
000+00:01:01.911 N 13: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 5A 64 00 0A FA 0F 02 06 64 64 02 08 F8 0F 0F 0F 0F 1E 05 04 09 09 00 28 00 3C
000+00:01:02.098 N 14: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20 (offset 25)
000+00:01:02.182 N 15: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 3C (offset 25)
000+00:01:02.911 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: 00 5A 64 00 0A FA 0F 02 06 64 64 02 08 F8 0F 0F 0F 0F 1E 05 04 09 09 00 28 00 3C
000+00:01:03.170 N 17: [emsesp] Controller(0x09) -> Boiler(0x08), UBAParameters(0x16), data: FF
000+00:01:13.293 N 18: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 5A 64 00 0A FA 0F 02 06 64 64 02 08 F8 0F 0F 0F 0F 1E 05 04 09 09 00 28 00 3C
```
The command works and boiler confirms, but the controller sets the command back to on after a few seconds in N17. In my system i can't set it from remote, but only because the MC10 does not allow it.

erykwieliczko, 2021-11-09:
[image]

As I already said, for some reason "call boiler heatingactivated" commands return "Command not found". This problem happens on both v3.3.0b3 and v3.2.1. Boiler is obviously connected and it's data can be read.
I tried both Putty and "telnet" command. So I doubt it's a terminal's fault.

The "successful" heatingactivated in above screenshot was made from the UI.

I can forward you telnet/http ports if that could help debug the problem.

proddy, 2021-11-09:
> For some reason "call boiler heatingactivated" commands return "Command not found". This problem happens on both v3.3.0b3 and v3.2.1. Boiler is obviously connected and it's data can be read.

You need to be admin to use the `call` command in the Console. So first do a `su`. See https://emsesp.github.io/docs/#/Console?id=root-level-commands. Ok, it could be clearer in the documentation. The reason is that telnet is not secure and you don't want someone hacking into your EMS-ESP and blowing up your boiler.

erykwieliczko, 2021-11-09:
```
ems-esp:# call boiler heatingactivated
{
  "name": "heatingactivated",
  "fullname": "heating activated",
  "value": 0,
  "type": "boolean",
  "writeable": true,
  "ha_entityname": "sensor.boiler_heating_activated"
}
ems-esp:# watch 16
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x16
ems-esp:# call boiler heatingactivated on
000+01:20:14.985 I 1: [command] Calling boiler command 'heatingactivated', value on, id is default
000+01:20:14.985 I 2: [boiler] Setting boiler heating on
000+01:20:15.097 N 3: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: FF
000+01:20:15.167 N 4: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: 00 00 64 00 00 FB 03 01 02 64 0A 02
000+01:20:15.441 N 5: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
000+01:20:15.477 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 00 64 00 00 FB 03 01 02 64 0A 02
000+01:21:00.371 N 7: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
000+01:21:00.405 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 00 64 00 00 FB 03 01 02 64 0A 02
000+01:22:00.285 N 9: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
000+01:22:00.323 N 10: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 00 64 00 00 FB 03 01 02 64 0A 02
000+01:23:00.325 N 11: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
000+01:23:00.361 N 12: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 00 64 00 00 FB 03 01 02 64 0A 02
000+01:24:00.365 N 13: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
000+01:24:00.401 N 14: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 00 64 00 00 FB 03 01 02 64 0A 02
```

proddy, 2021-11-10:
@MichaelDvP what do you make of this?

MichaelDvP, 2021-11-10:
Looks like the setting is ignored from boiler. It is the right telegram, right position and write is acknowlaged, but without effect. 
It's not a bug in emsesp, we can do nothing.

> If you can't set "heatingactivated" how do you enable/disable the heater? Is turning "selected flow temperature" to 0 enough?

Yes, my thermostat do so, this seems to be the way bosch handles it.

proddy, 2021-11-10:
right, that's what I thought. It'll be interesting if any of those official external Bosch devices can do this. But we don't have one (they are so expensive). I'm closing this and will note it down as a Known Issue

closed 2021-11-10 by proddy

erykwieliczko, 2021-12-02:
**If anybody experiences the same problem.**

It looks like just setting `boiler/selflowtemp` to 0°C every 10 seconds works well enough in practice.
The boiler turns on after a few minutes if you `selflowtemp` to 0°C... But repeating selflowtemp often keeps it turned off permanently.

proddy, 2021-12-02:
Thanks @erykwieliczko - I'll add to the troubleshooting guide