emsesp/EMS-ESP32 #190: Buderus RC35: datetime not updating

derretro, 2021-11-14:
I wanted to change the datetime value of the thermostat over web interface for my buderus RC35, but the value won't update...

[log.txt](https://github.com/emsesp/EMS-ESP32/files/7533391/log.txt)
.

`2021-11-14 09:01:24.065 D 23: [telegram] Tx read request to device 0x08 for type ID 0x19
2021-11-14 09:01:24.065 D 24: [telegram] Tx read request to device 0x08 for type ID 0x16
2021-11-14 09:01:24.065 D 25: [telegram] Tx read request to device 0x08 for type ID 0x33
2021-11-14 09:01:24.065 D 26: [emsesp] Fetching values for device ID 0x09
2021-11-14 09:01:24.065 D 27: [emsesp] Fetching values for device ID 0x10
2021-11-14 09:01:24.065 D 28: [telegram] Tx read request to device 0x10 for type ID 0x3E
2021-11-14 09:01:24.065 D 29: [telegram] Tx read request to device 0x10 for type ID 0x3D
2021-11-14 09:01:24.065 D 30: [telegram] Tx read request to device 0x10 for type ID 0x3F
2021-11-14 09:01:24.065 D 31: [telegram] Tx read request to device 0x10 for type ID 0xA5
2021-11-14 09:01:24.065 D 32: [telegram] Tx read request to device 0x10 for type ID 0x37
2021-11-14 09:01:24.136 D 33: [mqtt] Publishing topic ems-esp/boiler_data_ww (#285, retain=0, retry=1, size=461, pid=1)
2021-11-14 09:01:24.154 D 34: [telegram] Sending read Tx [#89], telegram: 0B 88 14 00 20
2021-11-14 09:01:24.218 D 35: [emsesp] Last Tx read successful
2021-11-14 09:01:24.237 D 36: [mqtt] Publishing topic ems-esp/thermostat_data (#286, retain=0, retry=1, size=702, pid=1)
2021-11-14 09:01:24.543 D 37: [mqtt] Publishing topic ems-esp/heartbeat (#287, retain=0, retry=1, size=208, pid=1)
2021-11-14 09:01:24.928 D 38: [telegram] Sending read Tx [#90], telegram: 0B 88 19 00 20
2021-11-14 09:01:25.046 D 39: [emsesp] Last Tx read successful
2021-11-14 09:01:25.163 D 40: [telegram] Sending read Tx [#91], telegram: 0B 88 16 00 20
2021-11-14 09:01:25.249 D 41: [emsesp] Last Tx read successful
2021-11-14 09:01:25.367 D 42: [telegram] Sending read Tx [#92], telegram: 0B 88 33 00 20
2021-11-14 09:01:25.433 D 43: [emsesp] Last Tx read successful
2021-11-14 09:01:25.547 D 44: [telegram] Sending read Tx [#93], telegram: 0B 90 3E 00 20
2021-11-14 09:01:25.628 D 45: [emsesp] Last Tx read successful
2021-11-14 09:01:25.748 D 46: [telegram] Sending read Tx [#94], telegram: 0B 90 3D 00 20
2021-11-14 09:01:25.908 D 47: [emsesp] Last Tx read successful
2021-11-14 09:01:26.017 D 48: [telegram] Sending read Tx [#98], telegram: 0B 90 3D 19 20
2021-11-14 09:01:26.132 D 49: [emsesp] Last Tx read successful
2021-11-14 09:01:26.249 D 50: [telegram] Sending read Tx [#95], telegram: 0B 90 3F 00 20
2021-11-14 09:01:26.407 D 51: [emsesp] Last Tx read successful
2021-11-14 09:01:27.135 D 52: [telegram] Sending read Tx [#99], telegram: 0B 90 3F 19 20
2021-11-14 09:01:27.290 D 53: [emsesp] Last Tx read successful
2021-11-14 09:01:27.925 D 54: [telegram] Sending read Tx [#100], telegram: 0B 90 3F 32 20
2021-11-14 09:01:28.040 D 55: [emsesp] Last Tx read successful
2021-11-14 09:01:28.156 D 56: [telegram] Sending read Tx [#101], telegram: 0B 90 3F 4B 20
2021-11-14 09:01:28.277 D 57: [emsesp] Last Tx read successful
2021-11-14 09:01:28.937 D 58: [telegram] Sending read Tx [#96], telegram: 0B 90 A5 00 20
2021-11-14 09:01:29.163 D 59: [emsesp] Last Tx read successful
2021-11-14 09:01:29.832 D 60: [telegram] Sending read Tx [#102], telegram: 0B 90 A5 19 20
2021-11-14 09:01:29.977 D 61: [emsesp] Last Tx read successful
2021-11-14 09:01:30.659 D 62: [telegram] Sending read Tx [#97], telegram: 0B 90 37 00 20
2021-11-14 09:01:30.741 D 63: [emsesp] Last Tx read successful
2021-11-14 09:01:39.059 I 64: [command] Calling thermostat command 'datetime', value 09:13:57 14/11/2021, id is default
2021-11-14 09:01:39.059 I 65: [thermostat] Date and time: 14.11.2021-09:13:57
2021-11-14 09:01:39.059 I 66: [thermostat] Setting date and time
2021-11-14 09:01:39.158 D 67: [telegram] Sending write Tx [#103], telegram: 0B 10 06 00 15 0B 09 0E 0D 39 48 7F
2021-11-14 09:01:39.330 D 68: [emsesp] Last Tx write successful
2021-11-14 09:01:39.330 D 69: [telegram] Sending post validate read, type ID 0x06 to dest 0x10
2021-11-14 09:01:40.349 D 70: [telegram] Sending read Tx [#104], telegram: 0B 90 06 00 20
2021-11-14 09:01:40.485 D 71: [emsesp] Last Tx read successful
2021-11-14 09:01:40.548 D 72: [mqtt] Publishing topic ems-esp/thermostat_data (#288, retain=0, retry=1, size=702, pid=1)
2021-11-14 09:01:54.035 D 73: [mqtt] Publishing topic ems-esp/boiler_data (#289, retain=0, retry=1, size=671, pid=1)
2021-11-14 09:01:54.137 D 74: [mqtt] Publishing topic ems-esp/boiler_data_ww (#290, retain=0, retry=1, size=461, pid=1)
2021-11-14 09:01:54.238 D 75: [mqtt] Publishing topic ems-esp/thermostat_data (#291, retain=0, retry=1, size=702, pid=1)
2021-11-14 09:02:24.066 D 76: [emsesp] Fetching values for device ID 0x08
2021-11-14 09:02:24.066 D 77: [telegram] Tx read request to device 0x08 for type ID 0x14
2021-11-14 09:02:24.066 D 77: [telegram] Tx read request to device 0x08 for type ID 0x14
2021-11-14 09:02:24.066 D 78: [telegram] Tx read request to device 0x08 for type ID 0x19
2021-11-14 09:02:24.066 D 79: [telegram] Tx read request to device 0x08 for type ID 0x16
2021-11-14 09:02:24.066 D 79: [telegram] Tx read request to device 0x08 for type ID 0x16
2021-11-14 09:02:24.066 D 80: [telegram] Tx read request to device 0x08 for type ID 0x33
2021-11-14 09:02:24.066 D 81: [emsesp] Fetching values for device ID 0x09
2021-11-14 09:02:24.066 D 82: [emsesp] Fetching values for device ID 0x10
2021-11-14 09:02:24.066 D 82: [emsesp] Fetching values for device ID 0x10
2021-11-14 09:02:24.066 D 83: [telegram] Tx read request to device 0x10 for type ID 0x3E
2021-11-14 09:02:24.066 D 83: [telegram] Tx read request to device 0x10 for type ID 0x3E
[... 154 lines cut ...]
2021-11-14 09:04:25.540 D 214: [telegram] Sending read Tx [#141], telegram: 0B 90 3E 00 20
2021-11-14 09:04:25.628 D 215: [emsesp] Last Tx read successful
2021-11-14 09:04:25.628 D 215: [emsesp] Last Tx read successful
2021-11-14 09:04:25.727 D 216: [telegram] Sending read Tx [#142], telegram: 0B 90 3D 00 20
2021-11-14 09:04:25.727 D 216: [telegram] Sending read Tx [#142], telegram: 0B 90 3D 00 20
2021-11-14 09:04:25.912 D 217: [emsesp] Last Tx read successful
2021-11-14 09:04:26.033 D 218: [telegram] Sending read Tx [#146], telegram: 0B 90 3D 19 20
2021-11-14 09:04:26.131 D 219: [emsesp] Last Tx read successful
2021-11-14 09:04:26.131 D 219: [emsesp] Last Tx read successful
2021-11-14 09:04:26.255 D 220: [telegram] Sending read Tx [#143], telegram: 0B 90 3F 00 20
2021-11-14 09:04:26.410 D 221: [emsesp] Last Tx read successful
2021-11-14 09:04:27.131 D 222: [telegram] Sending read Tx [#147], telegram: 0B 90 3F 19 20
2021-11-14 09:04:27.291 D 223: [emsesp] Last Tx read successful
2021-11-14 09:04:27.398 D 224: [telegram] Sending read Tx [#148], telegram: 0B 90 3F 32 20
2021-11-14 09:04:27.534 D 225: [emsesp] Last Tx read successful
2021-11-14 09:04:27.534 D 225: [emsesp] Last Tx read successful
2021-11-14 09:04:28.231 D 226: [telegram] Sending read Tx [#149], telegram: 0B 90 3F 4B 20
2021-11-14 09:04:28.403 D 227: [emsesp] Last Tx read successful
2021-11-14 09:04:28.521 D 228: [telegram] Sending read Tx [#144], telegram: 0B 90 A5 00 20
2021-11-14 09:04:28.663 D 229: [emsesp] Last Tx read successful
2021-11-14 09:04:28.776 D 230: [telegram] Sending read Tx [#150], telegram: 0B 90 A5 19 20
2021-11-14 09:04:28.854 D 231: [emsesp] Last Tx read successful
2021-11-14 09:04:28.967 D 232: [telegram] Sending read Tx [#145], telegram: 0B 90 37 00 20
2021-11-14 09:04:29.116 D 233: [emsesp] Last Tx read successful
`

proddy, 2021-11-14:
Hi @derretro,

From the log it looks like the command is sent correctly:

```
2021-11-14 09:04:03.420 I 181: [command] Calling thermostat command 'datetime', value 09:09:03 14/11/2021, id is default
2021-11-14 09:04:03.420 I 182: [thermostat] Date and time: 14.11.2021-09:09:03
2021-11-14 09:04:03.420 I 183: [thermostat] Setting date and time
2021-11-14 09:04:03.420 I 183: [thermostat] Setting date and time
2021-11-14 09:04:04.100 D 184: [telegram] Sending write Tx [#135], telegram: 0B 10 06 00 15 0B 09 0E 09 03 48 7F
2021-11-14 09:04:04.209 D 185: [emsesp] Last Tx write successful
```

I'm unable to validate it though as I only have an old RC20 but @MichaelDvP can check. 

Can you also post back the device information for your thermostat? (like ProductID)

BTW version 3.2.1 is quite old and we haven't had a chance to update it with the latest 3.3 release which has many new fixes and features. So worth upgrading by taking the firmware binary from the dev branch.

derretro, 2021-11-14:
DeviceID: 0x10, ProductID: 86, Version 01.11

I'll give the update a try immediately.

derretro, 2021-11-14:
Same behavior with v3.3.0b4 and also over telnet calls. I think this could be a "read only" issue.

proddy, 2021-11-14:
could be. I could never get it working on my RC20 either so it was disabled for those thermostats. Let's see what Michael comes back with

MichaelDvP, 2021-12-09:
The command have no syntax-checks, if something is wrong or missing the thermostat does not set the time. I expected the command is mostly used with `NTP` and nobody will type the complete date with day-of-week and daylightsaving information as needed for the thermostat. In this case exactly these two information are missing and the thermostat does not accept it.

proddy, 2021-12-09:
correct, as documented in https://emsesp.github.io/docs/#/Command?id=thermostats-thermostat

thanks Michael

MichaelDvP, 2021-12-09:
I'm using a template to remember the right format for free text inputs.
[image]

derretro, 2021-12-09:
Thanks Michael and proddy for your great help, I tried different inputs, but still can‘t get it to work. Could you give me a complete example String?

closed 2023-02-19 by proddy