emsesp/EMS-ESP32 #192: Heating does not work after night setback ??

zibous, 2021-11-14:
**Question**
In the morning (after the lowering), the temperature of 16 degrees is not changed over to the set operating temperature of 22 degrees. The heating temperature remains at 16 degrees, even if automatic operation is set.

Is it due to the settings or is it due to the current Homeassistant version or EMS-ESP32 version?

Another problem I have is that the "heating pump modulation" is usually very high (100%).

[image]

What can that be?

**Screenshots**
[image]

**Device information**

[image]

```
system
Uptime: 003+22:14:05.749
SDK version: v3.3.5-1-g85c43024c
CPU frequency: 240 MHz
Free heap: 163396 bytes

WiFi: Connected
SSID: ***************
BSSID: ***************
RSSI: -51 dBm (98 %)
MAC address: ***************
Hostname: ***************
IPv4 address: ***************
IPv4 gateway: ***************
IPv4 nameserver: ***************
```
*Copy-paste here the information as it is outputted by the device. You can get this information from http://ems-esp.local/api?device=system&cmd=info*

**Additional context**
```
Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)
  heating active: off
  warm water active: off
  selected flow temperature: 85 °C
  burner selected max power: 108 %
  heating pump modulation: 0 %
  outside temperature: 8.6 °C
  current flow temperature: 87.9 °C
  gas: off
  flame current: 0.0 uA
  heating pump: off
  fan: on
  ignition: off
  heating activated: on
  heating temperature: 60 °C
  burner pump max power: 100 %
  burner pump min power: 10 %
  pump delay: 5
  burner min period: 10
  burner min power: 0 %
  burner max power: 100 %
  hysteresis on temperature: -6 °C
  hysteresis off temperature: 6 °C
  burner current power: 0 %
  burner starts: 908 times
  total burner operating time: 10 days 9 hours 50 minutes
  total heat operating time: 7 days 7 hours 35 minutes
  total UBA operating time: 152 days 14 hours 23 minutes
  last error code: 0Y(276) 10.11.2021 03:20
  service code: 5H
  service code number: 268
  maintenance message:
  maintenance scheduled: date
  maintenance set time: 6000 hours
  maintenance set date: 07.07.2023
  ww selected temperature: 57
  ww set temperature: 57
  ww type: buffer
  ww comfort: eco
  ww flow temperature offset: 40
  ww max power: 100
  ww circulation pump available: off
  ww charging type: 3-way valve
  ww hysteresis on temperature: -5
  ww hysteresis off temperature: 0
  ww disinfection temperature: 70
  ww circulation pump frequency: 2x3min
  ww circulation active: off
  ww current intern temperature: 54.4
  ww current extern temperature: 54.4
  ww current tap water flow: 0.0
  ww storage extern temperature: 54.4
  ww activated: on
  ww one time charging: off
  ww disinfection: off
  ww charging: off
  ww recharging: off
  ww temperature ok: off
  ww active: off
  ww heating: on
  ww starts: 342
  ww active time: 3 days 2 hours 15 minutes

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)
  error code: (0)
  date/time: 10:25:12 14/11/2021
  floor drying: off
  damped outdoor temperature: 7.4 °C
  floor drying temperature: 0 °C
  building: light
  minimal external temperature: -10 °C
  ww set temperature: 60
  ww mode: own_prog
  ww set low temperature: 45
  ww circulation pump frequency: own_prog
  ww charge duration: 60
  ww charge: off
  ww circuit 1 extra: 0
  ww disinfection: on
  ww disinfection day: we
  ww disinfection time: 180
  hc1 selected room temperature: 26.0
  hc1 mode: auto
  hc1 mode type: comfort
  hc1 eco temperature: 15.0
  hc1 manual temperature: 24.0
  hc1 comfort temperature: 25.0
  hc1 summer temperature: 17
  hc1 design temperature: 45
  hc1 offset temperature: 0
  hc1 min flow temperature: 25
  hc1 max flow temperature: 60
  hc1 room influence: 0
  hc1 current room influence: 0.0
  hc1 nofrost temperature: 5
  hc1 target flow temperature: 41
  hc1 heating type: radiator
  hc1 set summer mode: auto
  hc1 summer mode: off
  hc1 control mode: optimized
  hc1 program: prog_1
  hc1 temporary room temperature automode: 24.0
  hc1 fast heatup: 0

Controller: BC25 (DeviceID:0x09, ProductID:125, Version:03.05)
```
Many thanks for the help.

proddy, 2021-11-14:
The heating question: are you saying the thermostat is not switching from 16 to 22 in the morning? I don't see how HA or EMS-ESP can affect that

The high heating pump modulation: I checked mine and I'm seeing the same behaviour so I guess it's normal? A lot of 100% (it's a combi heating/warm water boiler)

zibous, 2021-11-14:
@proddy 

Thank you for the information and checks 👍 

> The heating question: are you saying the thermostat is not switching from 16 to 22 in the morning? I don't see how HA or EMS-ESP can affect that

The setting to 16 degrees is not understandable for me, because I don't have that in any setting. Somehow that looks like a "default" setting. 

> The high heating pump modulation: I checked mine and I'm seeing the same behaviour so I guess it's normal? A lot of 100% (it's a combi heating/warm water boiler)

I just don't understand that the high heating pump modulation happens at night when the heating is in low temperature mode. In my opinion, it makes no sense, on the day when normal operation is set, the pump is working properly.

So I'm not sure whether my heating is set correctly?

Unfortunately, I cannot find any information about the optimal setting of the heating under the conditions when I operate 8 radiators with temperature controls.

proddy, 2021-11-14:
I don't understand the 16 degrees either. We would need more details to deep dive into what is happening. I'm not familiar with those thermostats - they have a lot of features and control modes.

The pump modulation looks strange if it's 100% in the middle of the night. It could be an error in reading. I checked mine and the pump mod does hit 100% but only when there is activity (heating on or someone having a shower). It's best to create a separate issue for this so we don't get mixed up between these two questions.

zibous, 2021-11-14:
@proddy 
> I don't understand the 16 degrees either. We would need more details to deep dive into what is happening. I'm not familiar with those thermostats - they have a lot of features and control modes.

The radiator thermostats do not control the heating, only the flow through the radiator.

The 16 degrees are displayed at "**climate.thermostat_hc1**" and when this happens, the heating mode is shut down. 

[image]

Doesn't always happen, but mostly after an update and restart of the home assistant. 

```
System Health

version: core-2021.11.3
installation_type: Home Assistant Core
dev: false
hassio: false
docker: false
user: homeassistant
virtualenv: true
python_version: 3.8.10
os_name: Linux
os_version: 5.8.0-63-generic
arch: x86_64
timezone: Europe/Vaduz

GitHub API: ok
Github API Calls Remaining: 5000
Installed Version: 1.16.0
Stage: running
Available Repositories: 980
Installed Repositories: 38

dashboards: 11
resources: 24
views: 18
mode: storage
```

> The pump modulation looks strange if it's 100% in the middle of the night. It could be an error in reading. I checked mine and the pump mod does hit 100% but only when there is activity (heating on [...]

Yes, this is better. I'll do a few more tests and then I'll create a separate issue.

zibous, 2021-11-14:
This was the trigger for the temperature change:
```
157902486	sensor	sensor.thermostat_hc1_selected_room_temperature	15	{"unit_of_measurement":"\u00b0C","friendly_name":"Thermostat hc1 selected room temperature","icon":"mdi:coolant-temperature"}	159399819	2021-11-13 22:02:52.212117	2021-11-13 22:02:52.212117	2021-11-13 22:02:52.212117	NULL	NULL	157686516
```
[image]

I just have to find out who did this.

glitter-ball, 2021-11-14:
I think I have the UK version (Sense II) of the same thermostat. It took me weeks to get the right settings and things like pump control were among the trickiest. I eventually found settings in mine that would shut the boiler and pump down overnight.

If your thermostat is located in the boiler, then the system would normally be set up for weather compensation based on outside temperature. If the thermostat is in one of the rooms, the system might be set up for weather compensation with room influence. In this case, the valve on the radiator in the same room should be set to maximum, so that it doesn't conflict with the thermostat.

The temperature settings can be done in two ways : basic levels (eco, comfort etc.) where the temperatures are fixed in the settings and the time programme jumps between them or individually-set temperatures, where these are set in the time programme. Might be worth checking the settings - my guess would that 'eco' is set to 15ºC somewhere.

The pump speed settings often depend on the boiler the thermostat is connected to.

zibous, 2021-11-14:
@glitter-ball
> my guess would that 'eco' is set to 15ºC somewhere.

 Thank you for the information. My conditions are:

  - Thermostat RC310 located in the boiler, settings: 25 °C
  - Temperature outside sensor (north side) for the set up for weather compensation.
  - 8 radiators with Thermostatic radiator Hama 00176593  that only regulate the flow rate depending on the set temperature
  - time programm hc1 program prog_1 06:30 - 23:00 22 ° C, 25:00 - 06:30 22:00 ° C 
  - hc1 max flow temperature: 60 °C
 
If I change the room temperature on the RC310 only slightly, e.g. b to 22.00 ° C or 25.00 ° C, then there is a strong increase in boiler heating pump modulation, although I cannot say whether this is also influenced by the external temperature.

In some cases, the hc1 selected room temperature thermostat is set to 15 ° C.
Then in the morning it no longer heats up. 

I'm now trying to figure out which parameters I need to change to make this more stable.   
Not an easy task with the many setting options, so I understand that it took you some time.

zibous, 2021-11-16:
Mea culpa. 
Found the bug. It was due to the setting "own_prog1", there the heating mode was set for the days **Mon-Fri** and therefore the economy mode was set on Saturday and Sunday. 
Thanks

glitter-ball, 2021-11-16:
Hurrah! I knew you'd find it buried in there somewhere. There are a *lot* of settings in that thermostat.

closed 2023-02-19 by proddy