emsesp/EMS-ESP32 #198: 3-way valve - unexpected behaviour

Karel311, 2021-11-16:
System: 
ESP-32 Running latest firmware (v3.3.0 b4)
FW200 with Junkers HT3, IPM module (2 heat circuits), warm water via buffer tank connected via 3-wau valve tot heating circuit.

Problem description
- parameter "ww heating" is ON whenever "heating active" is OFF)
--> The 3-way valve is then closing off the heating circuit (=no power to the valve) and opens the boiler tank circuit, even though no water is heated (Circulation pump  and "warm water active" stay OFF)
- But this speeds up heat loss from the boiler tank (normally 0,5C per hour , now >1C  per hour) even though the water inside the tank is heated above the threshold.

Expected behaviour: 
3-way valve only closes when the boiler tank water is heated (this is how it worked till last week).

Tried to disable the  circulation program and ww circulation pump on the FW200.

Don't really know whether this can be related to a firmware issue since my last upgrade to 3.3.0b4  or caused by some minor tweaks in the FW200 and HT3 temperature settings

**Device information**
(1) Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.08)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A

(3) Thermostat: Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.12)
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168 

(4) Mixer: Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.07)
 This Mixer will respond to telegram type IDs: 0x10C 0x11E

(5) Mixer: Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.07)
 This Mixer will respond to telegram type IDs: 0x10C 0x11E

(2) Controller: HT3 (DeviceID:0x09, ProductID:95, Version:18.08)

proddy, 2021-11-17:
this is beyond me...sorry. @BBQKees any ideas?

FredericMa, 2021-11-19:
I also have a FW200 and I have the same issue but I never saw it working correctly. Even before I was using ems-esp, the boiler would never reset the 3-way valve to the heating circuits when the water was heated. UNLESS heating for the floor heating or radiators is demanded!

So how I tackled this is to check if warm water heating has been completed but the 3-way valve is still open towards the water tank. If so, I fire up the radiators heating circuit and turn it off from the moment the 3-way valve is set towards the heating circuits. This is how I'm doing it since the last 5 years.

I guess this is just how the thermostat works. (He doesn't care what the position of the 3-way valve is when his job is done.)

So maybe you notice this now because your heating hasn't been active anymore since then?

Karel311, 2021-11-19:
You are probably right.

May I ask how you did this automation? (home assistant?)

Would be an enhancement for us with an option in the EMS-ESP that could control the 3-way valve position after warm water is heated (Close --> open  after warm water is heated)

FredericMa, 2021-11-19:
The automation is done via AppDaemon since I use more complex logic for additional stuff on the heating circuits but essentially this can also be done in an automation in Home Assistant. 

I'll outline a bit how this automation would look like in a HA automation:
Trigger:
- MQTT sensor: `wwcharging` from the `ems-esp/boiler_data_ww` topic is `OFF` for 10 minutes (if heating is required the switch will probably be made automatically within 10 minutes). See https://www.home-assistant.io/docs/automation/trigger/#holding-a-state

Condition:
- MQTT sensor: `wwheat` from the `ems-esp/boiler_data_ww` is `ON`. This indicates the 3-way valve is open towards the tank. @proddy, now I think about it, `wwheat` it is rather a abstract description for a valve and as in this case it also doesn't mean that any heating is still going on. Maybe something like `ww3waytobuffer` or `wwvalvetobuffer` is more clear?

Action:
- MQTT publish on topic `ems-esp/thermostat` with payload `'{"cmd":"mode" ,"data":"heat", "id":1}'`. This turn on the circuit 1 which are my radiators.
- MQTT publish on topic `ems-esp/thermostat` with payload `'{"cmd":"heattemp" ,"data":28, "id":1}'`. Put it high enough so you are sure the boiler will fire up. If your setpoint is below the actual temperature, nothing will happen.
- Wait_template that waits for `wwheat` to got to `OFF` state. I would suggest to add a timeout of 1 minute for example. In my system the valve resets within 10 seconds most of the time. See https://www.home-assistant.io/docs/scripts#wait
- MQTT publish on topic `ems-esp/thermostat` with payload `'{"cmd":"mode" ,"data":"nofrost", "id":1}'. This turns of the radiators circuit again.
- Additionally you can put the setpoint back to the temperature you want. (Payload of action 2).

Do keep in mind that you need to make sure you don't turn off your heating circuit by accident if you actually still need heat.

glitter-ball, 2021-11-19:
I don't think this unusual behaviour. Here's my system...
[image]
I have a Worcester-Bosch Greenstar 24i System with Sense II (CW400). I didn't get the diverter valve fitted in the boiler when it was installed (it was another £500...) so use a diverter valve motor to operate micro-switches which drive a fast (9-second stroke) external Belimo diverter valve. This works pretty well. Like you, the default position for my valve is 'HW' - that's just the way the boiler/thermostat works. The manual backs this up...
[image]

If the tank is losing heat by passive convective circulation, is the tank situated higher/lower than the boiler? That would cause this problem to be worse. You might be able to add a spring-loaded valve to close the circuit until the valve changes to 'HW' and the pump starts. Or, do something clever in EMS-ESP!

You might find a setting called "DHW priority" in your boiler/thermostat...
[image]
It might be worth changing this to see if this affects behaviour of the valve.

FredericMa, 2021-11-19:
I have a Junkers CerapurSmart A TOP 14-3CE. The boiler will also always give priority to hot water instead of heating circuits but only switches the valve to the position it needed for its last action. So if it was making warm water and no heat for the heating circuits is requested, the valve remains towards the buffer. If it was heating for the circuits and the buffer its temperature is still ok, it remains in position to the heating circuits. Only when the buffer its temperature is too low and also heat is requested for the circuits, this priority makes sure the buffer get heated first.

The problem is that the heater its pump is started once every hour in case it hasn't been started for the last hour (to prevent stalling if not started for a long time). So what happens (especially during summer when the heating circuits are off) is that the buffer gets heated and afterwards the valve remains in position towards the buffer. So every time the pump starts, it starts circulating water from the hot buffer through the cooled down circuit towards the boiler. On the other hand the cold water in the pipes goes also back in the buffer. In the end, when the pump turns back off, we have colder water in the buffer and hot water in the pipes towards the boiler (although insulated, but still unnecessary loss of heat). This cycle repeats every hour.
I checked it a few years ago and this unwanted circulation towards the buffer lowers the temperature of the whole buffer of 300 liter by an extra 4 degrees over one night compared to when the pump isn't circulating through the buffer.

glitter-ball, 2021-11-20:
Yes, it's a pain and I don't know why it has to exercise the pump every hour. Once every 24 hours (if it hasn't run anyway) would be plenty.

The other option in all this might be the Test menu. There's an option in my boiler and my thermostat to drive the 3-way valve to the HW position or the CH position for testing. I don't know how long it stays there when test ends, though. Maybe you could send a test telegram every 10 minutes?!

closed 2023-02-19 by proddy