emsesp/EMS-ESP32 #200: Buderus GB162 installation not detected properly

elektronicanet, 2021-11-16:
Using EMS-ESP v3.2.1, my Buderus GB162 is not detected properly. I have 3 HC (2x radiators, 1x floor) with RC35 + RC20 + RC20 thermostates. 

I noticed the RC20 thermostates have id 0x18 instead of 0x17. Could that be a problem? Only one is detected (I have 2) but instead I have an unknown device 0x1A.

The installation is working fine, and I am able to see the RC35 and the boiler info. But I can't see any RC20 info.

**Device information**
{
  "System": {
    "version": "3.2.1",
    "uptime": "000+20:38:49.844",
    "uptime_sec": 74329,
    "freemem": 165
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 136344,
    "read requests sent": 38429,
    "write requests sent": 0,
    "incomplete telegrams": 2,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "dallas sensors": 0,
    "dallas reads": 0,
    "dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus Topline/GB162 (DeviceID:0x08 ProductID:115, Version:03.06)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "ES72/RC20 (DeviceID:0x18, ProductID:113, Version:02.03)",
      "handlers": "0xA3 0x06 0xA2 0x12 0xAE 0xAD"
    },
    {
      "type": "Thermostat",
      "name": "RC35 (DeviceID:0x10, ProductID:86, Version:21.08)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x3E 0x3D 0x3F 0x48 0x47 0x49 0x52 0x51 0x53 0x5C 0x5B 0x5D 0xA5 0x37"
    },
    {
      "type": "Mixer",
      "name": "MM10 (DeviceID:0x22, ProductID:69, Version:02.01)",
      "handlers": "0xAA 0xAB 0xAC"
    },
    {
      "type": "Mixer",
      "name": "MM10 (DeviceID:0x21, ProductID:69, Version:02.01)",
      "handlers": "0xAA 0xAB 0xAC"
    },
    {
      "type": "Switch",
      "name": "WM10 (DeviceID:0x11, ProductID:71, Version:02.00)",
      "handlers": "0x9C 0x9D 0x1E"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)",
      "handlers": ""
    },
    {
      "type": "Unknown",
      "name": "unknown (DeviceID:0x1A, ProductID:151, Version:04.01)",
      "handlers": ""
    }
  ]
}

proddy, 2021-11-17:
The RC25 is productID 151, what's strange is why it's not been detected as a thermostat. I'll need to look into that.
Apart from that, is everything else working ok?

elektronicanet, 2021-11-17:
Ah, so one of the thermostats is RC25 and one is RC20. I was not aware of that, but they look the same, so that must be the case then 👍  

In general it seems to work fine, I can change parameters and they actually update, and the values seem realistic. I did get some warnings:

```
000+21:10:04.569    INFO 174:    [command]    Calling thermostat command 'hc1/tempautotemp', value 19, id is default
000+21:10:04.569    INFO 175:    [thermostat] Setting thermostat temperature to 19.0 for heating circuit 1, mode tempauto
000+21:20:50.498    INFO 176:    [command]    Calling boiler command 'heatingtemp', value 65, id is default
000+21:20:50.498    INFO 177:    [boiler]     Setting boiler heating temperature to 65 C
000+21:53:04.937 WARNING 178:    [telegram]   Incomplete Rx: F8 08 00 18 00 52 03 09 64 00 00 01 E4 62 80 00 02 5E 02 D9 00 01 13 30 41 01 31 00 01 00
001+00:19:39.644    INFO 179:    [command]    Calling thermostat command 'hc1/seltemp', value 15, id is default
001+00:19:39.644    INFO 180:    [thermostat] Setting thermostat temperature to 15.0 for heating circuit 1, mode auto
001+15:51:00.249    INFO 181:    [command]    Calling thermostat command 'hc2/mode', value auto, id is default
001+15:51:00.249    INFO 182:    [thermostat] Setting thermostat mode to auto for heating circuit 2
001+15:51:27.266    INFO 183:    [command]    Calling thermostat command 'hc2/daytemp', value 19, id is default
001+15:51:27.266    INFO 184:    [thermostat] Setting thermostat temperature to 19.0 for heating circuit 2, mode day
001+21:27:29.354 WARNING 185:    [telegram]   Incomplete Rx: E0 08 00 34 00 3C 02 6C 02 6C 21 00 05 03 00 04 D1 01 00 9E E4 00
```

But besides the RC20/RC25, I'm also missing all hc3 parameters in the RC35, so I'm not able to control the radiators in HC 3. I only see the radiators in HC 1 and the floor heating in HC 2.

proddy, 2021-11-17:
do you see the hc3 params in the WebUI? Also in the Telnet Console if you type `show` ?

proddy, 2021-11-17:
I made a quick change. Can you upload 3.3.0b8 from https://github.com/proddy/EMS-ESP32/releases/download/latest/EMS-ESP-3_3_0b8-ESP32.bin and see if your RC25 pops up?

elektronicanet, 2021-11-18:
> do you see the hc3 params in the WebUI? Also in the Telnet Console if you type `show` ?
I don't see the hc3 params in the WebUI. But I do see them in the Telnet Console.

elektronicanet, 2021-11-18:
With the new firmware, the RC25 is indeed recognized, and I see hc1, hc2 and hc3 in the WebUI. So it seems solved!

proddy, 2021-11-18:
that's great. I'll close this issue. If you notice any other peculiarities please re-open or create a new issue. Don't forget to like this project!

closed 2021-11-18 by proddy