emsesp/EMS-ESP32 #202: MQTT: can we have network status messages (ip address, etc)

g6094199, 2021-11-18:
Can we have IP (v4/v6) address send via MQTT  plz?
Also if its connected via ethernet or wifi (eg. important for w32-eth devices).

sometimes Gateway and DNS is also usefull for fast debugging (mostly on dhcp).

messages for SSID and MAC address will be a goodie on top.

proddy, 2021-11-18:
good idea. So add the IPv6 IP, MAC and SSID (if enabled) to the API output of http://ems-esp.local/api/system/settings  

There is a topic called `ems-esp/info` which is sent when EMS-ESP starts with the version and IP address, e.g. `{"event":"start","version":"3.3.0b7","ip":"192.168.1.134"}` . I'll add the MAC and SSID too. Does that work for you?

g6094199, 2021-11-18:
👍

> good idea. So add the IPv6 IP, MAC and SSID (if enabled) to the API output of http://ems-esp.local/api/system/settings
 I'll add the MAC and SSID too. Does that work for you?

maybe the bssid too  ;-)

proddy, 2021-11-18:
ok. A bit like the command `show system` from the console does.

g6094199, 2021-11-18:
in general, yes

proddy, 2021-11-18:
ok. I meant the URL http://ems-esp.local/api/system (the info command) not settings. I'll add it shortly.

proddy, 2021-11-18:
pls check 3.3.0b8

g6094199, 2021-11-18:
whow, that was quick!
but i cant see any new mqtt messages coming in.

[image]

proddy, 2021-11-18:
it's in the MQTT topic called `info`, not `heartbeat`. The heartbeat is used for dynamically changing data. It's a waste to send the Network details every 60 seconds if it doesn't change, and for Home Assistant to create sensors from them. If you want the information it's best to use the API and do a call to http://ems-esp.local/api/system. If you want to see this information in HA then manually create a sensor:

```yaml
sensor:
  - platform: rest
    resource: http://ems-esp.local/api/system/info
    name: emsesp_MAC
    value_template: "{{ value_json.Network.MAC }}"
```

closed 2021-12-02 by proddy