emsesp/EMS-ESP32 #204: "last error code" is not retrieved

pswid, 2021-11-18:
I'm currently using the latest development build v3.3.0b7 with Buderus WSW196i heat pump. It work's fine (although it shows a lot of parameters not used by heat pumps, so it would be worth giving the possibility to block/hide irrelevant parameters in the future versions, don't you think?). 

My heat pump currently (almost) all the time reports errors A01-5503 and H01-5504. However, EMS-ESP hardly ever shows "last error code". Restarting the EMS-ESP also deosn't help. I wonder, why? Maybe EMS-ESP should (at least at startup) "ask" the boiler for last error info? Maybe even more than only one, last error (on the heat pump display I can see the last 20 errors)?

Besides, in case the error is displayed on dashboard, there is also "uptime", like this:
**last error code          H01(5504) uptime:724749 min**
What is this uptime? Shouldn't it rather be decoded as the date and time the error occurred as it KM200 gateway does? In my case, response from KM200 looks e.g. like this:
**/notifications/Error-01          Time: 2021-11-16T14:57:34 -ErrorCode: A01 -AddCode: 5503 -Category: N
/notifications/Error-02          Time: 2021-11-16T14:57:34 -ErrorCode: H01 -AddCode: 5504 -Category: N**

If it helps I could make some tests if I only knew which EMS telegrams are related to error codes and what to look at...

tp1de, 2021-11-18:
A01/5503 indicates a problem with CAN-bus cable (disconnected or loose connection or to narrow to power cable)
H01/5504 indicates a problem with heat pump ventilation (filter to be cleaned or ventilation defect)

Most probably these are not permanent errors and overwritten afterwards. Within RC310 the last errors are stored locally while ems-esp polls the bus and shows the last status. I had a similar issue and RC310 error only disappeared on new start of heating system.

Within RC310 the date/time of first startup of the system is stored. This date + uptime should indicate the date-time of the error. But first startup is not available within km200 and as as I know neither in ems-esp.

Maybe Michael has another idea ...

pswid, 2021-11-19:
Errors are permanent and shows up after first compressor start since applying power to the HP.
In my case "Total UBA operating time" shows 699 days 7 hours and 5 min (=1 006 985 minutes). It seems to be total working time of the heat pump (it has been installed almost two years ago). An H01 error appeared few days ago, so time offset since first start should be only a few thousand less, shouldn't it? But "error uptime" shown is only 724496 min. So, data from 0xC2 telegrams seems to be incorrectly interpreted.
BTW, if I send "read 0x08 0xC2" in console I'm getting dozens(!) of telegrams in answer and as I see they contain many sequences 0x48 0x30 0x31 0x15 0x80 (=H01 in ASCII + 5504) and 0x41 0x30 0x31 0x15 0x7F (=A01 in ASCII + 5503). Maybe it's all 20 errors stored in controller memory? So, why not to decode all of them?

pswid, 2021-11-22:
WSW196i has HMC300 controller (firmware NF13.07) built-in. It also stores "installation date" in the memory. My controller shows 14.12.2019. So, it was approximately 1,012,000 not 724,749 minutes from installation to error (16.11.2021). Are you sure that it is an offset (in minutes)?

proddy, 2021-11-22:
I'm not sure. Take a look at the code and if the calculation is wrong, try and fix it!

https://github.com/emsesp/EMS-ESP32/blob/6532abd870cddeb616f0c1bb60cebd897eae71b6/src/devices/boiler.cpp#L829

pswid, 2021-11-22:
in my case, begining of response to **read 0x08 0xC2** console command is:

08 AC 00 10 31 48 30 31 15 80 95 0B 0E 10 38 00 7F FF FF FF  _**2021-11-16T14:57:34 -ErrorCode: H01 -AddCode: 5504 -Category: N**_
08 AC 00 10 09 41 30 31 15 7F 95 0B 0E 10 2A 00 7F FF FF FF  _**2021-11-16T14:57:34 -ErrorCode: A01 -AddCode: 5503 -Category: N**_
08 AC 00 0E 31 41 30 31 14 E3 95 0B 0D 10 26 95 0B 0E 10 2A
08 AC 00 10 31 48 30 31 15 80 95 0B 0F 0D 03 95 0B 0E 10 2A
08 AC 00 10 09 41 30 31 15 7F 95 0B 0E 0D 30 95 0B 0E 10 2A
08 AC 00 0E 31 41 30 31 14 E3 95 0B 0D 0D 1E 95 0B 0E 0D 30
08 AC 00 10 31 48 30 31 15 80 95 0B 0F 0B 2A 95 0B 0E 0D 30
08 AC 00 10 09 41 30 31 15 7F 95 0B 0F 0B 1B 95 0B 0E 0D 30
08 AC 00 0E 31 41 30 31 14 E3 95 0B 0F 0B 08 95 0B 0F 0B 1B
08 AC 00 10 31 48 30 31 15 80 95 0B 0C 0B 2B 95 0B 0F 0B 1B
08 AC 00 10 09 41 30 31 15 7F 95 0B 0C 0B 1C 95 0B 0F 0B 1B
08 AC 00 0E 31 41 30 31 14 E3 95 0B 0C 0B 18 95 0B 0C 0B 1C
08 AC 00 10 31 48 30 31 15 80 95 0B 0F 0A 24 95 0B 0C 0B 1C

Error code and number is decoded correctly (why only the first error?). Following bytes seems to contain info about start and end of the error. As you see first two rows have "empty" info about end because those errors are still valid (not ended). I can't figure out how to decode start of the first two errors to date stamps **2021-11-16T14:57:34.** Maybe you?

BTW, it seems that command "read 0x08 0xC2" doesn't work correctly. First two bytes in each row are equal to last two bytes from previous row. So offset or telegram length is set incorrectly. It also causes the same data to be read multiple times. See attachment.
[read 0x08 0xc2.txt](https://github.com/emsesp/EMS-ESP32/files/7581120/read.0x08.0xc2.txt)
.

proddy, 2021-11-22:
couple of initial thinkings
- the `read` command is read `<device ID> <type ID> [offset]`. Can you double check its actually working
- in the code for some reason any offset > 0 is ignored, which is why it's not working for you. I don't know why. I didn't write that function and there are no comments (grrr). Just remove those lines and try again. Code is here: https://github.com/emsesp/EMS-ESP32/blob/6532abd870cddeb616f0c1bb60cebd897eae71b6/src/devices/boiler.cpp#L818
- The timestamp is in the 3 bytes, so in your first examples it's 0x0B0E10 = 724496. If it was a date it would have 5 bytes and not 3, so it's correct to assume its an offset of something?
- There is also lastcode error warning from the Thermostat in telegram 0x12, if your thermostat supports it

pswid, 2021-11-23:
Unfortunately, deleting the condition (which has the effect that only the first 0xC2 response is being processed) does not cause all errors to be displayed.
If you look closer at the 0xC2 response, you'll see that there are two 5-bytes sections at the end of each error data. They are in fact dates coded in similar way to 0x10 and 0x11 messages. To be consistent with the controller indication, only minutes must be increased by 1. 
To display long error info, I've doubled size of _lastCode__ (boiler.h) and modified the code (boiler.cpp) as in attachment. Now last error is displayed in the same form as on controller's LCD (it seems that new error appeared this morning):

[image]

[process_UBAErrorMessage2.txt](https://github.com/emsesp/EMS-ESP32/files/7586569/process_UBAErrorMessage2.txt)

Could you implement something like this?

proddy, 2021-11-23:
nice change, well done. I can make the commit but why don't you try and do a Pull Request? I'm really in need for another contributor

pswid, 2021-11-23:
I'm just beginner in C++ programming, so the ems-esp looks very complicated to me. If I have free time and I may try to fix simple errors, sometimes. I'm also complete novice on github. I've just created my first "pull request". Is it ok?

proddy, 2021-11-24:
your first PR, yippy! I'll check later...

closed 2023-02-19 by proddy