emsesp/EMS-ESP32 #205: Supporting multiple 'smart' thermostats

axello, 2021-11-18:
I looked into the last year of old and new github issues and could not find a reference to this particular one.

My system is a `Nefit GBx72/Trendline/Cerapur/Greenstar Si/27i` with a BC-10 controller and I'm using a Bosch EasyControl (CT200) with Bosch Smart Thermostats in several rooms. Also I'm using 
For EMS I'm using BBQ-Kees ESP32 controller, which is very reliable by the way!
```
EMS-ESP Version
v3.2.2b13
Device (Platform / SDK)
ESP32 / v3.3.5-1-g85c43024c
```
**The solution I would like**
Currently the MQTT packets show the set temperature for the main thermostat, although that also does not seem to work (see below). However, I'm using the 'weather dependent control', which allows different temperatures *per room*. I have looked in the MQTT packets, in particular 
`{"id":203,"datetime":"23:41:44 18/11/2021","hc1":{"seltemp":0,"currtemp":0}}`
but have not been able to determine which setting show the set room temperatures.

In the picture you see the 4 different temperatures set in the thermostat: the main one at 16.5, the other ones at 15.0 and the 'Zolder' one at 19.0°C. This means the whole house is bloody cold, but my study is heated.
[image]

**Feature Request**
It would be lovely if somehow these extra settings, which are known by the thermostat (?), are in some way interceptable and could be delivered over MQTT.
I have attached the MQTT logs of the normal operation and the _weather dependent_ operation.
[EMS-ESP MQTT.txt](https://github.com/emsesp/EMS-ESP32/files/7566519/EMS-ESP.MQTT.txt)

If there is anything I can do to help the project along, by enabling logs or installing new firmware, please let me know!

proddy, 2021-11-19:
Those newer online thermostats (the Easy range) don't use EMS to communicate to the master/boiler so it's tricky to tap in and pull the data. There are however quite a few GitHub projects that use Nefit's web service to sniff out the information and also write back. 

EMS-ESP currently only listens to the telegram 0x0A because apparently, that's the only one we've been able to find so far. It has the current setpoint temperature and the current room temperature. It may also have the other zones somewhere in that telegram but I would need to see it. Can you do a `watch 0a` via the Console and send back some telegrams? And anything else you see coming back on the EMS line from your CT200 would be interesting to dive into

axello, 2021-11-19:
Oh, that's too bad, that it does not use the same protocol. 
I didn't know telnet was open! That seems like a security nightmare 😳
Here's the data dump.
[watchdump_0x0A.txt](https://github.com/emsesp/EMS-ESP32/files/7569957/watchdump_0x0A.txt)

As you can see: all 0a packets are null.
I did see some hopeful 0a packets, where I set the temperature to '20', but when I set other thermostats to 16 and 15, nothing showed up 👎 

I'll scavenge github to look for nefit webscraping projects!

proddy, 2021-11-19:
Telnet is read-only and password protected for any of the scary stuff. But it may be a good idea to have this optional (enable/disable via the WebUI).

For other projects look at https://emsesp.github.io/docs/#/Supported-EMS-Devices?id=thermostats

closed 2022-01-26 by proddy