emsesp/EMS-ESP32 #206: disabling irrelevant values

pswid, 2021-11-19:
I'm using Buderus WSW196i heat pump. As I see, some reported values ​​are irrelevant (e.g. those related to gas/oil burners). Maybe it is worth considering to add possibility to disable parameters by user? The best way is probably to add a checkbox to each value (in one column) in the dashboard. Additionally the "Hide disabled"/"Show disabled" button (to see disabled items and be able to enable them again). Disabled parameters should be "processed" normally but not sent via mqtt. It would minimize mqtt traffic and make dashboard much more clear. What do you think?
In my case, the parameters that could be hidden (because they are always 0/off):
**heating active off
warm water active off
system pressure 0 bar
exhaust temperature 0.0 ° C
gas off
flame current 0 uA
fan off
ignition off
burner starts 0 times
total burner operating time 0 minutes
total heat operating time 0 minutes
ww starts 0 times
ww active time 0 minutes**

Additionally I'd like to disable "working" but not used parameters, e.g.:
**burner current power** (there is no "burner" in heat pump and it shows value is identical to "heating pump 2 modulation")
all values ​​related to cooling (Buderus WSW196i doesn't support cooling)
all values ​​related to pool (this is an option I don't have)

proddy, 2021-11-19:
It's a great feature to add, to be able to enable/disable what information is sent via the API or MQTT. It's been on my todo list for longer than 1 year. I'll give it some thought and outline a suggestion as it's quite a bit a work to retain a list of 'disabled' entities due to its dynamic nature.

By design EMS-ESP checks each device value (or entity as we call it now) to see if it's valid or not. Otherwise it will be excluded everywhere (MQTT payload, Telnet Console, API and WebUI). I'd like to start with looking at the ones you listed above and figuring out why these still appear. Could you send me a couple of telegrams to examine?

- 0x18 and 0xE4  (for flame current, gas on/off, sys pressure, fan on/off an ign on/off)
- 0x19 and 0xE5  (for all the operating times, exhaust temp, and sometimes ign on/off, fan on/off)
- 0x34 (for ww starts and active time)

Use the console and type `watch <id>` and grab 2-3 telegrams.

g6094199, 2021-11-20:
> I'm using Buderus WSW196i heat pump. As I see, some reported values ​​are irrelevant (e.g. those related to gas/oil burners). Maybe it is worth considering to add possibility to disable parameters [...]

just to add...there a doubles, too, that are not needed to send.

e.g im my case:
 
 ww current intern temperature 43,9 °C
 ww current extern temperature 43,9 °C

which call internally the same sensor.

glitter-ball, 2021-11-20:
This is a great idea, if it can be implemented somehow without being a coding nightmare!

In my head I see this page, with a tick box on each line for "Enabled". Everything should still display in this dialog, because that becomes the one place where you can see what's available and switch it on or off.

[image]

I guess you have to decide how you treat new points as they're added in releases? Maybe they default to enabled so you can see they're present and then pop in and disable them?

PS Could alternate lines in these lists be shaded differently to aid reading across and/or the point name's column width minimised? Resizing the browser window does this but... *8)

proddy, 2021-11-20:
To keep the configuration separate from the dashboard showing and editing values I was thinking of adding this feature in an additional tab under Settings. And then basically a list of device entities (shortname, fullname) with a show/hide toggle next to it and a big Save button. These would be stored in Flash on the LittleFS file system. How does that sound?

glitter-ball, 2021-11-20:
That would work. Could you show the current value in the dialog too? That helps sometimes with the enable/disable decision!

Is the alternate line shading viable somehow or a pain?!

proddy, 2021-11-20:
yes, showing the current value would be helpful. I'll add the stripped rows to the table too. I'm not a UI designer so always open for styling suggestions. I'm not really happy with the table - the font is too small and the blue pen icon too large, but couldn't think of a better way to show that a value is writable. Ideas welcome!

glitter-ball, 2021-11-20:
I think you're being a bit hard on yourself, worrying about the UI! This remains a staggering project and a sexy UI could become a distraction and eat up valuable memory and CPU... I don't have an issue with the font, blue pen etc. I think it functions fine for what it's aiming to do and works better than some other offerings I've seen!

pswid, 2021-11-22:
sorry for delay. requested info:

ems-esp:$ watch 0x18
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x18
000+03:40:30.182 N 0: [emsesp] Gateway(0x48) <- Boiler(0x08), UBAMonitorFast(0x18), data: 02 (offset 11)
000+03:40:30.191 N 1: [emsesp] Boiler(0x08) -> Gateway(0x48), UBAMonitorFast(0x18), data: <empty> (offset 11)
000+03:42:07.730 N 2: [emsesp] Gateway(0x48) <- Boiler(0x08), UBAMonitorFast(0x18), data: 02 (offset 11)
000+03:42:07.741 N 3: [emsesp] Boiler(0x08) -> Gateway(0x48), UBAMonitorFast(0x18), data: <empty> (offset 11)
000+03:43:41.455 N 4: [emsesp] Gateway(0x48) <- Boiler(0x08), UBAMonitorFast(0x18), data: 02 (offset 11)
000+03:43:41.469 N 5: [emsesp] Boiler(0x08) -> Gateway(0x48), UBAMonitorFast(0x18), data: <empty> (offset 11)

ems-esp:$ watch 0xE4
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xE4
000+03:44:04.099 N 6: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 22 01 74 24 23 02 00 02 B2 00 00 01 4F 00 00 00 00 01 91 01 74
000+03:44:13.148 N 7: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 22 01 74 24 23 02 00 02 B2 00 00 01 4F 00 00 00 00 01 90 01 74
000+03:44:14.174 N 8: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 22 01 74 24 23 02 00 02 B2 00 00 01 4F 00 00 00 00 01 90 01 74
000+03:44:23.125 N 9: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 22 01 74 24 23 02 00 02 B2 00 00 01 4F 00 00 00 00 01 8F 01 74

ems-esp:$ watch 0x19
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x19
*** nothing captured ***

ems-esp:$ watch 0xE5
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xE5
000+03:50:54.928 N 10: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 00 A0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06
000+03:51:54.904 N 11: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04
000+03:52:54.904 N 12: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04
000+03:53:54.855 N 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03

ems-esp:$ watch 0x34
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x34
000+03:54:20.571 N 14: [emsesp] Gateway(0x48) <- Boiler(0x08), UBAMonitorWW(0x34), data: 02 (offset 3)
000+03:54:20.574 N 15: [emsesp] Boiler(0x08) -> Gateway(0x48), UBAMonitorWW(0x34), data: <empty> (offset 3)
000+03:55:57.993 N 16: [emsesp] Gateway(0x48) <- Boiler(0x08), UBAMonitorWW(0x34), data: 02 (offset 3)
000+03:55:58.006 N 17: [emsesp] Boiler(0x08) -> Gateway(0x48), UBAMonitorWW(0x34), data: <empty> (offset 3)
000+03:57:35.352 N 18: [emsesp] Gateway(0x48) <- Boiler(0x08), UBAMonitorWW(0x34), data: 02 (offset 3)
000+03:57:35.382 N 19: [emsesp] Boiler(0x08) -> Gateway(0x48), UBAMonitorWW(0x34), data: <empty> (offset 3)

pswid, 2021-11-22:
And regarding ww temperatures. In my  case values are swaped - external temperature is higher than internal (tank) temperature:

  ww current intern temperature: 38.4 °C
  ww current extern temperature: 44.2 °C

In attachment all values from my Buderus WSW196i
[show.txt](https://github.com/emsesp/EMS-ESP32/files/7581807/show.txt)

pswid, 2021-11-22:
responses for 0x18 and 0x34 telegrams was <empty>. I don't know why it's not visible

pswid, 2021-11-22:
I mean < empty >
:-)

pswid, 2021-12-03:
any ideas?

proddy, 2021-12-03:
I couldn't find anything in the telegrams that would hint that some of your device values don't exist. I need more time to fully look into it. For the time being we can just block them by this enhancement which I'll add to the upcoming v3.4 release

proddy, 2021-12-21:
@glitter-ball I have a first prototype of this. Wanna take it for a spin? It'll be good to get some feedback

proddy, 2021-12-22:
@MichaelDvP (and other developers) - writing this down so I don't forget. The way disabling of device values is implemented is that it persists a list of all the device value unique id's. These are generated based on the sequence they are added in the code via the register_device_value() function. That means if the sequence changes in the code, the mapping will be incorrect. It means when we add new device values to the code between releases we probably need to manually reset the customization. I choose to use numbers instead of the 'shortnames' (which are in itself unique) to save on memory.

glitter-ball, 2021-12-22:
I’d like to. Things a bit mad here getting ready for Christmas and this won’t be on my issued to-do list ;-) ! But I’ll see if I can sneak some time…GB

proddy, 2021-12-22:
@glitter-ball ok, let me know when you have some time and I'll send over a new binary

g6094199, 2021-12-22:
i would also step in to test

proddy, 2021-12-22:
The code is in https://github.com/emsesp/EMS-ESP32/tree/ft_reacthooks

And here's the binary:  [EMS-ESP-3_4_0b0-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7762645/EMS-ESP-3_4_0b0-ESP32.zip)

Let me know what you think. Remember to refresh your browser cache after the first upgrade

MichaelDvP, 2021-12-22:
Tested with this bin. I see a lot of values with NOT_SET value (255, 3200, 25.1, etc). Exclude list works, but after restarting emsesp it is empty again, seems not to be stored or reloaded. I'll take a deeper look into code later.

proddy, 2021-12-22:
> Tested with this bin. I see a lot of values with NOT_SET value (255, 3200, 25.1, etc). Exclude list works, but after restarting emsesp it is empty again, seems not to be stored or reloaded. I'll take a deeper look into code later.

oh, thought I fixed the exclude list loading problem that but see its broken again. I'll fix it tonight. Appreciate your time in reviewing the changes. There's quite a bit

proddy, 2021-12-22:
I fixed the formatting and checked in into my branch. The loading is strange as it works on my test system but not in product. For some reason the device's unique id (unique_id) is un-set. Need to dig deeper why this is happening or find another way to match exclude entity lists to the device (perhaps by a combination of deviceid/productid).

proddy, 2021-12-22:
new build in the ft_reacthooks branch. Here's the new bin to try out: [EMS-ESP-3_4_0b1-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7765484/EMS-ESP-3_4_0b1-ESP32.zip)

MichaelDvP, 2021-12-23:
Now it's working good, keeps the settings also after update. Checked web/api/mqtt, all as expected.
Is there a reason why you use a dropdown in the exclude list instead of the buttons of dashboard? 
Imo same look and feel would be better, but maybe leads to confusion?

proddy, 2021-12-23:
I had buttons first. Then was worried it took a lot of screen space when there are >4 devices. It starts to look chunky. I'm going to review the new UI design with Kees tonight so maybe I'll end up changing things again.

glitter-ball, 2021-12-23:
Tick box?

proddy, 2021-12-23:
@glitter-ball looks like this now:

[image]

proddy, 2021-12-25:
[EMS-ESP-3_4_0b1-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7775797/EMS-ESP-3_4_0b1-ESP32.zip)

proddy, 2021-12-28:
This is now part of the 3.4 branch. If we find bugs this issue can be re-opened.

closed 2021-12-28 by proddy