emsesp/EMS-ESP32 #207: Circuit command "mode" not working

FredericMa, 2021-11-19:
**Bug description**
I just upgraded to v3.3.0b8 (because of the MQTT reconnect issue I had before).
When I publish `{"cmd":"mode" ,"data":"heat", "id":1}` to the `ems-esp/thermostat` topic. The heating circuit its mode isn't changed.
I only see the following in the logs:
```
2021-11-19 12:11:00.111    INFO 80:     [command]    Calling thermostat command 'values', value cmd, id is default
```

This was working before. I guess I was on some 3.2.1 beta build.

**Steps to reproduce**
Publish `{"cmd":"mode" ,"data":"heat", "id":1}` to the `ems-esp/thermostat` topic.

**Expected behavior**
I expect the mode of circuit 1 to change to `heat`. Now it remains at `nofrost`

**Device information**
This API command to get the info doesn't seems to work anymore. I get this when browsing to `http://192.168.xxx.xxx/api?device=system&cmd=info`:
```
{
message: "missing command in path"
}
```
 Nonetheless, I am using a Junkers FW200 thermostat.

proddy, 2021-11-19:
yup, reproduced. working on a fix...

FredericMa, 2021-11-19:
That's quick! Thanks!

proddy, 2021-11-19:
It's because I switched the key from 'data` to 'value' but really both should be allowed for backwards compatibility as mentioned in https://emsesp.github.io/docs/#/Command?id=definitions

The firmware is building, it'll be b9.

closed 2021-11-19 by proddy

FredericMa, 2021-11-19:
I can confirm that it works. Thanks!

commit 30b111b9 (2021-11-19): Merge pull request #208 from proddy:dev

fixes #207
commit a63f2e61 (2021-11-19): fixes #207 - allow both data and value as MQTT keys