emsesp/EMS-ESP32 #210: I2C: do we have D18B20 support for non JST enabled "EMS interface board" interfaces?

g6094199, 2021-11-20:
hi,

i'm using the "EMS interface board" which doesnt have the JST plug for i2c sensors.
Is it possible to have (to add) the Dallas sensors to the ems-esp without (re)compiling from source?
If there is support where can i find the docu?

proddy, 2021-11-20:
Yes, you just need to wire up the Dallas to the ESP32. One pin to the Ground/GND, one to the power (3.3v or 5v) and one to any of the free ESP32 pins. Then configure it in the EMS-ESP settings. The default GPIO is 4:

[image]

g6094199, 2021-11-20:
jep, i see.
but if i use a preconfigured profile i have no chance to set it. otherwise when i use the custom profile i can set the gpio. on LAN devices LAN wont work anymore without the right settings....
imho for ease of use it would be nice to have the chance to set the gpio (for Dallas only) on every set profile!? exept the BBQKees Gateways which are  ready to go "appliances".

proddy, 2021-11-20:
Oh, yes, that's a really good point I hadn't thought about. Ideally, when choosing 'Custom..' from the board profiles there should be a drop down-menu for Ethernet with the options 'Not Activated', 'LAN8720' and 'TLK110'. 

How urgent is this for you?

g6094199, 2021-11-21:
or the other way round. there is already a point "Use Dallas Sensor parasite power" which covers the same area. So maybe the Whole Dallas thing should be regrouped to a new subtopic called "Sensors" or so?

proddy, 2021-11-21:
the board profile is only about configuring the hardware GPIOs and nothing else (led, dallas, rx, tx, button, and the 3 ethernet GPIOs). All the other settings like the Dallas parasite are software options. I'll just add Ethernet and we're done. I'll see if I can code it today

proddy, 2021-11-21:
I added the option. Can you try this build? https://github.com/proddy/EMS-ESP32/releases

bbqkees, 2021-11-21:
We already have specifically the board profile 'custom' for this.
And you can also just use f.i. the S32 or MH-ET live profile and use the assigned GPIO pin of the profile for the DS18B20.

proddy, 2021-11-21:
But not in combination with Ethernet. It’s added now in 3.3.0b10

closed 2021-11-22 by proddy

g6094199, 2021-11-22:
even if its working. its not intuitive/consistent to the user the way its implemented right now.
havnt checked but to the LAN versions have the same GPIO pinouts than the custom setting?
also for the olimex boards (ESP32-EVB, ESP32-POE, etc) which are 8710, too.
for consistency (its a bit redundant but i dont have a better solution) and easy of use it should be better to have each board in the "board profile" dropdown as before and additionalliy the chance to choose the phy type in the "custom profile"?

proddy, 2021-11-22:
The proper way would be to add these 6 additional LAN settings to the Custom section (if Ethernet is enabled):

    uint8_t       phy_addr;   // I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
    int              power;      // Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
    int              mdc;        // Pin# of the I²C clock signal for the Ethernet PHY
    int              mdio;       // Pin# of the I²C IO signal for the Ethernet PHY
    eth_phy_type_t   type;       // Type of the Ethernet PHY (LAN8720 or TLK110)
    eth_clock_mode_t clock_mode; // ETH_CLOCK_GPIO0_IN or ETH_CLOCK_GPIO0_OUT, ETH_CLOCK_GPIO16_OUT, ETH_CLOCK_GPIO17_OUT for 50Hz inverted clock

But I'm not going to do that. If you want create a PR and contribute back to the project. I already spent a few hours on my weekend trying to help you out.

g6094199, 2021-11-24:
i really appreciate that!
but i´m not a coder and never will be.

proddy, 2021-11-24:
which design did you use for your board? Anything you can share?