emsesp/EMS-ESP32 #211: Rename wwheat

proddy, 2021-11-20:
From #198, @FredericMa mentioned :

> MQTT sensor: `wwheat` from the `ems-esp/boiler_data_ww` is `ON`. This indicates the 3-way valve is open towards the tank. @proddy, now I think about it, `wwheat` it is rather a abstract description for a valve and as in this case it also doesn't mean that any heating is still going on. Maybe something like `ww3waytobuffer` or `wwvalvetobuffer` is more clear?

A lot of the short names for device entities were made up on the spot, loosely translated from what ever I could find in German, Italian of French. And often corrected by users.

`wwheat` got to be called "Warm Water Heating" which I agree is it bit dubious.  BTW you can see the complete list on https://emsesp.github.io/docs/#/MQTT?id=boiler 

so happy to change it and the rest.

glitter-ball, 2021-11-20:
What's the alternate state? Should probably consider that, or is it just ww... on or off?

proddy, 2021-11-20:
it's on/off. The type is a Boolean:

http://ems-esp.local/api/boiler/wwheat

```json
{
  "name": "wwheat",
  "fullname": "ww heating",
  "circuit": "ww",
  "value": "on",
  "type": "boolean",
  "writeable": false,
  "ha_entityname": "sensor.boiler_ww_heating"
}
```

FredericMa, 2021-11-20:
How about `3-way valve towards buffer` or `3-way valve buffer position`? Or is this too long?

proddy, 2021-11-21:
From the original famous [emswiki](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#wwbetriebsart) they call it "**3-way valve on WW**" and on [Norbert's excellent HT3 mapping](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf) he calls it "**3-way valve on storage charge**". We can pick either?

FredericMa, 2021-11-21:
+1 for "3-way valve on WW"

proddy, 2021-11-21:
and for the shortname `wwheat` -> `ww3wayon` ?

FredericMa, 2021-11-22:
Yeah, should be clear enough I guess.

proddy, 2021-11-25:
ok, since it's a breaking change I'll include in 3.4

proddy, 2021-11-29:
Also rename wwdisinfect -> wwdisinfecting

proddy, 2021-12-17:
in 3.4, coming soon to the dev branch

proddy, 2021-12-21:
@MichaelDvP said "rename of wwdisinfect to wwdisinfecting is right for boiler, but not for thermostat"

MichaelDvP, 2021-12-21:
For boiler it is the actual state disinfecting, For thermostat it is the command set disinfection on/off with time schedule of day and hour.

proddy, 2021-12-22:
@MichaelDvP rename `ww3wayon `to  `ww3waycharge` ?

MichaelDvP, 2021-12-22:
We have for the chargetype the enum: `charge pump` or `3-way valve`, so i would prefere ww3wayvalve, but since i don't have it, i don't care much. Only the setting `ww3wayon off` looks odd to me.

proddy, 2021-12-29:
> We have for the chargetype the enum: `charge pump` or `3-way valve`, so i would prefere ww3wayvalve, but since i don't have it, i don't care much. Only the setting `ww3wayon off` looks odd to me.

changed to ww3wayvalve in v3.4 and leaving disinfecting to how it is now.

closed 2021-12-29 by proddy