emsesp/EMS-ESP32 #212: Support for SM100/MS100 to control Warm Water circuit

aHVzY2g, 2021-11-20:
**Is your feature request related to a problem? Please describe.**
*A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]*
Currently, I am not able to control the warm water temperature, which actually was for me the reason to set up EMS-ESP in the first place.

**Describe the solution you'd like**
*A clear and concise description of what you want to happen.*
I would like to be able to control, the warm water temperature via MQTT and web. 

**Additional context**
*Add any other context or screenshots about the feature request here.*
For a more detailed overlook of my setup, please see https://github.com/emsesp/EMS-ESP/issues/359 

The only change since then is that I upgraded to a ESP32. Like @MichaelDvP suggested I did an`watch on 2A`, here you can see the output. I changed the value for warm water from 44 to 45 forward and backward. I also checked https://emsesp.github.io/docs/#/Solar?id=decoded-ems-messages , but it does not seem to be relevant in this context. What are the next steps?

From my understanding, the message:
```
000+01:59:03.854 N 678: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:03.917 N 679: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2C
```
is relevant to implement this feature.

Full `watch on 2a`
```
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x2A
000+01:59:01.798 N 677: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 2F
000+01:59:03.854 N 678: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:03.917 N 679: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2C
000+01:59:06.125 N 680: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2D (offset 10)
000+01:59:06.398 N 681: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 B4
000+01:59:06.581 N 682: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2D (offset 6)
000+01:59:07.397 N 683: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2D (offset 10)
000+01:59:09.401 N 684: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 93 00 93 80 00 00 80 00 80 00 80 00 00
000+01:59:16.722 N 685: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 F8 (offset 4)
000+01:59:17.705 N 686: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:17.769 N 687: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2D
000+01:59:19.375 N 688: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 93 00 92 80 00 00 80 00 80 00 80 00 00
000+01:59:20.201 N 689: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2C (offset 10)
000+01:59:20.845 N 690: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2C (offset 10)
000+01:59:21.920 N 691: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2C (offset 6)
000+01:59:27.800 N 692: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:27.863 N 693: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2C
000+01:59:29.301 N 694: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 00 93 80 00 00 80 00 80 00 80 00 00
000+01:59:30.022 N 695: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2D (offset 10)
000+01:59:30.895 N 696: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2D (offset 6)
000+01:59:31.078 N 697: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2D (offset 10)
000+01:59:31.799 N 698: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 2F
000+01:59:32.005 N 699: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 00 00 04 00 03 00 2D 01 0F
000+01:59:32.638 N 700: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:32.701 N 701: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2D
000+01:59:32.978 N 702: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7AD), data: 64 00 00 03
000+01:59:33.178 N 703: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7B1), data: 01 80 00 01 00 00 00 00
000+01:59:34.177 N 704: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 A6 00 00 01 DE 00 C7 00 C0 80 00 01 C2
000+01:59:34.408 N 705: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+01:59:34.998 N 706: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2C (offset 10)
000+01:59:35.296 N 707: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2C (offset 10)
000+01:59:36.796 N 708: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2C (offset 6)
000+01:59:37.630 N 709: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:37.692 N 710: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2C
000+01:59:39.350 N 711: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 01 2B 80 00 00 80 00 80 00 80 00 00
000+01:59:39.819 N 712: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2D (offset 10)
000+01:59:41.245 N 713: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2D (offset 6)
000+01:59:41.428 N 714: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2D (offset 10)
000+01:59:42.580 N 715: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:42.644 N 716: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2D
000+01:59:44.746 N 717: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2C (offset 10)
000+01:59:45.446 N 718: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2C (offset 6)
000+01:59:45.628 N 719: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2C (offset 10)
000+01:59:47.274 N 720: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:47.337 N 721: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2C
[... 5 lines cut ...]
000+01:59:52.394 N 727: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:52.457 N 728: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2D
000+01:59:54.401 N 729: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2C (offset 10)
000+01:59:54.696 N 730: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2C (offset 10)
000+01:59:55.796 N 731: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2C (offset 6)
000+01:59:56.391 N 732: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+01:59:56.455 N 733: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2C
000+01:59:58.325 N 734: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2D (offset 10)
000+01:59:58.621 N 735: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2D (offset 6)
000+01:59:59.900 N 736: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 00 52 80 00 00 80 00 80 00 80 00 00
000+02:00:00.671 N 737: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2D (offset 10)
000+02:00:02.881 N 738: [emsesp] Thermostat(0x10) <- Solar(0x2A), ParamCfg(0xF9), data: 11 FF 06 A6 0A
000+02:00:02.943 N 739: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2D
000+02:00:03.423 N 740: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 2F
000+02:00:03.608 N 741: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 C5 (offset 4)
000+02:00:04.125 N 742: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2C (offset 10)
000+02:00:04.995 N 743: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2C (offset 6)
[... cut ...]
000+02:00:06.469 N 746: [emsesp] Solar(0x2A) -> Thermostat(0x10), ParamCfg(0xF9), data: FF 06 A6 0A 0F 00 00 00 0F 00 00 00 2D 00 00 00 3C 00 00 00 2C
000+02:00:08.645 N 747: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7A6), data: 2D (offset 10)
000+02:00:09.095 N 748: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 2D (offset 6)
000+02:00:09.700 N 749: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 00 00 80 00 00 80 00 80 00 80 00 00
000+02:00:10.945 N 750: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A6), data: 2D (offset 10)
```

MichaelDvP, 2021-12-07:
Most of this is done in #164 and the ft_solar branch, but there is no activity anymore. You can compile yourself, or use the bins i've posted in #164(, or check my [tests](https://github.com/MichaelDvP/EMS-ESP32/releases/download/latest/EMS-ESP-3_3_1a1-ESP32.bin)).

aHVzY2g, 2021-12-10:
Thank you Michael, this seems to work just fine. I will do some test and let you know if I find anything.

aHVzY2g, 2021-12-11:
From what I can tell looking at this [commit](https://github.com/MichaelDvP/EMS-ESP32/commit/3fc18ede9dd9d352113be000d867ac2039efe8e3#L997) it should be possible to set the wwtemp of the solar circuit now. If try that by using call via telnet or via the dashboard itself, the value does not change. 

_using build EMS-ESP-3_2_2b14-ESP32 from #164_
```
ems:$ watch on 2a
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x2A
000+00:08:19.723 N 0: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwCirc(0x7A5), data: 20
000+00:08:19.760 N 1: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwCirc(0x7A5), data: 00 03 0A 02 FF 1E 32
000+00:08:20.102 N 2: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 07 (offset 8)
000+00:08:20.572 N 3: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwParam(0x7A6), data: 20
000+00:08:20.653 N 4: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwParam(0x7A6), data: FF 05 0F 5F 00 01 3C 3C 3C 3C 28 12 46 01 3C 1E 03 07 3C 00 0F 00 05
000+00:08:20.879 N 5: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 EB 00 EA 80 00 00 80 00 80 00 80 00 00
000+00:08:21.177 N 6: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 ED
000+00:08:21.361 N 7: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 ED (offset 12)
000+00:08:22.750 N 8: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: FF (offset 5)
000+00:08:22.932 N 9: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 9)
000+00:08:29.242 I 10: [command] Calling solar command 'wwtemp', value 50, id is default
000+00:08:30.348 N 11: [emsesp] Me(0x0B) -> Solar(0x2A), SM100wwParam(0x7A6), data: F4 (offset 9)
000+00:08:30.624 N 12: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 5)
000+00:08:31.379 N 13: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 EB 00 EA 80 00 00 80 00 80 00 80 00 00
000+00:08:31.872 N 14: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwParam(0x7A6), data: 20
000+00:08:31.954 N 15: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwParam(0x7A6), data: FF 05 0F 5F 00 01 3C 3C 3C 3C 28 12 46 01 3C 1E 03 07 3C 00 0F 00 05
000+00:08:32.250 N 16: [emsesp] Solar(0x2A) -> All(0x00), SM100wwParam(0x7A6), data: 3C (offset 9)
000+00:08:32.433 N 17: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 90 (offset 12)
000+00:08:34.375 N 18: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0A (offset 9)
000+00:08:35.625 N 19: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 (offset 8)
000+00:08:35.809 N 20: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 A9
000+00:08:36.574 N 21: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 9)
000+00:08:41.455 N 22: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 EB 00 EC 80 00 00 80 00 80 00 80 00 00
000+00:08:41.677 N 23: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 2F
000+00:08:46.625 N 24: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 07 (offset 8)
000+00:08:47.851 N 25: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 EF
000+00:08:48.037 N 26: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 EF (offset 12)
000+00:08:51.200 N 27: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: FF (offset 5)
000+00:08:51.383 N 28: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 9)
000+00:08:52.229 N 29: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 EB 00 EB 80 00 00 80 00 80 00 80 00 00
000+00:08:52.649 N 30: [emsesp] Controller(0x09) <- Boiler(0x08), MC110Status(0x2A), data: 01 (offset 20)
000+00:08:52.653 N 31: [emsesp] Boiler(0x08) -> Controller(0x09), MC110Status(0x2A), data: 00 (offset 20)
000+00:08:55.049 N 32: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 5)
000+00:08:57.977 N 33: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 90 (offset 12)
000+00:09:00.098 N 34: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 (offset 8)
000+00:09:01.698 N 35: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0A (offset 9)
000+00:09:01.903 N 36: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 EB 00 EC 80 00 00 80 00 80 00 80 00 00
000+00:09:04.250 N 37: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 92
000+00:09:04.434 N 38: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 9)
000+00:09:08.416 N 39: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 01 00 01
```

MichaelDvP, 2021-12-11:
Thanks, this is a bug, there is a wrong factor 10: input `5.0`  for 50°C should work. The same for `wwMaxTemp` and `wwRedTemp`.

aHVzY2g, 2021-12-11:
Yes, that does the trick, according to the EMS the temperature is now 50. However, the APP “Easy Remote” reports a value of 40. I will do some more tests and let you know what I find.

MichaelDvP, 2021-12-11:
I've updated the branch on my github and merged the latest dev. Test[ this bin file](https://github.com/MichaelDvP/EMS-ESP32/raw/ft_solar/EMS-ESP-3_3_1b0-ESP32.bin).

aHVzY2g, 2021-12-11:
Works fine, "Easy Remote" still does not show the correct value, but that does not matter. 

One thing I noticed is that the "minimum pump modulation" is displayed wrong, if it's set to 100% EMS reports 20%.

closed 2021-12-11 by aHVzY2g

MichaelDvP, 2021-12-12:
I guess the minimum pump modulation is set in units of 5%, please try with the updated file (same link). 
Is the "Easy Remote" value wrong or the emsesp? 
Your log shows a lot of changing values in telegram 0x7E0, 
offset 5: changes from 0 to 0xFF, something goes off->on
offset 8: 0x07->0x06->0x07
offset 9: 0x00->0x0A->0x00->0x0A->0x05
any idea what is this?

aHVzY2g, 2021-12-12:
- It works, and you are correct, the value is set in 5% steps.
- Great news with the new build, the values also show correctly in the "Easy Remote" app
- I have no clue regarding the log and the frequent changes. At the end of next week, I have access to the unit again and can do some further analyzes.