emsesp/EMS-ESP32 #213: CRC for loaded firmware

glitter-ball, 2021-11-20:
**Is your feature request related to a problem? Please describe.**
Would be great to be certain which firmware version is loaded. Version control should normally cover this, but a CRC would tighten this more.

**Describe the solution you'd like**
When EMS-ESP32 starts up, it calculates a CRC for the firmware that's loaded and displays this on the System page next to the version. The CRC should match the value on the Github download page (so hopefully there's a straightforward way to generate it at both ends!)

proddy, 2021-11-20:
yes, this is already on the to-do list. The idea was also to include the check in the WebUI to ensure a valid binary was installed. The code is there, it just needs to be matched up. I'll include this in v3.4 when all the web code goes in for a service

glitter-ball, 2021-11-20:
Sorry - yet more work!..

proddy, 2022-10-29:
fixed in #637

closed 2022-10-29 by proddy