emsesp/EMS-ESP32 #214: Boiler last error code only reported after restart [v3.3.0b8]

giovanne123, 2021-11-20:
Can it be that within v3.3.0b8 the boiler last error code is not instantly reported?

Saw it in HA when I had an error and in HA the information was not updated.
Had a look to EMS-ESP UI and there it was also not updated.

After a restart of EMS-ESP the new error code was read and could be seen in EMS-ESP UI and in HA.
[image]

(maybe also for thermostat last error code, but can test because have no error ;-))

proddy, 2021-11-20:
there's been quite some discussion about this lately. The error messages are sent on telegrams 0x10, 0x11 or 0xC2 (depending on the boiler model). They are broadcasted periodically, so EMS-ESP is not actively querying these. But I may be wrong. My EMS devices don't support this. @MichaelDvP implemented the logic and is in a better position to answer your question.

giovanne123, 2021-11-24:
hm, I don't think that a new error is periodically reported. Always waiting long time/days and I only see it after restart of EMS-ESP.
[In the meantime updated to 3.3.0b10]

pypb, 2022-03-30:
I'm seeing this issue as well, with 3.3.1. I've had quite a few error codes on my heat pump lately, due to having emptied and refilled the radiator system. The EMS-ESP never picks up the new error code until restart on my boiler.

It is actually currently showing an error code from two days ago, but looking at the alert history on the heat pump control panel there was an error yesterday while I was air bleeding.

If there is a way to actively query for the error code, perhaps it would be a good idea to do that.

MichaelDvP, 2022-03-30:
@giovanne123 , @pypb could you please check what error telegrams are used from your boiler. In terminal use: `read 8 10`, `read 8 11` and `read 8 c2`.
My boiler uses telegram 10 and 11 and these telegrams are published if an error happens (not periodically). Ems-esp only fetches the telegrams on start and waits for changes. If you both have C2 telegram for the error, we need to fetch this periodically. If you have 10/11 telegrams, different boilers handle this different and we also need to fetch these long telegrams often.
Please check also for thermostat errors `read 10 12` and `read 10 A2`

pypb, 2022-03-30:
@MichaelDvP here you go:

```
ems-esp:$ read 8 10
001+17:38:51.242 N 0: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage1(0x10), data: <empty>

ems-esp:$ read 8 11
001+17:38:59.246 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage2(0x11), data: <empty>

ems-esp:$ read 8 c2
001+17:39:04.996 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 08 AC 00 0C 11 41 30 31 14 7E 96 03 0F 1D 12 96 03 0F 1D 14 08 AC 00 10 11 41 30
001+17:39:05.247 N 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 31 14 A4 96 03 04 1C 3B 96 03 0F 1C 3B 08 AC 00 0C 11 41 30 31 14 7E 96 03 12 19 (offset 27)
001+17:39:05.547 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 1C 96 03 06 1C 0E 08 AC 00 0C 11 41 30 31 14 7E 96 03 02 15 0A 96 03 04 15 0C 08 (offset 54)
001+17:39:05.772 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: AC 00 0C 11 41 30 31 14 7E 96 03 0B 08 36 96 03 0D 08 00 08 AC 00 0C 11 41 30 31 (offset 81)
001+17:39:06.122 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 14 7E 96 03 06 01 33 96 03 07 01 2F 08 AC 00 0C 11 41 30 31 14 7E 96 02 17 1C 0F (offset 108)
001+17:39:06.546 N 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 96 03 06 01 12 08 AC 00 0C 09 41 30 31 15 93 96 02 12 1C 03 96 02 12 1C 07 08 AC (offset 135)
001+17:39:06.798 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 0C 09 41 30 31 15 93 96 02 11 1C 38 96 02 12 1C 00 08 AC 00 0C 11 41 30 31 14 (offset 162)
001+17:39:07.247 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 7E 96 02 00 0B 26 96 02 06 0B 01 08 AC 00 10 11 41 30 31 15 4D 96 01 0B 12 1B 96 (offset 189)
001+17:39:07.497 N 10: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 01 0B 12 1C 08 AC 00 0E 11 41 30 31 14 E3 96 01 0B 12 1A 96 01 0B 12 1C 08 AC 00 (offset 216)
001+17:39:07.798 N 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 0C 31 48 30 31 15 4C 96 01 0B 12 1A 96 01 0B 12 1C 08 AC 00 0E 11 41 30 31 14 E3 (offset 243)

ems-esp:$ read 10 12
001+17:39:53.192 N 12: [emsesp] Thermostat(0x10) -> Me(0x0B), RCErrorMessage(0x12), data: <empty>

ems-esp:$ read 10 A2
001+17:39:58.592 N 13: [emsesp] Thermostat(0x10) -> Me(0x0B), RCError(0xA2), data: <empty>
```

MichaelDvP, 2022-03-30:
Thanks, so it's the 0xC2, that is not broadcasted, maybe because it is a very long telegram. 

@proddy What do you think, set it to fetch? It's a quick solution, but making a lot of bus-traffic. 
Or we can try to register the 0xBF actual error message, this is sent on error and on error end, making the logging in 0xC2. 
We can add a simple logic to read the errorlog 0xC2 only if we receive a 0xBF: 
```
// 0xBF
void Boiler::process_ErrorMessage(std::shared_ptr<const Telegram> telegram) {
    EMSESP::send_read_request(0xC2, device_id()); // read last errorcode
}
```

proddy, 2022-03-30:
yup, i like that approach

MichaelDvP, 2022-03-30:
@pypb If you have sometimes errors, please test this approach, I have the[ actual dev with this addition here](https://github.com/MichaelDvP/EMS-ESP32/releases). If it works we can merge to official. My boiler does not have the C2 telegram, I can not test by pulling a sensor or something like that.
If coming from 3.3.x you have to reload the page in browser after update.

pypb, 2022-03-30:
> @pypb If you have sometimes errors, please test this approach, I have the[ actual dev with this addition here](https://github.com/MichaelDvP/EMS-ESP32/releases). If it works we can merge to [...]

I've updated and it works as intended. I provoked an error code by cutting power to the outdoor unit and the error code was published to MQTT within a few seconds. When the error cleared that was updated as well. 

Nice work! :)

proddy, 2022-04-01:
great, closing this issue.

closed 2022-04-01 by proddy