emsesp/EMS-ESP32 #216: Why is there an increase in pump modulation during the night?

zibous, 2021-11-21:
**Question**
I don't understand the increase in the pump modulation, because on some nights it is very high and then again it is not.

The settings are always the same:
- Hot water boiler off between 23:00 - 05:00
- Night temperature decrease between 23:00 - 05:00 20 ° C
- All radiators have the same setting as during the day, the line pressure (flow and return) show no major deviations.

Who is responsible for the increase in pump modulation and why does it only happen at night? 
During the day, operations are normal. 

**Screenshots**

[image]

**Device information**
```
{
  "System": {
    "version": "3.3.0b9",
    "uptime": "000+05:33:12.986",
    "uptime (seconds)": 19992,
    "freemem": 155,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-heizung.siebler.home",
    "SSID": "Microwelle",
    "BSSID": "78:8A:20:2A:70:8D",
    "RSSI": -50,
    "MAC": "A4:E5:7C:4A:08:CC",
    "IPv4 address": "10.1.1.147/255.255.255.0",
    "IPv4 gateway": "10.1.1.254",
    "IPv4 nameserver": "10.1.1.217"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 13202,
    "read requests sent": 1295,
    "write requests sent": 3,
    "incomplete telegrams": 0,
    "tx fails": 19,
    "rx line quality": 100,
    "tx line quality": 99,
    "MQTT": "connected",
    "MQTT publishes": 6484,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Controller",
      "name": "BC25 (DeviceID:0x09, ProductID:125, Version:03.05)"
    }
  ]
}
```

**Additional context**
*Add any other context about the problem here.*

bbqkees, 2021-11-22:
Well the more data you can log, the more you see the quirks of the system :-)
To me it seems the approximate inverse of the selected room temperature time frames, so perhaps as the system cools down the pump start running faster to pump the lower temperature water round to compensate.

zibous, 2021-11-22:
@bbqkees 
> Well the more data you can log, the more you see the quirks of the system :-)

Thank you for the test and information, sounds good 👍 

How can I run the detailed log overnight?
Can you do this via  GUI with`http: //ems-heizung.home/system/log?` with the settings **log level "ALL"**. Is the buffer size 100 big enough?

[image]

Or can I evaluate this via other parameters that are supplied via MQTT? 

Thanks for your help and time.

glitter-ball, 2021-11-28:
Is the problem still happening?
Has it always happened or is it new?
I see you have CW400/Sense II like me. It is possible to stop the pump running at night but you need the right combination of settings. It took me a few weeks to find them...

glitter-ball, 2021-11-28:
Have you a graph of outside temperature vs pump speed? Wonder if it's some form of frost protection cutting in?

zibous, 2021-11-29:
@glitter-ball 

It's gotten a little better. I have lowered the ### hc1 selected room temperature from 25 ° C to **24 ° C.**

But I don't know what the **hc1 comfort temperature** is and whether it has any influence? 

>Have you a graph of outside temperature vs pump speed? 

[image]

I have been looking for an optimal setting for my heating for a long time, but can hardly find any information about it.

Somehow the behavior is strange, because I don't see any connection between room temperatures, outside temperatures and line pressure. The night setback is also moderate - only from 24 ° C to 20 ° C. 

**Controller: BC25 (DeviceID:0x09, ProductID:125, Version:03.05)**
**Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)**
```text
  heating active: off
  warm water active: off
  selected flow temperature: 42.0 °C
  burner selected max power: 100 %
  heating pump modulation: 37 %
  outside temperature: 2.0 °C
  current flow temperature: 42.5 °C
  gas: off
  flame current: 0.0 uA
  heating pump: on
  fan: off
  ignition: off
  heating activated: on
  heating temperature: 60.0 °C
  burner pump max power: 100 %
  burner pump min power: 10 %
  pump delay: 5
  burner min period: 10
  burner min power: 0 %
  burner max power: 100 %
  hysteresis on temperature: -6.0 °C
  hysteresis off temperature: 6.0 °C
  burner current power: 0 %
  burner starts: 1420 times
  total burner operating time: 17 days 16 hours 19 minutes
  total heat operating time: 14 days 3 hours 45 minutes
  total UBA operating time: 171 days 12 hours 1 minutes
  service code: 0Y
  service code number: 204
  maintenance message:
  maintenance scheduled: date
  maintenance set time: 6000 hours
  maintenance set date: 07.07.2023
  ww selected temperature: 57.0 °C
  ww set temperature: 57.0 °C
  ww type: buffer
  ww comfort: eco
  ww flow temperature offset: 40
  ww max power: 100 %
  ww circulation pump available: off
  ww charging type: 3-way valve
  ww hysteresis on temperature: -5.0 °C
  ww hysteresis off temperature: 0.0 °C
  ww disinfection temperature: 70.0 °C
  ww circulation pump frequency: 2x3min
  ww circulation active: off
  ww current intern temperature: 56.7 °C
  ww current extern temperature: 56.7 °C
  ww current tap water flow: 0.0 l/min
  ww storage extern temperature: 56.7 °C
  ww activated: on
  ww one time charging: off
  ww disinfection: off
  ww charging: off
  ww recharging: off
  ww temperature ok: on
  ww active: off
  ww heating: off
  ww starts: 432 times
  ww active time: 3 days 12 hours 34 minutes
```
**Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)**
```text
  error code: (0)
  date/time: 08:48:48 29/11/2021
  floor drying: off
  damped outdoor temperature: 1.9 °C
  floor drying temperature: 0.0 °C
  building: light
  minimal external temperature: -10.0 °C
  ww set temperature: 60.0 °C
  ww mode: own_prog
  ww set low temperature: 45.0 °C
  ww circulation pump frequency: own_prog
  ww charge duration: 60
  ww charge: off
  ww circuit 1 extra: 0.0 °C
  ww disinfection: on
  ww disinfection day: we
  ww disinfection time: 180
  hc1 selected room temperature: 24.0 °C
  hc1 mode: auto
  hc1 mode type: comfort
  hc1 eco temperature: 15.0 °C
  hc1 manual temperature: 24.0 °C
  hc1 comfort temperature: 25.0 °C
  hc1 summer temperature: 17.0 °C
  hc1 design temperature: 45.0 °C
  hc1 offset temperature: 0.0 °C
  hc1 min flow temperature: 25.0 °C
  hc1 max flow temperature: 60.0 °C
  hc1 room influence: 0.0 °C
  hc1 current room influence: 0.0 °C
  hc1 nofrost temperature: 5.0 °C
  hc1 target flow temperature: 42.0 °C
  hc1 heating type: radiator
  hc1 set summer mode: auto
  hc1 summer mode: off
  hc1 control mode: optimized
  hc1 program: prog_1
  hc1 fast heatup: 0 %
```

zibous, 2021-12-01:
I now suspect that it is the water heating. Because whenever "ww charging" or "binary_sensor.boiler_gas" is on, the pump modulation increases.

[image]

I just don't know how to narrow it down so that I can be sure that it is the hot water preparation and not the heating circuit. Also do not understand why this happens at night, because the water heating is switched off from 23:00 to 05:00.

firegnom, 2021-12-01:
Not sure about the heat pump and how it is connected, but just in case I would add to the chart :

**boiler_ww_set_temperature** -  this is the temperature that is requested currently by the system, please note that even if the WW is off during the night, water temp can't fall below this value (at least that what I see on my installation as this value is set to 10C when system is off)
**boiler_ww_current_intern_temperature** - current WW temperature , at least in my case.

So what could be happening is that **boiler_ww_current_intern_temperature** falls below **boiler_ww_set_temperature** but as WW is switched off for some reason heat is not transferring from heatpump to WW

zibous, 2021-12-02:
@firegnom 
Thank you, as I understand it, if the temperature in the hot water tank falls by more than 10 K (° C) below the set temperature, the hot water tank is heated up to the set temperature again. The device then goes into heating mode.

Therefore, if the hot water operation is switched off, the pump may run at 100% and, since the set temperature is not reached, it will run until 5:00 in the morning. Do not understand, because if there is no heating of the hot water tank, it makes no sense that the pump is running.

I will try to make the hot water supply overnight, because I'm not sure which one is more economical. Then I could see if the effect of the pump modulation is reduced.

firegnom, 2021-12-02:
You should be able to see that in Grafana that data should be already in the database I just made a quick sample chart from my instance:
[image]

So if the **boiler_ww_current_intern_temperature** goes below **boiler_ww_set_temperature** and not growing means this is the reason otherwise it is something else. 

Please note I am using **boiler_burner_current_power**  in your case it has to be heat pump modulation

The 3 queries used in his graph : 
```sql
SELECT mean("value") FROM "%" WHERE ("entity_id" = 'boiler_burner_current_power') AND $timeFilter GROUP BY time($__interval) fill(previous)

SELECT mean("value") FROM "°C" WHERE ("entity_id" = 'boiler_ww_set_temperature') AND $timeFilter GROUP BY time($__interval) fill(previous)

SELECT mean("value") FROM "°C" WHERE ("entity_id" = 'boiler_ww_current_intern_temperature') AND $timeFilter GROUP BY time($__interval) fill(previous)
```

zibous, 2021-12-02:
@firegnom 
Thanks, but it is still a mystery to me why the pump modulates so highly at night when everything is in reduced mode. 

[image]

[image]

crbble, 2022-11-17:
Due to this issue still being open some thoughts.

You say that you see that _All radiators have the same setting as during the day, the line pressure (flow and return) show no major deviations._ How do you measure the line pressure? Some pumps run in Delta-P mode with pressures somwhere between 150 mbar and 300 mbar.

Now at night when you switch off the heating the radiator valves open because it is getting cold, pressure from the pump falls and the pump increases its power.

closed 2023-02-19 by proddy