emsesp/EMS-ESP32 #219: Alternative font for log window

sciurius, 2021-11-24:
**Is your feature request related to a problem? Please describe.**
I find it very hard to read the log information in the log window.

**Describe the solution you'd like**
Currently the (built-in) stylesheet has "Courier New, monospace". The Courier New font is rather thin, which makes it hard to read when white on black. Just "monospace" gives the user the possibility to specify a better readable font in the browser settings.

**Additional context**
It is not a very big deal, but highly appreciated (by me and other people with bad eyesight).

proddy, 2021-11-24:
I did experiment with many different fonts and settled on this in the end. I'll add it to 3.3 or perhaps 3.4

proddy, 2021-11-25:
in b11

closed 2021-11-25 by proddy

sciurius, 2021-11-25:
Built and running. Thanks!