emsesp/EMS-ESP32 #222: 3.3.0b10 - Many Errors - "tx line quality": 86 %

zibous, 2021-11-25:
**Question**
Probably not critical, but I am unsettled that I get so many error messages.

[image]

I've tried different cables, but the errors remain the same.
Is there something wrong with my configuration or is it due to the hardware configuration used? 

**Screenshots**

[image]

**Device information**
```
{
  "System": {
    "version": "3.3.0b10",
    "uptime": "003+00:08:09.583",
    "uptime (seconds)": 259689,
    "freemem": 158,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-heizung.home",
    "SSID": "Microwelle",
    "BSSID": "78:8A:20:2A:70:8D",
    "RSSI": -52,
    "MAC": "A4:E5:7C:4A:08:CC",
    "IPv4 address": "10.1.1.147/255.255.255.0",
    "IPv4 gateway": "10.1.1.254",
    "IPv4 nameserver": "10.1.1.217"
  },
  "Status": {
    "bus status": "connected, instable tx",
    "bus protocol": "Buderus",
    "telegrams received": 140954,
    "read requests sent": 1333,
    "write requests sent": 36,
    "incomplete telegrams": 16,
    "tx fails": 197,
    "rx line quality": 100,
    "tx line quality": 86,
    "MQTT": "connected",
    "MQTT publishes": 17144,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0,
    "syslog IP": "10.1.1.96",
    "syslog active": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Controller",
      "name": "BC25 (DeviceID:0x09, ProductID:125, Version:03.05)"
    }
  ]
}
```
[... 879 lines cut ...]
161026:Nov 24 09:16:34 ems-heizung.home telegram 001+21:36:52.976 T 158633: Incomplete Rx: 0B 90 FF 00 20 01 A5 19
161035:Nov 24 09:16:36 ems-heizung.home telegram 001+21:36:55.530 T 158642: Incomplete Rx: 0B 90 FF 19 20 01 A5 D1
163150:Nov 24 09:54:24 ems-heizung.home telegram 001+22:14:42.961 W 160737: Incomplete Rx: E0 08 00 18 00 29 01 6A 64 17 09 01 25 40 80 00 02 C7 80 00 00 68 FF 2D 48 00 C8 00 02 00
169654:Nov 24 11:53:47 ems-heizung.home telegram 002+00:14:06.639 W 167190: Incomplete Rx: E0 08 00 18 00 28 01 7B 64 00 01 01 20 40 80 00 02 B6 80 00 00 00 FF 30 59 00 CC 00 00 00
202794:Nov 24 22:03:13 ems-heizung.home telegram 002+10:23:32.261 W 200052: Incomplete Rx: FE 08 00 18 00 2A 01 90 64 00 01 01 20 40 80 00 02 59 80 00 00 00 FF 30 59 00 CC 00 00 00
212017:Nov 25 00:52:24 ems-heizung.home telegram 002+13:12:43.633 W 209193: Incomplete Rx: F8 08 00 34 00 0A 02 34 02 34 20 00 00 03 00 00 13 09 00 01 90 00 80 00
220229:Nov 25 03:24:01 ems-heizung.home telegram 002+15:44:19.881 W 217351: Incomplete Rx: FE 08 00 34 00 0A 02 23 02 23 20 00 00 03 00 00 13 09 00 01 90 00 80 00
236732:Nov 25 08:27:24 ems-heizung.home telegram 002+20:47:43.298 W 233664: Incomplete Rx: F8 08 00 18 00 29 01 96 64 00 01 01 20 40 80 00 02 04 80 00 00 00 FF 30 59 00 CC 00 00 00
244655:Nov 25 10:52:55 ems-heizung.home telegram 002+23:13:14.559 E 241517: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 5A 00 00 64 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
244674:Nov 25 10:53:04 ems-heizung.home telegram 002+23:13:23.013 E 241536: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 5A 00 00 64 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
244711:Nov 25 10:53:17 ems-heizung.home telegram 002+23:13:36.838 E 241573: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
244740:Nov 25 10:53:34 ems-heizung.home telegram 002+23:13:52.852 E 241602: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
245481:Nov 25 11:05:47 ems-heizung.home telegram 002+23:26:06.590 E 242343: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 5A 00 00 64 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
245503:Nov 25 11:05:56 ems-heizung.home telegram 002+23:26:15.674 E 242365: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 5A 00 00 64 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
245526:Nov 25 11:06:11 ems-heizung.home telegram 002+23:26:30.827 E 242388: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 5A 00 00 64 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
245554:Nov 25 11:06:27 ems-heizung.home telegram 002+23:26:46.242 E 242416: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 5A 00 00 64 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
245576:Nov 25 11:06:36 ems-heizung.home telegram 002+23:26:55.261 E 242438: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
245617:Nov 25 11:06:54 ems-heizung.home telegram 002+23:27:13.197 E 242479: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
245647:Nov 25 11:07:11 ems-heizung.home telegram 002+23:27:30.822 E 242509: Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

```

*all log entries*

[ems_heizung.log.zip](https://github.com/emsesp/EMS-ESP32/files/7602331/ems_heizung.log.zip)

proddy, 2021-11-25:
that is not normal. You should see the odd failed Rx or Tx but not that many. You see the really count by going into the the Console and typing `show ems`. Look for "#incomplete telegrams"

So the Rx is cutting out. The break signal is being detected and this could mean
 - loss of data in the cabling
 - cable is too long
 - you have a specific boiler which uses very special timing (Tx mode). We had about 10 in the past and now made it 3 which works for most people. Michael spent a lot of time on this
 - faulty board

has it always been this bad? Did you build the circuit yourself? Is it the same using the jack or EMS lines directly? Is the 5V power to the ESP32 stable?

zibous, 2021-11-25:
@proddy 

> cable is too long

I think not, only 10cm

> Did you build the circuit yourself? 

No bought at bbqkees-electronics.nl 

> Is it the same using the jack or EMS lines directly? 

Yes

> Is the 5V power to the ESP32 stable?

i think so, connected on plug power supply.

```
show ems
EMS Bus is connected, but Tx is not stable.

EMS Bus info:
  Tx mode: 2
  Bus protocol: Buderus
  #telegrams received: 141872
  #read requests sent: 1333
  #write requests sent: 36
  #incomplete telegrams: 16
  #tx fails (after 3 retries): 197
  Rx line quality: 100%
  Tx line quality: 86%

Rx Queue is empty
Tx Queue is empty
```

bbqkees, 2021-11-25:
If RX is perfect, but TX is not I suspect the TX mode.
Is your boiler EMS or EMS+? Try outTX mode EMS instead of EMS+ (See Settings).
Indeed the RC310 and one other EMS thermostat is polarity sensitive. All other EMS thermostats are not polarity sensitive.
Don't know why they did this. If you have a RC310 in your system you have to observe polarity on all EMS devices.
I believe the right pin on the screw terminal is the EMS+ pin, the left on EMS-.

zibous, 2021-11-25:
@bbqkees 

Thanks for your information. 

> Is your boiler EMS or EMS+?

I don't know or I can't find any information about it in the manuals Bruderus Buderus GB172 with RC310. 

> Try outTX mode EMS instead of EMS+ 

I will use EMS mode instead of EMS + for the near future.

bbqkees, 2021-11-25:
Is it the old GB172 or the new one? Buderus is reusing model numbers now.
But just try EMS instead of EMS+ and see if that helps.
Reboot the Gateway so the counter resets.

zibous, 2021-11-25:
> Is it the old GB172 or the new one? Buderus is reusing model numbers now.

I hope a new one, installed 07/2021. GB172 14 (DE -II2ELL38P AT,CH,LU II2H3P), 
Logomax Plus and there are a lot of  model numbers and serial numbers on each
part..

> But just try EMS instead of EMS+ and see if that helps.
> Reboot the Gateway so the counter resets.

Yes i will try it, thanks

bbqkees, 2021-11-26:
Then it's likely EMS+ but I am not sure.
There are so many different boiler configurations, might still be 'normal' EMS.

zibous, 2021-11-28:
If i use EMS all is now o.k

```
EMS Bus is connected.

EMS Bus info:
  Tx mode: 1
  Bus protocol: Buderus
  #telegrams received: 102676
  #read requests sent: 13136
  #write requests sent: 20
  #incomplete telegrams: 7
  #tx fails (after 3 retries): 12
  Rx line quality: 100%
  Tx line quality: 100%

Rx Queue is empty
Tx Queue is empty

ems-heizung.siebler.home:$ showsystem
Command not found
ems-heizung.siebler.home:$ show system
Uptime: 001+15:36:36.318
SDK version: v3.3.5-1-g85c43024c
CPU frequency: 240 MHz
Free heap: 174448 bytes
```

Thanks.

proddy, 2021-11-28:
great, so to fix this you switched from using the 3.5" jack plug to using the 2 EMS wires?

zibous, 2021-11-28:
@proddy 
Yes, I use the 2 wire solution because I can simply connect it. With the 3.5 "jack plug I had to re-solder the cable because the polarity of the original cable does not match.

I hope I have no disadvantages if I switch from EMS + to EMS. 

```
EMS Bus info:
  Tx mode: 1
  Bus protocol: Buderus
  #telegrams received: 117002
  #read requests sent: 14831
  #write requests sent: 23
  #incomplete telegrams: 9
  #tx fails (after 3 retries): 13
  Rx line quality: 100%
  Tx line quality: 100%
```

```
Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)
  heating active: off
  warm water active: off
  selected flow temperature: 40.0 °C
  burner selected max power: 100 %
  heating pump modulation: 41 %
  outside temperature: 3.7 °C
  current flow temperature: 35.5 °C
  gas: off
  flame current: 0.0 uA
  heating pump: on
  fan: off
  ignition: off
  heating activated: on
  heating temperature: 60.0 °C
  burner pump max power: 100 %
  burner pump min power: 10 %
  pump delay: 5
  burner min period: 10
  burner min power: 0 %
  burner max power: 100 %
  hysteresis on temperature: -6.0 °C
  hysteresis off temperature: 6.0 °C
  burner current power: 0 %
  burner starts: 1396 times
  total burner operating time: 17 days 5 hours 24 minutes
  total heat operating time: 13 days 17 hours 15 minutes
  total UBA operating time: 170 days 16 hours 57 minutes
  last error code: 0Y(276) 24.11.2021 03:51
  service code: 0Y
  service code number: 204
  maintenance message:
  maintenance scheduled: date
  maintenance set time: 6000 hours
  maintenance set date: 07.07.2023
  ww selected temperature: 57.0 °C
  ww set temperature: 57.0 °C
  ww type: buffer
  ww comfort: eco
  ww flow temperature offset: 40
  ww max power: 100 %
  ww circulation pump available: off
  ww charging type: 3-way valve
  ww hysteresis on temperature: -5.0 °C
  ww hysteresis off temperature: 0.0 °C
  ww disinfection temperature: 70.0 °C
  ww circulation pump frequency: 2x3min
  ww circulation active: off
  ww current intern temperature: 58.5 °C
  ww current extern temperature: 58.5 °C
  ww current tap water flow: 0.0 l/min
  ww storage extern temperature: 58.5 °C
  ww activated: on
  ww one time charging: off
  ww disinfection: off
  ww charging: off
  ww recharging: off
  ww temperature ok: on
  ww active: off
  ww heating: off
  ww starts: 428 times
  ww active time: 3 days 12 hours 9 minutes

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)
  error code: (0)
  date/time: 13:52:26 28/11/2021
  floor drying: off
  damped outdoor temperature: 2.7 °C
  floor drying temperature: 0.0 °C
  building: light
  minimal external temperature: -10.0 °C
  ww set temperature: 60.0 °C
  ww mode: own_prog
  ww set low temperature: 45.0 °C
  ww circulation pump frequency: own_prog
  ww charge duration: 60
  ww charge: off
  ww circuit 1 extra: 0.0 °C
  ww disinfection: on
  ww disinfection day: we
  ww disinfection time: 180
  hc1 selected room temperature: 24.0 °C
  hc1 mode: auto
  hc1 mode type: comfort
  hc1 eco temperature: 15.0 °C
  hc1 manual temperature: 24.0 °C
  hc1 comfort temperature: 25.0 °C
  hc1 summer temperature: 17.0 °C
  hc1 design temperature: 45.0 °C
  hc1 offset temperature: 0.0 °C
  hc1 min flow temperature: 25.0 °C
  hc1 max flow temperature: 60.0 °C
  hc1 room influence: 0.0 °C
  hc1 current room influence: 0.0 °C
  hc1 nofrost temperature: 5.0 °C
  hc1 target flow temperature: 40.0 °C
  hc1 heating type: radiator
  hc1 set summer mode: auto
  hc1 summer mode: off
  hc1 control mode: optimized
  hc1 program: prog_1
  hc1 fast heatup: 0 %

Controller: BC25 (DeviceID:0x09, ProductID:125, Version:03.05)

proddy, 2021-11-28:
ok, thanks. I'll update the documentation incase other people run into the same problem

zibous, 2021-11-28:
@proddy 

I get now a few error messages with TX mode "**EMS**",  but I only notice that there is almost always an error after "**Setting warm tap water on**", but could it also be a coincidence?

```
2021-11-27 20:39:02.623    INFO 73:     [boiler]     Setting warm tap water on

2021-11-27 20:46:18.998    INFO 74:     [command]    Calling boiler command 'wwtapactivated', value false, id is default
2021-11-27 20:46:18.998    INFO 75:     [boiler]     Setting warm tap water off
2021-11-27 20:46:28.999    INFO 76:     [command]    Calling boiler command 'wwtapactivated', value true, id is default

2021-11-27 20:46:28.999    INFO 77:     [boiler]     Setting warm tap water on
2021-11-27 20:46:31.670   ERROR 78:     [telegram]   Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2021-11-27 23:18:21.608 WARNING 79:     [telegram]   Incomplete Rx: F8 08 00 34 00 0A 02 3B 02 3B 20 00 00 03 00 00 13 A5 00 01 A9 00 80 00
2021-11-28 07:32:31.537 WARNING 80:     [telegram]   Incomplete Rx: F8 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00
2021-11-28 09:12:59.036 WARNING 83:     [telegram]   Incomplete Rx: FE 08 00 18 00 2A 01 C6 64 17 09 01 25 40 80 00 02 01 80 00 00 71 FF 2D 48 00 C8 00 02 00
2021-11-28 09:13:06.322    INFO 85:     [shell]      Shutdown console pty0 for connection from [10.1.1.98]:50093
2021-11-28 11:20:20.691 WARNING 86:     [telegram]   Incomplete Rx: FE 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00
2021-11-28 11:53:53.568    INFO 87:     [command]    Calling boiler command 'wwtapactivated', value false, id is default
2021-11-28 11:53:53.568    INFO 88:     [boiler]     Setting warm tap water off

2021-11-28 12:01:30.950    INFO 94:     [boiler]     Setting warm tap water on
2021-11-28 12:01:34.792   ERROR 95:     [telegram]   Last Tx Write operation failed after 3 retries. Ignoring request: 0B 08 1D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

.....
```

bbqkees, 2021-11-29:
@zibous Did you also try to switch the EMS wires on the RC310?
It's the first time I hear that the polarity of the jack is not correct, and there are lots of RC300/RC310 users out there already.
So I think it's something specific to your setup.

zibous, 2021-11-29:
@bbqkees 
> So I think it's something specific to your setup.
Maybe, i use t2 wire from the EMS on the Buderus GB172, works perfect.

I use the Gateway S32 (Standard Edition) w/o any modification, but when i use the original jack cable the RC310 is deactivated. Same as when i do not have the correct polarity with the 2 wire cable.
No problem, the 2 wire cable is working.

closed 2023-02-19 by proddy