emsesp/EMS-ESP32 #224: Add LED status indication for EMS bus connected

bbqkees, 2021-11-29:
Currently the LED blinks fast until WiFi is connected, then it will start blinking slowly until EMS has been found.
But if WiFi/network has not been connected, but EMS is, it will still blink fast forever. 
So if you plug in a new Gateway (with WiFi not configured yet), you have no idea if it found the EMS bus or not. 

To enhance the setup procedure, it would be better if we can show that the EMS bus was found and connected (so both RX/TX working).

Maybe we could introduce a second blink mode to indicate EMS fully connected yes, network no?
Like 2 short flashes and a longer pause?

glitter-ball, 2021-12-04:
Maybe there are two flashes in the sequence - one for network, one for EMS? Potentially a third for MQTT?

Healthy for each is 800ms on:200ms off, unhealthy is 200ms on:800ms off (or other numbers which look nice to humans). So two/three long pulses is all good. Any short pulse indicates something bad. Short gap - 2 seconds? - between each group?

L=Long flash, S=Short flash
L-L-L : all good
S-L-S : network failed, EMS good, MQTT failed (since network has failed)
L-S-S : network good, EMS failed, MQTT failed
L-L-S : network good, EMS good, MQTT failed
etc. or is this getting a bit much?

bbqkees, 2021-12-07:
Lets not go overboard. Its hard to decipher elaborate blinking variations.
It mostly comes in handy when you are setting up the device for the first time.
I think a solid EMS connection is the most useful indication, followed by a blink indication for network connected.
MQTT is not something you would need a hardware LED for, that's a kind of issue you need network and EMS connected anyway and must be troubleshooted from the pc.

proddy, 2022-01-03:
I've done the first implementation in v3.4 for you to try out @bbqkees . It works like

- on boot, no LEDs, all dark, so no more fast neurotic flashing 
- if there are errors you will see a series of short flashes
- 1 flash is the EMS bus is not connected
- 2 flashes if the network (wifi or ethernet) is not connected
- 3 flashes is both the bus and the network are not connected. Then you know you're truly f*cked.
- a solid LED will show only if 'Hide LED' is not selected in the Settings

closed 2022-01-03 by proddy