emsesp/EMS-ESP32 #225: Unrecognized EMS devices.

anothersupermario, 2021-11-29:
Hello,

first, I want to thank for this software and all the effort done.
In my setup I get a lot of log entries like

_[emsesp] Unrecognized EMS device (device ID 0x08, no product ID). Please report on GitHub.
[emsesp] Unrecognized EMS device (device ID 0x10, no product ID). Please report on GitHub.
[emsesp] Unrecognized EMS device (device ID 0x11, no product ID). Please report on GitHub.
[emsesp] Unrecognized EMS device (device ID 0x09, no product ID). Please report on GitHub._

In addition, there are many log entries like:
_[telegram] Incomplete Rx:
[telegram] Incomplete Rx: BA 10 0F 00 F3 A3 00 00 87 04 8D 01 7A 40 EF 6F 7F 5A 10
[telegram] Incomplete Rx: 64 C4 87
[telegram] Incomplete Rx: 6F C8 7A_

On Rx (Pin 23) the signal looks like this.
Can somebody tell what to do to get rid of the messages in the log and  the whole thing to work, or does this look like a problem on the input side?
We are using a BUDERUS GB142 with RC30, MM10 and WM10 installed.

Thank you,
Mario

[log.txt](https://github.com/emsesp/EMS-ESP32/files/7621290/log.txt)
[system.txt](https://github.com/emsesp/EMS-ESP32/files/7621314/system.txt)

proddy, 2021-11-29:
the Rx looks garbled so I'm thinking its a hardware or wiring problem. Check the cables and the polarity - try the EMS lines and also the jack.  If you bought an interface board from BBQKees he's got some good tips on what and what not to use. Also check out https://emsesp.github.io/docs/#/Troubleshooting?id=ems-connectivity

anothersupermario, 2021-11-29:
Thank you for the reply. The interface was built by my on and was successfully used with the AVR Net IO (described [here](https://emswiki.thefischer.net/doku.php?id=wiki:ems:net_io ) before.
Now used with 3.3 V the Rx line looks like this (with 1 V / div):

[image]

bbqkees, 2021-11-30:
The problem here is not unrecognized EMS devices but the EMS bus signal is not properly converted to 'normal' serial such that the ESP32 can decode it. So it's something in your hardware path.
If you recreated the Net IO schematic from your link exactly it should actually work fine with an ESP32.

The RX line needs to be pulled lower, are you sure ground is connected to the board?

Although I do not think it will make a lot of difference but one thing you can try is to lower R5.
A few years back with the oldest of my interface boards you could switch between 4K7 and 100R, that helped with some ESP boards.

anothersupermario, 2021-12-01:
Thank you for the reply. I've changed R5 to 100R, but there where no good results so far. 
The "screenshot" was taken with wrong connections. Right now it looks like this:
[image]

High and low levels are valid.
It seems I have to buy one of the boards ;-) to replace my old solution.

anothersupermario, 2021-12-07:
Works with the "EMS interface board" by bbqkees.

closed 2023-02-19 by proddy