emsesp/EMS-ESP32 #228: lastcode broke MQTT JSON structure

firegnom, 2021-11-30:
**Isssue with the lastCode json format**

Welcome, all I am very fresh in this community so sorry if this is already reported. 
I am running version :
```
EMS-ESP Version: v3.3.0
Device (Platform / SDK): ESP32 / v3.3.5-1-g85c43024c

```
My new boiler is Buderus GB212  and i just installed EMS-ESP , after the first configuration all worked with no problem, but then I have started to see that in MQTT Explorer the ems-esp/boiler_data was not parsed correctly by the MQTT server .

it turned out that the issue was:
```json
"lastcode":" 4/7760 start: 2060.149.11 11:1, end: none"
```

when I use lint on that message i get the error : 
```
Invalid characters in string. Control characters must be escaped.json(262)
```

I have attached sample MQTT messages - [sample.json.txt](https://github.com/emsesp/EMS-ESP32/files/7629055/sample.json.txt)
System log, that I have gathered during restart - [ALL.log](https://github.com/emsesp/EMS-ESP32/files/7629085/ALL.log)

I would really appreciate the help here.

firegnom, 2021-11-30:
Can confirm the same issue on : 
```
EMS-ESP Version
v3.3.1b0
```

```json
"lastcode":"�\f/7760 start: 2021.11.19 12:24, end: none"
```

[sample.json.2.txt](https://github.com/emsesp/EMS-ESP32/files/7629228/sample.json.2.txt)

firegnom, 2021-11-30:
Switched to : 

```
EMS-ESP Version
v3.2.1
```

All works correctly but of course last code is not in json

proddy, 2021-11-30:
It’s a simple fix. I’ll look into it later this week and make a patch.
Thanks for reporting.

pswid, 2021-12-01:
it's because in commit #220 following condition has been removed. 

if (telegram->offset > 0 || telegram->message_length < 14) {
     return;
}

For decoding last error code, we need 20 bytes starting from offset 0. So, now condition should look e.g. like this. Am I right?
if (telegram->offset > 0 || telegram->message_length < 20) {
    return;
 }

proddy, 2021-12-01:
yes, that's it. We need those extra bytes to accommodate the extra data. Do you want to fix with a PR?

proddy, 2021-12-04:
@firegnom can you try the latest dev build so we can close this, if it works

firegnom, 2021-12-06:
Hi, I can confirm It works now. I was running the latest version over the weekend all works now. 

Thank you!

closed 2021-12-06 by firegnom