emsesp/EMS-ESP32 #237: Make climate entity configurable or disable climate entity in WebUI

tschamm, 2021-12-04:
**Is your feature request related to a problem? Please describe.**
I have two heating circuits in my boiler. While one is reporting the current room temperature, the other does not have a room temp associated. 

**Describe the solution you'd like**
A feature to configure the climate entities in the WebUI would allow to either disable a climate entity related to a heating circuit or to configure the current temperature to a value reported by EMS (e.g. boiler current flow temp). 

**Describe alternatives you've considered**
Disabling the climate entity in home assistant.

**Additional context**
[image]
Attached a screenshot with two heat circuits, one which does not has a current room temp because it is connect to a buffer water tank, the other one is a mixed heat circuit.

proddy, 2022-01-03:
@tschamm if you use MQTTExplorer or something similar, how many entries do you have under `homeassistant/climate/ems-esp`? Is it showing you two config entries for each of the heating circuits?

tschamm, 2022-01-03:
Two config entries are listed, one for each hc.

proddy, 2022-01-03:
ok thanks. Just thinking of what the best way forward is here.

proddy, 2022-01-08:
@tschamm if you want to try it, build the latest from the v3.4 branch or grab the binary from Discord

tschamm, 2022-01-09:
Thanks, works fine for me!

commit 59f4e942 (2022-01-08): ha improvements, remove hamode/hatemp, fixes #237
closed 2022-01-10 by proddy