emsesp/EMS-ESP32 #240: Error 500 while uploading a new bin file

rvanderven, 2021-12-05:
I tried to download the bin file for esp32 for 3.3.0 as well as 3.3.1b0 and get an error 500 while uploading the bin file (both drag/drop as well as opening from file explorer.
Next to Chrome I also tried using Safari.

Any idea why this happens.

proddy, 2021-12-05:
no, haven't seen this before. Perhaps a timeout issue. Is it Wifi or Ethernet? If WifI, is the Wifi strength strong? Does it happen all the time? When it fails at what %?

rvanderven, 2021-12-05:
Thanks for your respons.
I forgot to mention that I have the feeling the bin file is not downloaded in full but I dont get a error there.
So it might be that that is causing the issue.
I tried both my tablet as well as my macbook.
I did all over wifi which normally isnt an issue.
I will go for your option to try a cable connection . Will keep you updated

rvanderven, 2021-12-05:
I think I have a download issue: the download bar in the top (Chrome) stops just before the end, but I don't get a timeout error or whatsoever. And with that the .bin is most likely incomplete

Nunak, 2021-12-06:
I have same issue. I did try on Android phone and also on Mac ever got Error 500 while uploading. I think it do a 99% of upload.

proddy, 2021-12-06:
wow, ok, so that's 2 people now. Never heard of it before. Can you both try using platformio and OTA? Also is there anything different with your setups, like do you have Telnet console open at the same time or other browser windows?

It may be related to https://github.com/emsesp/EMS-ESP32/issues/177

rvanderven, 2021-12-06:
I think it already goes wrong at download. Proddy: did you try a download from github with after that a successful upload? If so are you able to offer us a bin file via another way?

rvanderven, 2021-12-06:
Update: I downloaded my current used 3.2.1 bin file, compared it with the one I had: it had the same size. I uploaded it successfully. Also my browser download bar at the top stopped at the same location (not fully to the right). 
I checked the article #177 and tried the ota option: I get "No answer". Will try esptool tomorrow

rvanderven, 2021-12-06:
Update #2: I used ota to upload my 3.2.1.bin again and that works. So this makes me think the issue is pointing to the downloaded bin file for 3.3.0 and 3.3.1b

MichaelDvP, 2021-12-07:
@rvanderven I guess your OTA partition is to small, you are using the default partion table and not the one from emsesp. Maybe you have flashed first another software (espeasy, tasmota, emsesp2.0) or used a flash-tool that uses only the default partition table. (You can check in System if the FileSystem-partition is 64k (right value for emsesp)). Flash with USB to use the right partition table from the bin.

rvanderven, 2021-12-07:
@MichaelDvP : according my system info I have 64Kb
[image]
I can try a USB upload

MichaelDvP, 2021-12-07:
> according my system info I have 64Kb

Oh, thats only 60k (it should be 65536 bytes), very strange, i don't know what software uses this partitons scheme.
None of the [standard tables](https://github.com/espressif/arduino-esp32/tree/master/tools/partitions) use this.
> I can try a USB upload

Yes do this. Use the espressive tool or a gui version that is based on it.

rvanderven, 2021-12-07:
Solved: it took some effort, but I managed to flash my esp32 via usb.
I needed to update my esptool.py. esphomeflasher didn't work (strange error on 'Encrypfiles', also error on littleFS), so in the end I used the esptool example as mentioned on the site of bbqkees (upload firmware) with the downloaded additional .bin files as available on this github location.
I also downloaded the 3.3.1b0.bin file and uploaded it to see whether this went without any error and so it did.
Almost forgot:  I also used 'esptool.py erase_flash' to reset my flash. Not sure whether this was really necessary, but I tried because of the errors I received with the esphomeflasher tool.
So I needed to configure the device again, but that was my least problem. 

Thanks all for reaching out to me. 
@Nunak : I hope this helps for you as well.

Nunak, 2021-12-07:
Thank you @rvanderven i did upload via USB and esp flasher and it's working for me also.

proddy, 2021-12-08:
@rvanderven I expect it was the incorrect File system partition as Michael pointed out, which you fixed with the erase_flash. I'll add a comment to https://emsesp.github.io/docs/#/Uploading-firmware explaining this if others run into the same problem.

There's also a Enhancement on the EMS-ESP todo list for to validate the CRC after an upload, instead of throwing up the HTTP 500 error.

rvanderven, 2021-12-08:
Sounds reasonable as well. Thanks for the followup

Grt.
Richard

closed 2023-02-19 by proddy