emsesp/EMS-ESP32 #241: Refactor Dallas

proddy, 2021-12-11:
As part of the 3.4 cleanup the Dallas console needs some code cleanup too. Things to look at
- the custom sensor names and temperature offsets are stored in the Settings file like `"sensor_id0":"","sensor_name0":"","sensor_offset0":0,"sensor_id1":"","sensor_name1":"","sensor_offset1":0,"sensor_id2":"","sensor_name2":"","sensor_offset2":0,"sensor_id3":"","sensor_name3":"","sensor_offset3":0,"sensor_id4":"","sensor_name4":"","sensor_offset4":0,"sensor_id5":"","sensor_name5":"","sensor_offset5":0,"sensor_id6":"","sensor_name6":"","sensor_offset6":0,"sensor_id7":"","sensor_name7":"","sensor_offset7":0,"sensor_id8":"","sensor_name8":"","sensor_offset8":0,"sensor_id9":"","sensor_name9":"","sensor_offset9":0,"sensor_id10":"","sensor_name10":"","sensor_offset10":0,"sensor_id11":"","sensor_name11":"","sensor_offset11":0,"sensor_id12":"","sensor_name12":"","sensor_offset12":0,"sensor_id13":"","sensor_name13":"","sensor_offset13":0,"sensor_id14":"","sensor_name14":"","sensor_offset14":0,"sensor_id15":"","sensor_name15":"","sensor_offset15":0,"sensor_id16":"","sensor_name16":"","sensor_offset16":0,"sensor_id17":"","sensor_name17":"","sensor_offset17":0,"sensor_id18":"","sensor_name18":"","sensor_offset18":0,"sensor_id19":"","sensor_name19":"","sensor_offset19":0}`. They will be moved to a new configuration service using the unique ID to identify the chip and only storing the alias if there is one.
- the Settings file is read each time the Dashboard is rendered which is costly, so move the name, id, offset to the Dallas  Sensor class object
- Remove the Dallas Sensor Format as it's no longer needed. The Name is the only identifier
- Fix the MQTT, as it's sending the changed value twice each time

glitter-ball, 2021-12-11:
I was going to say good spot on detecting the duplicate MQTT transmission - I'd not noticed that.

That's because I've got mine set to publish on change (setting=0) and it doesn't publish duplicate records...
[image]

... in case that's useful info!

glitter-ball, 2021-12-11:
I set it to publish at 10-second intervals and it still doesn't send duplicate messages. Weird... I'm running v3.3.0b11.

proddy, 2021-12-11:
> I set it to publish at 10-second intervals and it still doesn't send duplicate messages. Weird... I'm running v3.3.0b11.

it's only sometimes. I think its a rounding error, comparing 19.00000 against 19.00001 but haven't looked into it yet

MichaelDvP, 2021-12-11:
Publish interval 10 sec sends out every 10 sec, not 10 sec after the last. If one message is delayed because of other queued messages (maybe HA), then the time to next message is shorter. But your example shows 10:06:29 and 10:06:39, i can't see a double message.

proddy, 2021-12-15:
I've included this in 3.4 which I'll merge into dev this weekend

closed 2021-12-15 by proddy